ptrkrysik/gr-gsm #514: some problems when run command

vespertillo, 2020-03-28:
my hardware info
Linux 5.4.0-kali4-amd64 x86_64
Distributor ID:	Kali
Description:	Kali GNU/Linux Rolling
Release:	2020.1
Codename:	kali-rolling
processor	: 0
vendor_id	: AuthenticAMD
cpu family	: 22
model		: 0
model name	: AMD E1-2100 APU with Radeon(TM) HD Graphics
stepping	: 1
microcode	: 0x700010f
cpu MHz		: 864.471
cache size	: 1024 KB
physical id	: 0
siblings	: 2
core id		: 0
cpu cores	: 2
apicid		: 0
initial apicid	: 0
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep_good nopl nonstop_tsc cpuid extd_apicid aperfmperf pni pclmulqdq monitor ssse3 cx16 sse4_1 sse4_2 movbe popcnt aes xsave avx f16c lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw ibs skinit wdt topoext perfctr_nb bpext perfctr_llc hw_pstate proc_feedback ssbd vmmcall bmi1 xsaveopt arat npt lbrv svm_lock nrip_save tsc_scale flushbyasid decodeassists pausefilter pfthreshold overflow_recov
bugs		: fxsave_leak sysret_ss_attrs null_seg spectre_v1 spectre_v2 spec_store_bypass
bogomips	: 1996.31
TLB size	: 1024 4K pages
clflush size	: 64
cache_alignment	: 64
address sizes	: 40 bits physical, 48 bits virtual
power management: ts ttp tm 100mhzsteps hwpstate proc_feedback

processor	: 1
vendor_id	: AuthenticAMD
cpu family	: 22
model		: 0
model name	: AMD E1-2100 APU with Radeon(TM) HD Graphics
stepping	: 1
microcode	: 0x700010f
cpu MHz		: 829.922
cache size	: 1024 KB
physical id	: 0
siblings	: 2
core id		: 1
cpu cores	: 2
apicid		: 1
initial apicid	: 1
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep_good nopl nonstop_tsc cpuid extd_apicid aperfmperf pni pclmulqdq monitor ssse3 cx16 sse4_1 sse4_2 movbe popcnt aes xsave avx f16c lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw ibs skinit wdt topoext perfctr_nb bpext perfctr_llc hw_pstate proc_feedback ssbd vmmcall bmi1 xsaveopt arat npt lbrv svm_lock nrip_save tsc_scale flushbyasid decodeassists pausefilter pfthreshold overflow_recov
bugs		: fxsave_leak sysret_ss_attrs null_seg spectre_v1 spectre_v2 spec_store_bypass
bogomips	: 1996.31
TLB size	: 1024 4K pages
clflush size	: 64
cache_alignment	: 64
address sizes	: 40 bits physical, 48 bits virtual
power management: ts ttp tm 100mhzsteps hwpstate proc_feedback
i try this app for fun,simply try to found my phone in 2G but with no result,if i remember but i am not sure, install all that i need via get apt apparently your fork work but when i run some stuff nothing to be interest appear
rtl_test work in loop: stop only with ctrl+c
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6 
[R82XX] PLL not locked!
Sampling at 2048000 S/s.

Info: This tool will continuously read from the device, and report if
samples get lost. If you observe no further output, everything is fine.

Reading samples in async mode...
Allocating 15 zero-copy buffers
lost at least 28 bytes
-------------------------------------------------------------------
run grgsm_livemon after this output open graph but output show 000 in loop
qt5ct: using qt5ct plugin
_IceTransSocketUNIXConnect: Cannot connect to non-local host localhost
_IceTransSocketUNIXConnect: Cannot connect to non-local host localhost
Qt: Session management error: Could not open network socket
[INFO] [UHD] linux; GNU C++ version 9.2.1 20200104; Boost_106700; UHD_3.15.0.0-2

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (hw:1,0), Dispositivo o risorsa occupata.

[ERROR] avahi_service_browser_new() failed: Stato errato
Found Rafael Micro R820T tuner
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000,052982 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers

** (python3:12557): WARNING **: 20:08:10.408: Invalid borders specified for theme pixmap:
        /usr/share/themes/Kali-Dark/gtk-2.0/assets/trough-scrollbar-horiz.png,
borders don't fit within the image

** (python3:12557): WARNING **: 20:08:10.409: invalid source position for vertical gradient

** (python3:12557): WARNING **: 20:08:10.409: invalid source position for vertical gradient

** (python3:12557): WARNING **: 20:08:10.418: invalid source position for vertical gradient

** (python3:12557): WARNING **: 20:08:10.422: invalid source position for vertical gradient

00000000 loop
-------------------------------------------------------------------------------------
grgsm_scanner -s 0.8e6 and grgsm_scanner -s 1.2e6 aborted in few minutes without showing nothing,grgsm_scanner only show aborted in few minutes if run with sudo
root@localhost:~/gr-gsm-maint-3.8# sudo grgsm_scanner
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_scanner", line 33, in <module>
    import grgsm
ModuleNotFoundError: No module named 'grgsm'
if run
grgsm_scanner -b GSM850 --debug

Args= 
[INFO] [UHD] linux; GNU C++ version 9.2.1 20200104; Boost_106700; UHD_3.15.0.0-2
[ERROR] avahi_service_browser_new() failed: Bad state

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (hw:1,0), Device or resource busy.

Found Rafael Micro R820T tuner
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOArgs= 
Assertion 'close_nointr(fd) != -EBADF' failed at src/basic/fd-util.c:71, function safe_close(). Aborting.
-------------------------------------------------
if run same command but with --args=rtl have a loop without aborting and error
 if shutdown my pc and restart im unable to run command if dont first start command export PYTHONPATH=/usr/local/lib/python3/dist-packages/:$PYTHONPATH
im not sure but i think install Pybombs without use it and grgsm 3.8 without remove
any help  appreciated thank for yours patience

velichkov, 2020-03-29:
Hi @vespertillo,

Please format your next messages using [markdown](https://guides.github.com/features/mastering-markdown/) as right now your previous message is hardly readable. 

> grgsm_scanner -s 0.8e6 and grgsm_scanner -s 1.2e6 aborted in few minutes without showing nothing,grgsm_scanner only show aborted in few minutes if run with sudo

Do you have a suitable antenna attached to your device? Are you sure that there are active 2G cells in the GSM900 band in the area where you're testing? In case you are inside in a big building try moving near a window or outside.

And also you don't have to run the gr-gsm's commands with `sudo`.

> if run same command but with --args=rtl have a loop without aborting and error

That's good, always run with `--args=rtl`.

> if shutdown my pc and restart im unable to run command if dont first start command export PYTHONPATH=/usr/local/lib/python3/dist-packages/:$PYTHONPATH

That's normal. You can make this change permanent by adding the export command to `~/.profile`.

vespertillo, 2020-03-29:
no antenna attached my phone stay around 50 cm distance,yes my cell when turn on 2G work my country always work in 2G/3G/4G and not 100% in 5G at moment.
run grgsm_livemon work
rtl_test work somes word and loop 0
grgsm_scanner after run show empty line and aborted
grgsm_scanner -b GSM850 --debug have error
i use your fork because i am unable to compile official 3.8 got error,install pybombs but now see hoe to remve and finally remove and delete folder 3.8,on my pc there is only your fork and all dependecie and not work in vm

velichkov, 2020-03-29:
> no antenna attached 

You do need an antenna.

> my phone stay around 50 cm distance

Both `grgsm_scanner` and `grgsm_livemon` capture and decode the downlink only so it doesn't really matter how close your phone is. If you want to capture the uplink then you need two RTL dongles and a small hardware modification, you can read more about this [here](https://ptrkrysik.github.io/).

> yes my cell when turn on 2G work

Do you have your own BTS?

>  my country always work in 2G/3G/4G and not 100% in 5G at moment.

Your phone has to be configured in [2G only mode](https://www.google.com/search?q=android+2g+only+mode).

> run grgsm_livemon work
> rtl_test work work and loop 0

What do you mean by "loop 0"? What most probably you are seeing is capital letter O (not 0) and this means **O**verflow/**O**verrun. 

> grgsm_scanner after run show empty line and aborted
> grgsm_scanner -b GSM850 --debug have error

You do need an antenna.

> i use your fork because i am unable to compile official 3.8 got error,install pybombs but now see how to remove and finally remove and delete folder 3.8,on my pc there is only your fork and all dependecie and not work in vm

OK.

vespertillo, 2020-03-29:
Do you have your own BTS?  what meaning?????
turn my phone only 2G,sorry is O not 0zero this means Overflow/Overrun. i reads many issue before open issue,without antenna every command dont work? i see now my version
gr-gsm         0.42.2-1+b2  amd64

vespertillo, 2020-03-29:
if i try with other machine more powerful intel i5 6GB ram? overflow/overrun caused on low hardware or need first antenna?

velichkov, 2020-03-30:
>  i see now my version
> gr-gsm 0.42.2-1+b2 amd64

How did you got this?

> if i try with other machine more powerful intel i5 6GB ram? 

If you have more powerful machine then do try on it as well.

> overflow/overrun caused on low hardware i read 

Yes, that's usually correct. Also test on different USB ports and in case you are using a USB hub also try without it.

> or need first antenna?

In all cases you do need an antenna.

vespertillo, 2020-03-31:
i got this with dpkg -l gr-gsm,i am not try at moment the big machine because use for mining at moment : )

velichkov, 2020-03-31:
> i got this with dpkg -l gr-gsm,

So you have two versions installed on your system - one installed with apt-get and a second one build from source from my fork. The `gr-gsm` package [has been removed](https://pkg.kali.org/pkg/gr-gsm) from the Kali and Debian testing repositories last October so my recommendation is to remove the package as having multiple versions installed at the same time can cause a lot of hard to debug problems.
```
sudo apt-get remove gr-gsm
```

vespertillo, 2021-04-06:
I try with antenna work,i forget to update post

closed 2021-04-06 by vespertillo