ptrkrysik/gr-gsm #515: grgsm wants an old version of the library, which I don't have, but I have a new one

antoniprzybylik, 2020-03-30:
Yesterday I tried to run grgsm_livemon. I got this error:

`
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/grgsm/__init__.py", line 48, in <module>
from .grgsm_swig import *
  File "/usr/lib/python3.8/site-packages/grgsm/grgsm_swig.py", line 13, in <module>
    from . import _grgsm_swig
ImportError: libboost_program_options.so.1.71.0: cannot open shared object file: No such file or directory
Traceback (most recent call last):
  File "/usr/bin/grgsm_livemon", line 37, in <module>
from grgsm import arfcn
  File "/usr/lib/python3.8/site-packages/grgsm/__init__.py", line 48, in <module>
from .grgsm_swig import *
  File "/usr/lib/python3.8/site-packages/grgsm/grgsm_swig.py", line 13, in <module>
from . import _grgsm_swig
ImportError: libboost_program_options.so.1.71.0: cannot open shared object file: No such file or directory
`

I installed grgsm again and got this error:

`
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/grgsm/__init__.py", line 48, in <module>
    from .grgsm_swig import *
  File "/usr/lib/python3.8/site-packages/grgsm/grgsm_swig.py", line 13, in <module>
    from . import _grgsm_swig
ImportError: libvolk.so.2.1: cannot open shared object file: No such file or directory
Traceback (most recent call last):
  File "/usr/bin/grgsm_livemon", line 37, in <module>
    from grgsm import arfcn
  File "/usr/lib/python3.8/site-packages/grgsm/__init__.py", line 48, in <module>
    from .grgsm_swig import *
  File "/usr/lib/python3.8/site-packages/grgsm/grgsm_swig.py", line 13, in <module>
    from . import _grgsm_swig
ImportError: libvolk.so.2.1: cannot open shared object file: No such file or directory
`

I have both of these libraries in the newer version. I have libvolk.so.2.2.1 and libboost_program_options.so.1.72.0 . I don't know why you are importing this version. Can't you just import libboost_program_options.so without the version suffix? What can I do to make it work?

system: Arch Linux
gnuradio: 3.8
I used maint-3.8: git clone -b maint-3.8 https://github.com/velichkov/gr-gsm.git

velichkov, 2020-03-30:
Hi @antoniprzybylik,

>  I have libvolk.so.2.2.1 and libboost_program_options.so.1.72.0 . 

How did you install VOLK and do you have multiple versions installed at the same time?

> I don't know why you are importing this version. 

These libraries are not imported but are linked in `libgrgsm.so` as they are used in the gnuradio libraries.

> Can't you just import libboost_program_options.so without the version suffix? 

No, that's not possible.

> What can I do to make it work?

How did you install gr-osmosdr and gr-iqbal and did you updated them recently? Every time you update boost, volk or gnuradio you need to rebuild and reinstall those projects.

If you have not updated them recently then remove the theirs build directories and then rebuild and reinstall them.

> system: Arch Linux

I'm not an Arch user and I'm not familiar with it.

> I used maint-3.8: git clone -b maint-3.8 https://github.com/velichkov/gr-gsm.git

That's the correct branch for gnuradio 3.8, you can also use [porting_to_gr38](https://github.com/ptrkrysik/gr-gsm/tree/porting_to_gr38) branch from the Piotr's repo.

velichkov, 2020-03-30:
Hi @antoniprzybylik,

You can use triple back ticks ``` to enclose code or output and not a single one, see also https://guides.github.com/features/mastering-markdown/

> > Every time you update boost, volk or gnuradio you need to rebuild and reinstall those projects.
> 
> It's quite a bit hard...

Yes, but it's needed especially when you update to a new major version (1.71 -> 1.72) that is ABI incompatible.

> There are such files in my /lib directory :
> `./libboost_program_options.so.1`
> `./libboost_program_options.so`
> `./libboost_program_options.so.1.72`
> `./libboost_program_options.so.1.72.0`
> So I meant you could link ./libboost_program_options.so instead of ./libboost_program_options.so.1.71.0. Couldn't you?

No, that's not how linking with dynamic libraries (`.so`) files works in Linux.

> Anyway, I tried to reinstall gr-iqbal.
> I don't remember how I installed it, but now I get this error:

The `gr-iqbal` is an optional dependency and probably you did not have it previously.

> `-- Checking for module SWIG`
> `-- Found SWIG version 4.0.1.`
> `-- Found SWIG: /usr/bin/swig `
> `-- Found PythonLibs: /usr/lib/libpython3.8.so (found suitable version "3.8.2", minimum required is "3") `
> `-- Configuring done`
> `CMake Error at lib/CMakeLists.txt:30 (add_library):`
> `Cannot find source file:`
> 
> `~/Pobrane/gr-iqbal/libosmo-dsp/src/cxvec.c`

You first need to install [libosmo-dsp](https://git.osmocom.org/libosmo-dsp/), `gr-iqbal` has a really old version of it as a submodule (`git submodule update --init`) so my advice is to clone and install the latest version of it (not as a submdule).

antoniprzybylik, 2020-03-30:
I rebuilt and reinstalled them in this order:
libosmo-dsp
gr-iqbal
gr-osmosdr
gr-gsm
However, this error still occurs. ( this first with libvolk )

velichkov, 2020-03-30:
> I rebuilt and reinstalled them in this order:
> libosmo-dsp
> gr-iqbal
> gr-osmosdr
> gr-gsm

That's the correct order.

> However, this error still occurs. ( this first with libvolk )

Check how many `libgrgsm.so.v0.*` files there are in `/usr/lib/` and if there are more then one then delete them all and reinstall `gr-gsm`. If this does not solve the problem provide (attach as file) the full output of the following commands
```
ldd -r -v /usr/lib/python3.8/site-packages/grgsm/_grgsm_swig.so
objdump -p /usr/lib/python3.8/site-packages/grgsm/_grgsm_swig.so
ldd -r -v /usr/lib/libgrgsm.so
objdump -p /usr/lib/libgrgsm.so
ls -alt /usr/lib/libgrgsm.so*
```

antoniprzybylik, 2020-03-30:
In /usr/lib I had only one libgrgsm.so.v0.* file but also I had two similar files so I managed to delete them anyway and reinstall gr-gsm.
./libgrgsm.so
./libgrgsm.so.1.0.0git
./libgrgsm.so.v0.41.1-346-g2efaa49e
and as a result now I don't have any files with libgrgsm prefix in this directory and grgsm_livemon execution attempt end with such an error:
```
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/grgsm/__init__.py", line 48, in <module>
    from .grgsm_swig import *
  File "/usr/lib/python3.8/site-packages/grgsm/grgsm_swig.py", line 13, in <module>
    from . import _grgsm_swig
ImportError: libgrgsm.so.1.0.0git: cannot open shared object file: No such file or directory
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_livemon", line 37, in <module>
    from grgsm import arfcn
  File "/usr/lib/python3.8/site-packages/grgsm/__init__.py", line 48, in <module>
    from .grgsm_swig import *
  File "/usr/lib/python3.8/site-packages/grgsm/grgsm_swig.py", line 13, in <module>
    from . import _grgsm_swig
ImportError: libgrgsm.so.1.0.0git: cannot open shared object file: No such file or directory
```

antoniprzybylik, 2020-03-30:
And you still wanted the results of these commands. Here they are:

ldd -r -v /usr/lib/python3.8/site-packages/grgsm/_grgsm_swig.so :
https://pastebin.com/XHVcUiqM

objdump -p /usr/lib/python3.8/site-packages/grgsm/_grgsm_swig.so:
https://pastebin.com/8c1H6Avh

Of course, I have to skip these files because I deleted them ...
ldd -r -v /usr/lib/libgrgsm.so
objdump -p /usr/lib/libgrgsm.so
ls -alt /usr/lib/libgrgsm.so*

velichkov, 2020-03-30:
What prefix are you installing in? When running cmake do you specify `CMAKE_INSTALL_PREFIX`? At some point have you attempted installation in `/usr/` prefix instead of the default `/usr/local/`? 

Now you most probably have multiple installations in different directories so my advice is to first clean up and remove everything related to gr-gsm from `/usr` and `/usr/local/`

```
cd build
sudo make uninstall
rm CMakeCache.txt
cmake -DCMAKE_INSTALL_PREFIX=/usr/  ..
make
sudo make uninstall
```
Then check for gr-gsm related files and manually delete them
```
sudo find /usr -name "*grgsm*"
```
As I said I'm not familiar with Arch distribution but for other distributions (Debina/Ubuntu/Fedora/...) I would not recommend installing from source in `/usr/`. When building from source I usually install in `/usr/local/` (the default) in order to keep installations from source separate from the software installed using the package manager.

So after cleaning up choose a location and if it is not `/usr/local` always specify it when running `cmake`.

antoniprzybylik, 2020-03-30:
ukh... I did it! 
I really thank you for your help. You helped me a lot. I learned to use tools such as cmake, autotools and github better. And to put manually installed programs in /usr/local. However, the grgsm installation is extremely complicated and must be repeated every time you update the libraries it use. I'm surprised that there is no script that updates it all. I will probably write it for myself, and then I'll be able to share it to the others and they do not have to bother like me, and you won't have to help them.

velichkov, 2020-03-30:
> ukh... I did it!

Cool! Can you post a short summary of what you did to resolve the problem here (and in your stackoverflow question as well) and close this issue.

>  However, the grgsm installation is extremely complicated

I disagree. Once you get yourself familiar with the tools and get some more experience building software from source you'll see that the installation is really not that difficult.

> I'm surprised that there is no script that updates it all. I will probably write it for myself, and then I'll be able to share it to the others and they do not have to bother like me, and you won't have to help them.

The usual problem with such scripts is that sooner or later the author loose interest in testing and updating them so they become outdated. Also such scripts tends to include distribution specific things and this limits their usability.

In my opinion it's better to try helping the package maintainer update the package for your distribution of choice or create a package and maintain it if such package does not exists yet. See https://aur.archlinux.org/packages/gr-gsm-git/

antoniprzybylik, 2020-03-30:
To resolve this problem you have to:
**1. Remove everything related to gr-gsm from /usr and /usr/local**
You can enter the cloned repository and do:
```
cd build
sudo make uninstall
rm CMakeCache.txt
cmake -DCMAKE_INSTALL_PREFIX=/usr/ ..
make
sudo make uninstall
```
Then you should check for gr-gsm related files using the command below and manually delete them:
`sudo find /usr -name "*grgsm*"`

**2. Once you have deleted all previous gr-gsm related files you need to reinstall it**
Install it and its dependencies in this order:
```
libosmo-dsp
gr-iqbal
gr-osmosdr
gr-gsm
```
**3. If you decided to install grgsm in /usr/local/lib then you may have a problem like me because / usr/local/lib is not the standard location for ld libraries in some distributions**
[This question on stackoverflow](https://stackoverflow.com/questions/4743233/is-usr-local-lib-searched-for-shared-libraries) shows how to add /usr/local/lib to ld libraries locations

closed 2020-03-30 by antoniprzybylik