ptrkrysik/gr-gsm #516: When I Try To Start The Livemon I get :- ModuleNotFoundError: No module named 'grgsm'

Mr-Momen, 2020-04-02:
hi there
see sir i really wanna try this project and i bought the sdr rtl nd i really wanna test it but i face a problem first of all you should know that i install gr-gsm successfully and i get this output :-
==============================
root@kali:~# pybombs install gr-gsm
[INFO] Prefix Python version is: 3.7.7
[INFO] PyBOMBS Version 2.3.4a0
[INFO] Phase 1: Creating install tree and installing binary packages:
[INFO] No packages to install.
root@kali:~#
=============================
but unfortunately when i try to start the live_mon i get this error :-
====================================
root@kali:~# grgsm_livemon
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_livemon", line 37, in <module>
    from grgsm import arfcn
ModuleNotFoundError: No module named 'grgsm'
==========================================
so please help me

velichkov, 2020-04-02:
Hi @Mr-Momen,

> but unfortunately when i try to start the live_mon i get this error :-
> # root@kali:~# grgsm_livemon
> Traceback (most recent call last):
> File "/usr/local/bin/grgsm_livemon", line 37, in
> from grgsm import arfcn
> ModuleNotFoundError: No module named 'grgsm'

See https://github.com/ptrkrysik/gr-gsm/issues/480#issuecomment-532305188

And please post such questions to our [mailing list](https://github.com/ptrkrysik/gr-gsm/#mailing-list)

ptrkrysik, 2026-11-09:
Duplicate of #480. Python path problem of the old install; see #480 for the solution. The new release installs under the GNU Radio python path.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik