ptrkrysik/gr-gsm #517: grgsm_livemon errore

greencorsairs, 2020-04-03:
root@kali:~/Desktop# grgsm_livemon 
[INFO] [UHD] linux; GNU C++ version 9.2.1 20200104; Boost_106700; UHD_3.15.0.0-2
[ERROR] avahi_service_browser_new() failed: Bad state
Found Rafael Micro R820T tuner
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
** (python3:22596): WARNING **: 01:42:30.147: Invalid borders specified for theme pixmap:
        /usr/share/themes/Kali-Dark/gtk-2.0/assets/trough-scrollbar-horiz.png,
borders don't fit within the image
** (python3:22596): WARNING **: 01:42:30.156: invalid source position for vertical gradient
** (python3:22596): WARNING **: 01:42:30.156: invalid source position for vertical gradient
** (python3:22596): WARNING **: 01:42:30.161: invalid source position for vertical gradient
** (python3:22596): WARNING **: 01:42:30.162: invalid source position for vertical gradient

ChillVibesMushroom, 2022-05-12:
Did you ever figure it out

ptrkrysik, 2026-11-09:
The log contains only harmless warnings (use --args=rtl) and no actual failure; please open a new issue with details if the problem persists with the new release.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik