ptrkrysik/gr-gsm #518: grgsm_livemon Error Ubuntu 18.04.4 LTS

kjellxvx, 2020-04-03:
I'm struggling since 3 days now to get this work... 
I'm using a fresh install of Ubuntu 18.04.4 LTS (Jetpack Version) on a NVIDIA Jetson Nano.

No matter what I do, always errors. First I tried to do the manual installation of gr-gsm which worked. But running grgsm_livemon didn't. Then tried sudo apt-get install gr-gsm and copied the grgsm_livemon file here /usr/local/bin/. Still an error: I would really appreciate some help! 

Traceback (most recent call last):
  File "/usr/bin/grgsm_livemon", line 45, in <module>
    from gnuradio import qtgui
  File "/usr/lib/python2.7/dist-packages/gnuradio/qtgui/__init__.py", line 37, in <module>
    from range import Range, RangeWidget
  File "/usr/lib/python2.7/dist-packages/gnuradio/qtgui/range.py", line 67, in <module>
    class RangeWidget(QtWidgets.QWidget):
AttributeError: 'module' object has no attribute 'QWidget'

theoden8, 2020-04-03:
Ubuntu 18.04 here. I am struggling to get it running as well, but I got around that particular issue by downloading https://github.com/ptrkrysik/gr-gsm/blob/master/apps/grgsm_livemon.grc and installing with

    $ grcc grgsm_livemon.grc
    $ mv grgsm_livemon.py grgsm_livemon
    # move it to /usr/bin

kjellxvx, 2020-04-03:
How do you mean that? When I try to compile it the way you said I get A LOT of these errors:
grgsm_livemon.grc:172:267:FATAL:PARSER:ERR_UNDECLARED_ENTITY: Entity 'rarr' not defined

theoden8, 2020-04-04:
That's not a python error though, so I assume it's trying to start at least?

theoden8, 2020-04-04:
Sorry, my bad. The command is `grcc -d . grgsm_livemon.grc`, like it says [here](https://osmocom.org/projects/gr-gsm/wiki/Installation).

kjellxvx, 2020-04-04:
Still a lot of errors like the one I described in my previous post. But I did this before and it worked.

velichkov, 2020-04-04:
Hi @kjellxvx,

>  But running grgsm_livemon didn't. Then tried sudo apt-get install gr-gsm and copied the grgsm_livemon file here /usr/local/bin/.

You should not do this. You can't mix scripts from different versions and expect them to work.

Right now you most probably have two different versions installed at the same time and this can leads to various hard to debug problems.

My advice is to either remove the version you've installed from source (go in the build directory and execute `sudo make uninstall`) and do what @theoden8 have described or remove the apt version (`sudo apt-get remove gr-gsm`) and provide the exact error you are getting when running `grgsm_livemon`

kjellxvx, 2020-04-04:
Thanks for the help so far. I'm going to do a fresh install now because things seem to be messed up. Can you tell which which way I should try it this time? Manuel mode or sudo apt-get install? As I said, I'm using Ubuntu 18.04.4 LTS. A few days ago I tried both installation types (not at the same time ;)) and had errors on both.

kjellxvx, 2020-04-07:
I tried to install it the apt-get way on a fresh system and it ends with the error: 

Processing triggers for shared-mime-info (1.9-2) ...
/usr/share/mime/packages/gnuradio.xml:1: parser error : Start tag expected, '<' not found
/usr/share/gnuradio/grc/freedesktop/gnuradio-grc.xml
^
Failed to parse '/usr/share/mime/packages/gnuradio.xml'
Processing triggers for gnome-menus (3.13.3-11ubuntu1.1) ...
Processing triggers for hicolor-icon-theme (0.17-2) ...

velichkov, 2020-04-07:
> /usr/share/mime/packages/gnuradio.xml:1: parser error : Start tag expected, '<' not found
> /usr/share/gnuradio/grc/freedesktop/gnuradio-grc.xml
> ^
> Failed to parse '/usr/share/mime/packages/gnuradio.xml'

This is a [known bug](https://bugs.launchpad.net/ubuntu/+source/gnuradio/+bug/1769432) in the gnuradio package. You can either ignore it or replace `gnuradio.xml` with a symlink to `gnuradio-grc.xml`

```
sudo rm /usr/share/mime/packages/gnuradio.xml
sudo ln -s /usr/share/gnuradio/grc/freedesktop/gnuradio-grc.xml /usr/share/mime/packages/gnuradio.xml
sudo update-mime-database -V /usr/share/mime/
```

kjellxvx, 2020-04-07:
Alright great! Thanks! When running grgsm_livemon I get the following error:

Traceback (most recent call last):
  File "/usr/bin/grgsm_livemon", line 41, in <module>
    from PyQt4 import Qt
ImportError: No module named PyQt4

Shouldn't the apt-get method install all modules in order to make it run?

velichkov, 2020-04-07:
> Alright great! Thanks! When running grgsm_livemon I get the following error:
> 
> Traceback (most recent call last):
> File "/usr/bin/grgsm_livemon", line 41, in
> from PyQt4 import Qt
> ImportError: No module named PyQt4
> 
> Shouldn't the apt-get method install all modules in order to make it run?

It does install all required modules but the `gr-gsm` package in Ubuntu 18.04 is a bit broken and that's why you  need to download `grgsm_livemon.grc` and rebuild it manually as described by @theoden8 and in our [wiki](https://osmocom.org/projects/gr-gsm/wiki/Installation#Installation-from-packages-on-Debian-Testing-and-Ubuntu-1804)

```
wget http://git.osmocom.org/gr-gsm/plain/apps/grgsm_livemon.grc
grcc -d . grgsm_livemon.grc
sudo cp grgsm_livemon.py /usr/bin/grgsm_livemon
```

kjellxvx, 2020-04-07:
The second command gives me the error: 

Gtk-Message: 15:45:31.643: Failed to load module "canberra-gtk-module"
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

velichkov, 2020-04-07:
> Gtk-Message: 15:45:31.643: Failed to load module "canberra-gtk-module"

You can either ignore this or install the canberra gtk module.
```
sudo apt install libcanberra-gtk-module libcanberra-gtk3-module
```

kjellxvx, 2020-04-07:
Tried running grgsm_livemoon again: 

linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.11
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
[ERROR] SoapySSDPEndpoint::sendTo(udp://[ff02::c]:1900) = -1
  sendto(udp://[ff02::c]:1900) [99: Cannot assign requested address]
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
[INFO] Using format CF32.
Segmentation fault (core dumped)

velichkov, 2020-04-07:
> Segmentation fault (core dumped)

What SDR device do you have and is it connected to your computer? 

To suppress most of the above messages you can specify the device using the `--args=` option, for example for RTL dongles you can run `grgsm_livemon --args=rtl`.

kjellxvx, 2020-04-07:
That seems to work right now. At least the GUI does start and show something. So it seems to work, right? 
However it says that it can't find a supported tuner, what I probably have to fix with the rtl-sdr tools. 

linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.11
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
[ERROR] SoapySSDPEndpoint::sendTo(udp://[ff02::c]:1900) = -1
  sendto(udp://[ff02::c]:1900) [99: Cannot assign requested address]
No supported tuner found
Enabled direct sampling mode, input 1
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
No supported tuner found
Enabled direct sampling mode, input 1
Exact sample rate is: 2000000,052982 Hz
/home/name/.gnuradio/prefs/vmcircbuf_default_factory: No such file or directory
vmcircbuf_createfilemapping: createfilemapping is not available
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO

Thank you for all the feedback and the fast replies!

velichkov, 2020-04-07:
> Cannot connect to server socket err = No such file or directory
> Cannot connect to server request channel
> jack server is not running or cannot be started
> JackShmReadWritePtr::~JackShmReadWritePtr - [...]

To suppress the above always run with `--args=rtl`

> No supported tuner found

I don't now about this one.

> Enabled direct sampling mode, input 1

Not quite sure what direct sampling mode is. Try finding a way to disable this.

> /home/name/.gnuradio/prefs/vmcircbuf_default_factory: No such file or directory
> vmcircbuf_createfilemapping: createfilemapping is not available

You are going to see this only on the first start so ignore the above two lines.

> OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO

The multiple `O` characters indicate buffer Overrun/Overflow and this means your computer can't read the incoming data fast enough. Try running `rtl_test`,  try a different USB port, a small sample rage (`-s 1e6`) and stop other processes that consume a lot of CPU. In case you are using USB hub try without it.

Also note that when running livemon you need to specify frequency of a known GSM C0 cell (`-f 941.8M`), you can find such frequencies using the `grgsm_scanner` script.

kjellxvx, 2020-04-07:
The hint about Overrun/Overflow was really useful! Thank you! It looks like the jetson nano is kind of to weak for this process. All 4 CPUs immediately use 100%. 
Running rtl_test -s 1e6 keeps the GPUs at around 5-15% and doesn't show any output. Which is probably that what should happen. 

Running grgsm_scanner makes the CPUs use 100% (probably overrun again) and shows after a few minutes either nothing or 0 stations found.

velichkov, 2020-04-07:
Also to get some more output from `grgsm_scanner` you need to open `/usr/bin/grgsm_scanner` and comment out lines 320 and 321
```
            os.dup2(null_fds[0], 1)
            os.dup2(null_fds[1], 2)
```
The newer versions have a `--debug` flag that is not available in version `0.41.2-1`

My other advice is to first install `gr-gsm` on a normal PC and make yourself familiar with the various tools - how to scan, capture and decode GSM messages and calls. After that you can run all tools on the Jetson board and compare the results with what you get on your PC.

theoden8, 2020-04-11:
I fixed my hackrf by updating firmware. Still, after running grgsm_scanner I have to press "Reset" for the board to be usable again.

marwon1, 2021-01-31:
l have a problem with this 

sudo grgsm_livemon                                              
Warning: failed to XInitThreads()
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
[INFO] [UHD] linux; GNU C++ version 10.2.1 20201207; Boost_107400; UHD_3.15.0.0-4+b1
[ERROR] avahi_service_browser_new() failed: Bad state
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_livemon", line 385, in <module>
    main()
  File "/usr/local/bin/grgsm_livemon", line 361, in main
    tb = top_block_cls(args=options.args, collector=options.collector, collectorport=options.collectorport, fc=options.fc, gain=options.gain, osr=options.osr, ppm=options.ppm, samp_rate=options.samp_rate, serverport=options.serverport, shiftoff=options.shiftoff)
  File "/usr/local/bin/grgsm_livemon", line 113, in __init__
    args="numchan=" + str(1) + " " + str(grgsm.device.get_default_args(args))
  File "/usr/lib/python3/dist-packages/grgsm/device.py", line 55, in get_default_args
    raise RuntimeError("Unable to find any supported SDR devices")
RuntimeError: Unable to find any supported SDR devices

darkhacknet, 2022-05-21:
> grgsm_livemon --args=rtl

Hey velichkov hope you are well, could you please help me with grgsm_livemon is always sending the command not found error ! Im searching what is happening with no success !

SDR888, 2022-06-13:
> It does install all required modules but the `gr-gsm` package in Ubuntu 18.04 is a bit broken and that's why you need to download `grgsm_livemon.grc` and rebuild it manually as described by [...]

 I did find where is grgsm_livemon.py   ,i search everywhere but can't find   grgsm_livemon.py .  help please!!! thank you so much!

ptrkrysik, 2026-11-09:
Resolved in thread (rebuild livemon via grcc, use --args=rtl; overruns 'O' mean the CPU cannot keep up). The new GNU Radio 3.10 release replaces this setup.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik