ptrkrysik/gr-gsm #519: I have tried all ways possible i couldn't solve it , help please

samnq, 2020-04-11:
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_capture.py", line 257, in <module>
    if grgsm.arfcn.is_valid_downlink(options.fc, band):
TypeError: is_valid_downlink() takes exactly 1 argument (2 given)

velichkov, 2020-04-15:
Hi @samnq,

> File "/usr/local/bin/grgsm_capture.py", line 257, in
> if grgsm.arfcn.is_valid_downlink(options.fc, band):
> TypeError: is_valid_downlink() takes exactly 1 argument (2 given)

You most probably have two versions installed at the same time - one using `apt-get` in `/usr/` and a second one installed from source in `/usr/local/`. 

Having multiple versions can cause various hard to debug problems so my advice is to remove the version installed using `apt-get` - execute `sudo apt-get remove gr-gsm`

samnq, 2020-04-15:
> Hi @samnq,
>
>
> You most probably have two versions installed at the same time - one using `apt-get` in `/usr/` and a second one installed from source in `/usr/local/`.
>
> Having multiple versions can [...]

sudo apt-get remove gr-gsm
[sudo] password for user: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package gr-gsm

velichkov, 2020-04-15:
How did you install gr-gsm and which version?

samnq, 2020-04-15:
git clone 
from here 
https://github.com/ptrkrysik/gr-gsm

velichkov, 2020-04-15:
When did you clone it? Go in the gr-gsm's directory and give me the full output of the following commands.
```
git describe
git status
grep is_valid_downlink apps/helpers/grgsm_capture
```

samnq, 2020-04-17:
Thank you problem solved, i just changed this 

if grgsm.arfcn.is_valid_downlink(options.fc, band):
change it to this
if grgsm.arfcn.is_valid_downlink(band):
line 257

velichkov, 2020-04-17:
> Thank you problem solved, i just changed this
> 
> if grgsm.arfcn.is_valid_downlink(options.fc, band):
> change it to this
> if grgsm.arfcn.is_valid_downlink(band):
> line 257

There is no such line in grgsm_capture in the [latest version](https://github.com/ptrkrysik/gr-gsm/blob/fa184a9447a90aefde2ca0dea1347b702551015d/apps/helpers/grgsm_capture) from the current master branch (fa184a9447a90aefde2ca0dea1347b702551015d) so you seems to have a really old version. My advice is to update and reinstall from the current master branch.

samnq, 2020-04-17:
I am new to this can you help me and give step by step for how to remove old and install new or update please?

velichkov, 2020-04-17:
It should be something like

```
cd gr-gsm/build
sudo make uninstall
git pull
make
sudo make install
```

samnq, 2020-04-17:
user@user-OptiPlex-7010:~/gr-gsm/build$ git pull
Already up-to-date.

user@user-OptiPlex-7010:~/gr-gsm/build$ sudo make install
[ 62%] Built target grgsm
[ 62%] Built target grgsm_swig_swig_doc
[ 64%] Built target _grgsm_swig_swig_tag
[ 68%] Built target grgsm_swig_swig_2d0df
[ 72%] Built target _grgsm_swig
[ 75%] Built target pygen_swig_94726
[ 78%] Built target pygen_python_95ac7
[ 80%] Built target pygen_python_misc_utils_f4465
[ 82%] Built target pygen_python_receiver_44dea
[ 84%] Built target pygen_python_demapping_20f36
[ 86%] Built target pygen_python_transmitter_9f516
[ 89%] Built target pygen_python_trx_5ce45
[ 90%] Generating grgsm_livemon_headless
Warning: Block with key "osmosdr_sink" already exists.
	Ignoring: /usr/share/gnuradio/grc/blocks/osmosdr_sink.xml
Warning: Block with key "osmosdr_source" already exists.
	Ignoring: /usr/share/gnuradio/grc/blocks/osmosdr_source.xml
Warning: Block with key "rtlsdr_source" already exists.
	Ignoring: /usr/share/gnuradio/grc/blocks/rtlsdr_source.xml
/home/user/gr-gsm/apps/grgsm_livemon_headless.grc:1:1:FATAL:PARSER:ERR_DOCUMENT_EMPTY: Start tag expected, '<' not found
Error during file compilation.
apps/CMakeFiles/pygen_apps.dir/build.make:89: recipe for target 'apps/grgsm_livemon_headless' failed
make[2]: *** [apps/grgsm_livemon_headless] Error 1
CMakeFiles/Makefile2:1317: recipe for target 'apps/CMakeFiles/pygen_apps.dir/all' failed
make[1]: *** [apps/CMakeFiles/pygen_apps.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

velichkov, 2020-04-17:
Which Linux distribution do you use and which version? How did you install gnuradio and which version? 

> ```
>/home/user/gr-gsm/apps/grgsm_livemon_headless.grc:1:1:FATAL:PARSER:ERR_DOCUMENT_EMPTY: Start tag expected, '<' not found
>```

Hmm, what is the content `/home/user/gr-gsm/apps/grgsm_livemon_headless.grc` and is it really empty? Give me the full output of
```
git describe
git status
```

samnq, 2020-04-17:
yes everything in empty now :(
how to reinstall grgsm again ?

velichkov, 2020-04-17:
Give me the full output of
```
git describe
git status
```

samnq, 2020-04-17:
user@user-OptiPlex-7010:~/gr-gsm$ git describe
v0.42.2-55-gfa184a9
user@user-OptiPlex-7010:~/gr-gsm$ git status
On branch master
Your branch is up-to-date with 'origin/master'.
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

	modified:   apps/grgsm_livemon_headless.grc

Untracked files:
  (use "git add <file>..." to include in what will be committed)

	apps/grgsm_livemon.py

no changes added to commit (use "git add" and/or "git commit -a")

velichkov, 2020-04-17:
Please next time use [markdown](https://guides.github.com/features/mastering-markdown/) to format your messages and enclose outputs with a triple backticks symobls (```)

> modified:   apps/grgsm_livemon_headless.grc

Why did you modify it?

```
cd gr-gsm
rm apps/grgsm_livemon.py
git reset --hard origin/master
cd build
make
sudo make install 
```

samnq, 2020-04-17:
user@user-OptiPlex-7010:~$ cd gr-gsm
user@user-OptiPlex-7010:~/gr-gsm$ rm apps/grgsm_livemon.py
rm: cannot remove 'apps/grgsm_livemon.py': No such file or directory
user@user-OptiPlex-7010:~/gr-gsm$ git reset --hard origin/master
HEAD is now at fa184a9 Fix uplink sub_slot and sub_types assignment in the sdcch4 and bcch_ccch demappers blocks
user@user-OptiPlex-7010:~/gr-gsm$ cd build
bash: cd: build: No such file or directory
user@user-OptiPlex-7010:~/gr-gsm$ make
make: *** No targets specified and no makefile found.  Stop.
user@user-OptiPlex-7010:~/gr-gsm$ sudo make install
[sudo] password for user: 
make: *** No rule to make target 'install'.  Stop.

velichkov, 2020-04-17:
>```
>user@user-OptiPlex-7010:/gr-gsm$ cd build
>bash: cd: build: No such file or directory
>```

Did you remove the build subdirectory and why? You need to create it again, cd into it and proceed fro m there like you did it before

closed 2020-04-18 by samnq