ptrkrysik/gr-gsm #520: gr-gsm Issue

techybilal, 2020-04-15:
[ 79%] Generating grgsm_livemon
<<< Welcome to GNU Radio Companion Compiler 3.8.1.0 >>>

Block paths:
        /home/kali/Desktop/gr-gsm/grc
        /usr/local/share/gnuradio/grc/blocks

>>> Loading: /home/kali/Desktop/gr-gsm/apps/grgsm_livemon.grc
>>> Load Error: /home/kali/Desktop/gr-gsm/apps/grgsm_livemon.grc: Flowgraph invalid
Compilation error
make[2]: *** [apps/CMakeFiles/pygen_apps.dir/build.make:85: apps/grgsm_livemon] Error 1
make[1]: *** [CMakeFiles/Makefile2:1064: apps/CMakeFiles/pygen_apps.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

velichkov, 2020-04-16:
Hi @techybilal,

Which Linux distribution do you use and which version? How did you install gnuradio? Which gr-gsm version are you trying to compile and from which repository and branch?

You can disable building of `grgsm_livemon` and `grgsm_livemon_headless` by running - `cmake -D ENABLE_GRCC=OFF ..` in the gr-gsm's build directory.

techybilal, 2020-04-17:
Sir Velichkov That Issue is Fixed Now  I  am Using Kali Linux New Version 
I Used this All Commands
rm -rf gr-gsm
git clone -b maint-3.8 https://github.com/velichkov/gr-gsm.git
cd gr-gsm
mkdir build
cd build
cmake ..
make
sudo make install
sudo ldconfig

and when i run grgsm_livemon i Got this error

root@kali:~# grgsm_livemon
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_livemon", line 37, in <module>
    from grgsm import arfcn
ModuleNotFoundError: No module named 'grgsm'

i am beginner so i don't have too much knowledge

velichkov, 2020-04-17:
> ModuleNotFoundError: No module named 'grgsm'

See https://github.com/ptrkrysik/gr-gsm/issues/480#issuecomment-532305188

naythuraintun, 2020-06-06:
Hello, Sir @velichkov  help me to fix this make error

>>> [ 73%] Generating grgsm_livemon

>>> <<< Welcome to GNU Radio Companion Compiler 3.8.1.0 >>>

>>> Block paths:
>>> 	/root/Documents/gsm/gr-gsm latest/grgsm/gr-gsm/grc
>>>	/usr/share/gnuradio/grc/blocks
>>>	/usr/local/share/gnuradio/grc/blocks

>>> Loading: /root/Documents/gsm/gr-gsm latest/grgsm/gr-gsm/apps/grgsm_livemon.grc
>>> Load Error: /root/Documents/gsm/gr-gsm latest/grgsm/gr-gsm/apps/grgsm_livemon.grc: Flowgraph invalid
Compilation error
make[2]: *** [apps/CMakeFiles/pygen_apps.dir/build.make:85: apps/grgsm_livemon] Error 1
make[1]: *** [CMakeFiles/Makefile2:982: apps/CMakeFiles/pygen_apps.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

sumerpepper45, 2022-05-08:
> Hello, Sir @velichkov help me to fix this make error

me to get this make error , how to fix it ?

ptrkrysik, 2026-11-09:
Duplicate of #480. See #480 for the python path solution; the GR 3.8 flowgraph compile problem does not apply to the GNU Radio 3.10 port.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik