ptrkrysik/gr-gsm #521: grgsm_decode problem - help

samnq, 2020-04-17:
Sam@user# grgsm_decode -c sms.cfile -s 1e6 -a 49 -m TCHH -t 0 
Using Volk machine: avx_64_mmx_orc

freezing , package not send to wireshark... 
wireshark not receiving anything
what is the problem ?

velichkov, 2020-04-17:
Which Linux distribution do you use and which version? How did you install gnuradio and which version? What SDR device do you have and what command (including all arguments) did you use to capture the file? Are you sure that there an active GSM C0 Cell on ARFCN 49? 

From #519 I know that you are using an old version of gr-gsm so my advice is to update to the latest one.

samnq, 2020-04-17:
Ubuntu 16.04.6 LTS
install gnuradio this way 
sudo apt-get update
 sudo apt-get install gnuradio
gnuradio version
3.7.9
SDR hackrf
yes i use the capture file first time grgsm decode was deliver all to wireshark , next day i was trying to find the sms again and it didn't work, nothing on wireshark was delivered , i am not sure it is grgsm problem or wireshark . 
thank you so much for your fast replay and support

velichkov, 2020-04-17:
> Ubuntu 16.04.6 LTS
> 3.7.9

These are really old versions, my advice is to update to a more recent versions.

> SDR hackrf

The HackRF has a big DC spike/offset and one solution to this is to capture with an offset, see https://github.com/ptrkrysik/gr-gsm/issues/473#issuecomment-513846774

> yes i use the capture file first time grgsm decode was deliver all to wireshark , next day i was trying to find the sms again and it didn't work, nothing on wireshark was delivered , i am not sure it is grgsm problem or wireshark .

When was this first time - yesterday, last month, last year? Did you use the same sms.cfile or did you do a new capture for your second attempt?

samnq, 2020-04-17:
first sms.cfile i check yesterday , today i captured a new one and tried to but i didn't work

velichkov, 2020-04-17:
Do you keep the sms.cfile from yesterday and what happens if you try again with it?

samnq, 2020-04-17:
not sending to wireshark

samnq, 2020-04-17:
please how to remove gr-gsm completely from mu ubuntu and make fresh install

velichkov, 2020-04-17:
> please how to remove gr-gsm completely from mu ubuntu and make fresh install

As you've installed it from source you need to execute `sudo make uninstall` from the `build` directory but you've already deleted it so you other option is to manually find and delete every gr-gsm file in `/usr/local/`  sub-directories.

Victor6112, 2020-04-18:
I am coming for the first time so I wanted to thank for your project, it is very necessary and cool!
I have the same problem that samnq encountered and exactly the same conditions, I couldn’t decide for a long time, I will understand now.
Question: There is a sylvain / burst_ind branch in OsmocomBB. When starting Motorola we get bursts - files bursts_20200317_0118_874_2017270_61.dat
1. Can we view them through Wereshark as well as .cfile files?
2. If we can, then how? I could not find it on the network. In grgsm_decode -h there is -b switch for burst but I couldn’t start viewing traffic.
[bursts_20200317_0118_874_2017270_61.dat.tar.gz](https://github.com/ptrkrysik/gr-gsm/files/4495722/bursts_20200317_0118_874_2017270_61.dat.tar.gz)
[bursts_20200317_0118_874_2018961_71.dat.tar.gz](https://github.com/ptrkrysik/gr-gsm/files/4495723/bursts_20200317_0118_874_2018961_71.dat.tar.gz)

closed 2020-04-18 by samnq

ChillVibesMushroom, 2022-10-01:
> Ubuntu 16.04.6 LTS install gnuradio this way sudo apt-get update sudo apt-get install gnuradio gnuradio version 3.7.9 SDR hackrf yes i use the capture file first time grgsm decode was deliver all to [...]

I capture with 

Grgsm_capture -a 588 -T 60 --args=hacker mops1.cfile

I decode with 

Grgsm_decode -a 588 -m SDCCH8 --cfile=mops1.cfile -v

But Wireshark will not open a cfile and it's really frustrating I just did a quick run tbh