ptrkrysik/gr-gsm #523: cmake .. kali 2020

alimka1, 2020-04-20:
67%] Generating doxygen xml for grgsm_swig_doc docs
warning: Tag 'PERL_PATH' at line 1654 of file '/root/gr-gsm/build/swig/grgsm_swig_doc_swig_docs/Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'MSCGEN_PATH' at line 1675 of file '/root/gr-gsm/build/swig/grgsm_swig_doc_swig_docs/Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"

chandu-77, 2020-04-21:
Hello bro. Use the manual installation method. Don't use automated installation by pybomb in debian distributions like kali, parrot etc. I have tested these codes it works fine in both kali and parrot os. But you have to install it manually. Pybomb will give you error. 
You can check my website at [Speed Test](https://netspeed.live)

velichkov, 2020-04-21:
Hi @alimka1,

> warning: Tag 'PERL_PATH' at line 1654 of file '/root/gr-gsm/build/swig/grgsm_swig_doc_swig_docs/Doxyfile' has become obsolete.
> To avoid this warning please remove this line from your configuration [...]

These are just two harmless warnings that you can safely ignore. If you can't build `gr-gsm` provide the full output attached as file or upload it somewhere (dropbox, google drive, ...) and provide a link to it.

ptrkrysik, 2026-11-09:
These doxygen warnings are harmless and can be ignored.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik