ptrkrysik/gr-gsm #525: Gnuradio could not be found because dependency Volk could not be found.

sahilc556, 2020-04-26:
root@kali:~/gnuradio/src/gr-gsm/build# cmake ..
-- The CXX compiler identification is GNU 9.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Extracting version information from git describe...
-- Found Boost: /usr/include (found suitable version "1.67.0", minimum required is "1.35") found components: filesystem system thread chrono date_time atomic 
-- 
-- Checking for module SWIG
-- Found SWIG version 4.0.1.
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_INCLUDE_DIR) 
-- Found Boost: /usr/include (found suitable version "1.67.0", minimum required is "1.67.0") found components: date_time program_options filesystem system regex thread unit_test_framework chrono atomic 
CMake Warning at CMakeLists.txt:135 (find_package):
  Found package configuration file:

    /root/gnuradio/lib/cmake/gnuradio/GnuradioConfig.cmake

  but it set Gnuradio_FOUND to FALSE so package "Gnuradio" is considered to
  be NOT FOUND.  Reason given by package:

  Gnuradio could not be found because dependency Volk could not be found.

CMake Error at CMakeLists.txt:150 (message):
  GnuRadio Runtime required to compile gr-gsm

-- Configuring incomplete, errors occurred!
See also "/root/gnuradio/src/gr-gsm/build/CMakeFiles/CMakeOutput.log".
See also "/root/gnuradio/src/gr-gsm/build/CMakeFiles/CMakeError.log".
root@kali:~/gnuradio/src/gr-gsm/build#

velichkov, 2020-04-26:
> CMake Warning at CMakeLists.txt:135 (find_package):
> Found package configuration file:
>
> ```
> /root/gnuradio/lib/cmake/gnuradio/GnuradioConfig.cmake
> ```
>
> but it set Gnuradio_FOUND to FALSE so package [...]

See https://github.com/ptrkrysik/gr-gsm/issues/480#issuecomment-532195846 and https://github.com/ptrkrysik/gr-gsm/issues/475 and please read all messages in the linked issues.

ptrkrysik, 2026-11-09:
This was a dependency problem (Volk not found by a self-built GNU Radio 3.8) on an old setup, answered in #480/#475. Please install GNU Radio 3.10 with its Volk dependency and try the new release; reopen with full logs if it still fails.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik