ptrkrysik/gr-gsm #526: Build dir arleady exists /usr/local/src/gnuradio/build

ilhuss, 2020-04-26:
Hello, on this part of the installation at 85%, this package "gnuradio / build" stops loading. How to deal with this problem?

velichkov, 2020-04-26:
Hi @ilhuss,

> this package "gnuradio / build" stops loading

What do you mean exactly by "stops loading". Compilation takes time, make sure you have enough free memory, wait until it finishes and if you get an error provide the full output.

ilhuss, 2020-05-03:
Thank you, it helped, but I had to wait a very long time)))

ptrkrysik, 2026-11-09:
Resolved by the reporter: the PyBOMBS compilation just takes a long time. Pybombs is no longer the supported install path for the new release.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik