ptrkrysik/gr-gsm #527: Mobile broadband card is not a supported device ?

e2002e, 2020-04-26:
Hello, I have a laptop with a card named:
usb-Foxconn_Dell_Wireless_5630__EVDO-HSPA__Mobile_Broadband_Mini-Card
It allows me to send and receive sms, but is there a way this can be used with gr-gsm ?
The error I get from grgsm_livemon is:
RuntimeError: Unable to find any supported SDR devices
If that comes from my hardware, are there supported mobile broadband cards that would let me do so while not having external hardware ? I was also wondering if I could use gr-gsm on an android table that uses a gsm chipset  (this is the google nexus 7 (2012) 3g).
Regards.

velichkov, 2020-04-26:
Hi @e2002e,

No. You need an SDR device like [RTL SDR](https://www.rtl-sdr.com/about-rtl-sdr/), [Ettus USRP](https://www.ettus.com/products/), [LimeSDR](https://limemicro.com/products/boards/), ...

e2002e, 2020-04-29:
Thank you. I can't wait to play with radio frequency !

closed 2020-04-29 by e2002e