ptrkrysik/gr-gsm #528: No results on grgsm_scanner

BogdanPau, 2020-04-27:
Hi,
I'm using RTL SDR (2838 DVB-T+DAB+FM) on Kali Linux
(Linux kali 5.5.0-kali2-amd64 #1 SMP Debian 5.5.17-1kali1 (2020-04-21) x86_64 GNU/Linux).
So I've installed all the necessary packages and as well gr-gsm.
Now I'm trying to run "grgsm_scanner -b GSM900 -d" to see nearby stations and I get no real results but just "00000" running in loops then aborted with this line:
"Assertion 'close_nointr(fd) != -EBADF' failed at src/basic/fd-util.c:71, function safe_close(). Aborting.
Aborted"

I tried to reduce the sample rate, but couldn't find a valid number, apparently all values I tried that are not 2Mhz are considered an error or invalid but still trying to search.
I tried rtl_test and I don't get any repeating problems of "lost at x bytes".

I will highly appreciate your help how can I continue, I tried online research and got stuck with this situation now.

Screenshot is attached:
https://imgur.com/hXnnrG4.png

velichkov, 2020-04-27:
Hi @BogdanPau,

> I'm using RTL SDR (2838 DVB-T+DAB+FM) on Kali Linux
> (Linux kali 5.5.0-kali2-amd64 #1 SMP Debian 5.5.17-1kali1 (2020-04-21) x86_64 GNU/Linux).
> So I've installed all the necessary packages and as well [...]

It's capital O and not zero and indicates buffer *O*verrun/*O*verflow. In short this means that the program can't read the samples from the SDR fast enough and its buffers get full.

Are you running it in a VM? What CPU do you have and with how many cores? Do you observer Os when you run `grgsm_livemon_headless -s 2e6`?

Try setting CPU frequency governor to `performance` mode.
```
cpupower frequency-set --governor performance  
```

> "Assertion 'close_nointr(fd) != -EBADF' failed at src/basic/fd-util.c:71, function safe_close(). Aborting.
> Aborted"

Hmm, that's interesting, this assert is in the (systemd's code)[https://github.com/systemd/systemd/blob/master/src/basic/fd-util.c#L72]. I have never seen such asserts.

> I tried to reduce the sample rate, but couldn't find a valid number, apparently all values I tried that are not 2Mhz are considered an error or invalid but still trying to search.

It should works with `1.2e6` or `1.6e6`, my dongle does not accepts `0.8e6` but in theory it should works as well.  

> I tried rtl_test and I don't get any repeating problems of "lost at x bytes".

That's a good indication but it should be noted that `rtl_test` just reads and discards all the data without doing any processing while `grgsm_scanner` performs channelization, resampling, frequency and ppm corrections, ... and this takes lot of CPU.

BogdanPau, 2020-04-27:
Hi,
Thank you for your fast response, much appreciated!

> Are you running it in a VM? What CPU do you have and with how many cores? Do you observer Os when you run grgsm_livemon_headless -s 2e6?

Yes, I'm running VirtualBox machine with amd64 (tried 2 cpus, that what my laptop has (core M))
with 2GB RAM, also tried 4GB RAM.
When first I ran "grgsm_livemon_headless -s 2e6" with 2 cpus, I didn't see immediately O's but after a minute or two I did see some O's flowing slowly. When I powered off the VM and increased to 4 logical cpus I tried again to run this command and I see the O's appereard after longer time, like 5 minutes without them and then few slowly again.

I also experience some weird error of avahi: "[ERROR] avahi_service_browser_new() failed: Bad state" between sessions that I'm restarting my VM. This error goes away when I'm reinstalling avahi daemon. Not sure if it has relevance to the case here..

> cpupower frequency-set --governor performance 

I tried to install cpupower with synaptic pkg manager and when trying to run this command I get:

root@kali:/home/kali# cpupower frequency-set --governor performance 
Setting cpu: 0
Error setting new values. Common errors:
- Do you have proper administration rights? (super-user?)
- Is the governor you requested available and modprobed?
- Trying to set an invalid policy?
- Trying to set a specific frequency, but userspace governor is not available,
   for example because of hardware which cannot be set to a specific frequency
   or because the userspace governor isn't loaded?

In general, is it possible to run grgsm on VM?

Thank you very much @velichkov 

Bogdan

velichkov, 2020-04-27:
Hi @BogdanPau,

> > Are you running it in a VM? What CPU do you have and with how many cores? Do you observer Os when you run grgsm_livemon_headless -s 2e6?
> 
> Yes, I'm running VirtualBox machine with amd64 (tried 2 cpus, that what my laptop has (core M))
> with 2GB RAM, also tried 4GB RAM.

OK. What is the exact CPU model? Give me the output of `cat /proc/cpuinfo`.

> When first I ran "grgsm_livemon_headless -s 2e6" with 2 cpus, I didn't see immediately O's but after a minute or two I did see some O's flowing slowly. When I powered off the VM and increased to 4 [...]

OK. Try running it also with `-s 1e6` and check for overflows. If you have a Samsung smartphone you can configure it to 2G mode only (somewhere in the network settings), then enter *#0011# and from there you can get all ARFCNs of the nearby base stations. Then convert one ARFCN to a frequency using [this](https://www.sqimway.com/gsm_arfcn.php) site and start `grgsm_livemon_headless` on that frequency.

> I also experience some weird error of avahi: "[ERROR] avahi_service_browser_new() failed: Bad state" between sessions that I'm restarting my VM. This error goes away when I'm reinstalling avahi daemon. Not sure if it has relevance to the case here..

You can ignore this error or to suppress it you can run with `--args=rtl`.

> > cpupower frequency-set --governor performance
> 
> I tried to install cpupower with synaptic pkg manager and when trying to run this command I get:
> 
> root@kali:/home/kali# cpupower frequency-set --governor performance
> Setting cpu: 0
> Error setting new values. Common errors:

I'm not familiar with VirtualBox and I'm not sure if it's possible to control this from the guest OS. Try setting this directly in your host OS.

> In general, is it possible to run grgsm on VM?

It's definitely possible but you may need more cores or a better CPU.

BogdanPau, 2020-04-27:
@velichkov Thank you so much! I'm very glad you are helping me here so promptly!
Here is the CPU model output of the command you gave me, meanwhile I will try the other things you wrote as well.

root@kali:/home/kali# cat /proc/cpuinfo
processor       : 0
vendor_id       : GenuineIntel
cpu family      : 6
model           : 61
model name      : Intel(R) Core(TM) M-5Y10c CPU @ 0.80GHz
stepping        : 4
cpu MHz         : 997.692
cache size      : 4096 KB
physical id     : 0
siblings        : 4
core id         : 0
cpu cores       : 4
apicid          : 0
initial apicid  : 0
fpu             : yes
fpu_exception   : yes
cpuid level     : 20
wp              : yes
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx rdtscp lm constant_tsc rep_good nopl xtopology nonstop_tsc cpuid tsc_known_freq pni pclmulqdq ssse3 cx16 pcid sse4_1 sse4_2 x2apic movbe popcnt aes xsave avx rdrand hypervisor lahf_lm abm 3dnowprefetch invpcid_single pti fsgsbase avx2 invpcid rdseed flush_l1d
bugs            : cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs itlb_multihit
bogomips        : 1995.38
clflush size    : 64
cache_alignment : 64
address sizes   : 39 bits physical, 48 bits virtual
power management:

processor       : 1
vendor_id       : GenuineIntel
cpu family      : 6
model           : 61
model name      : Intel(R) Core(TM) M-5Y10c CPU @ 0.80GHz
stepping        : 4
cpu MHz         : 997.692
cache size      : 4096 KB                                                                                                      
physical id     : 0                                                                                                            
siblings        : 4                                                                                                            
core id         : 1                                                                                                            
cpu cores       : 4                                                                                                            
apicid          : 1
initial apicid  : 1
fpu             : yes
fpu_exception   : yes
cpuid level     : 20
wp              : yes
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx rdtscp lm constant_tsc rep_good nopl xtopology nonstop_tsc cpuid tsc_known_freq pni pclmulqdq ssse3 cx16 pcid sse4_1 sse4_2 x2apic movbe popcnt aes xsave avx rdrand hypervisor lahf_lm abm 3dnowprefetch invpcid_single pti fsgsbase avx2 invpcid rdseed flush_l1d
bugs            : cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs itlb_multihit
bogomips        : 1995.38
clflush size    : 64
cache_alignment : 64
address sizes   : 39 bits physical, 48 bits virtual
power management:

processor       : 2
vendor_id       : GenuineIntel
cpu family      : 6
model           : 61
model name      : Intel(R) Core(TM) M-5Y10c CPU @ 0.80GHz
stepping        : 4
cpu MHz         : 997.692
cache size      : 4096 KB
physical id     : 0
siblings        : 4
core id         : 2
cpu cores       : 4
apicid          : 2
initial apicid  : 2
fpu             : yes
fpu_exception   : yes
cpuid level     : 20
wp              : yes
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx rdtscp lm constant_tsc rep_good nopl xtopology nonstop_tsc cpuid tsc_known_freq pni pclmulqdq ssse3 cx16 pcid sse4_1 sse4_2 x2apic movbe popcnt aes xsave avx rdrand hypervisor lahf_lm abm 3dnowprefetch invpcid_single pti fsgsbase avx2 invpcid rdseed flush_l1d
bugs            : cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs itlb_multihit
bogomips        : 1995.38
clflush size    : 64
cache_alignment : 64
address sizes   : 39 bits physical, 48 bits virtual
power management:

processor       : 3
vendor_id       : GenuineIntel
cpu family      : 6
model           : 61
model name      : Intel(R) Core(TM) M-5Y10c CPU @ 0.80GHz
stepping        : 4
cpu MHz         : 997.692
cache size      : 4096 KB
physical id     : 0
siblings        : 4
core id         : 3
cpu cores       : 4
apicid          : 3
initial apicid  : 3
fpu             : yes
fpu_exception   : yes
cpuid level     : 20
wp              : yes
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx rdtscp lm constant_tsc rep_good nopl xtopology nonstop_tsc cpuid tsc_known_freq pni pclmulqdq ssse3 cx16 pcid sse4_1 sse4_2 x2apic movbe popcnt aes xsave avx rdrand hypervisor lahf_lm abm 3dnowprefetch invpcid_single pti fsgsbase avx2 invpcid rdseed flush_l1d
bugs            : cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs itlb_multihit
bogomips        : 1995.38
clflush size    : 64
cache_alignment : 64
address sizes   : 39 bits physical, 48 bits virtual
power management:

root@kali:/home/kali#

BogdanPau, 2020-04-27:
Ok, so I tried "grgsm_livemon_headless -s 1e6" I don't get any overflows at all and neither any results. Just the "not locked" output last as usual. I tried also to do the running with the exact frequency with my Samsung as you instructed.. I tried to make/receive a call meanwhile the command is running but no significant output.
Please bear with me :) I'm going to move my Kali image to a new machine with more CPU and stronger to see how this will result.

Meanwhile I tried also on this machine the scanner command with some invalid sample rate but I got a new kind of output, maybe it tells you something? This is the command and the result of it.

root@kali:/home/kali# grgsm_scanner -b GSM900 -s 1e8 -d --args=rtl

Args= rtl
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Invalid sample rate: 100000000 Hz
[R82XX] PLL not locked!
gr_remez: insufficient extremals -- cannot continue
Warning: set ripple to 0.1100 dB. If this is a problem, adjust the attenuation or create your own filter taps.
gr_remez: insufficient extremals -- cannot continue
Warning: set ripple to 0.1200 dB. If this is a problem, adjust the attenuation or create your own filter taps.
gr_remez: insufficient extremals -- cannot continue
Warning: set ripple to 0.1300 dB. If this is a problem, adjust the attenuation or create your own filter taps.
gr_remez: insufficient extremals -- cannot continue
Warning: set ripple to 0.1400 dB. If this is a problem, adjust the attenuation or create your own filter taps.

velichkov, 2020-04-27:
> model name : Intel(R) Core(TM) M-5Y10c CPU @ 0.80GHz

The frequency scaling seems to be enabled as this CPU supports frequencies up to 2GHz. Try to disable it in your host OS. What is the OS on the host?

> flags : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx rdtscp lm constant_tsc rep_good nopl xtopology nonstop_tsc cpuid [...]

It's good that it supports AVX2 instructions. The Rotator block was broken until recently on AVX2 hosts. Which gnuradio and volk versions did you install and how - manually or using `apt-get`?

Run `volk_profile`, it will test the performance of all supported VOLK kernels and as a result will create `~/.volk/volk_config`. Open the file with a text editor, find `volk_32fc_s32fc_x2_rotator_32fc` and change the line to `volk_32fc_s32fc_x2_rotator_32fc a_sse4_1 u_sse4_1` in order to use the SSE4.1 kernel. 

Also try rebuilding `gr-gsm` with `-ftree-vectorize -march=native -mtune=native` in C and CXX flags.

```
cd build
rm CMakeCache.txt
cmake -D CMAKE_BUILD_TYPE=RelWithDebInfo -D CMAKE_C_FLAGS="-ftree-vectorize -march=native -mtune=native" CMAKE_CXX_FLAGS="-ftree-vectorize -march=native -mtune=native" ..
make clean
make
make test
sudo make install
git submodule update --init
./tests/scripts/decode.sh
```
Run `decode.sh` with and without the modifications in `volk_config` to determine if you are affected or not by the Rotator bug.

> Ok, so I tried "grgsm_livemon_headless -s 1e6" I don't get any overflows at all 

That's good!

> and neither any results.

You need to specify the frequency of your base station (`-f 957.2M`) otherwise it will capture on the default frequency (`957.0M`) but probably there is no base station on that frequency where you live.

Also make sure you have a suitable antenna attached to your RTL dongle and in case you are inside of a big building move near a window or go outside if possible.

> Meanwhile I tried also on this machine the scanner command with some invalid sample rate

Try with `-s 1.2e6` and `-s 1.6e6`.

> grgsm_scanner -b GSM900 -s 1e8 -d --args=rtl
>
> gr_remez: insufficient extremals -- cannot continue
> Warning: set ripple to 0.1100 dB. If this is a problem, adjust the attenuation or create your own filter taps.

I see this for the first time and it does not tell me anything.

velichkov, 2020-04-27:
I just fixed the cmake command as one `-D` was missing
```
cmake -D CMAKE_BUILD_TYPE=RelWithDebInfo -D CMAKE_C_FLAGS="-ftree-vectorize -march=native -mtune=native" -D CMAKE_CXX_FLAGS="-ftree-vectorize -march=native -mtune=native" ..
```

BogdanPau, 2020-04-28:
Hi @velichkov !
Thank you very much, once again :) for the kind detailed help!
So I took the extra mile to move the Kali image for another computer, much powerful.
4 physical CPU's running at 2.6Ghz. The CPU's details are attached here.

Then I followed step by step your instructions.
gnuradio version: 3.8.1.0-1
volk version: 2.2.1-2
I installed them using Synaptic package manager and I can see their versions in the Synaptic as well.
If I can run a command you need regarding testing them, please let me know!

Now regarding your instructions:
I ran the volk_profile and then I entered the config file and the only line with `volk_32fc_s32fc_x2_rotator_32fc`  was actually written `volk_32fc_s32fc_x2_rotator_32fc a_avx u_avx`
I did change them to `a_sse4_1 u_sse4_1`
Then I rebuild the gr-gsm as you instructed step by step, all went fluent.
I ran `decode.sh`, output is attached here as well - no Rotator bug was specified in the running. and then I rolled back the config file to the original line of `volk_32fc_s32fc_x2_rotator_32fc a_avx u_avx` and ran again `decode.sh` with what seems to be exact same output.

I believe the antenna is suitable for the wavelength, it is the stock default antenna came with the RTL-SDR, might not be the best, but I think it should work.
Here is an approximate picture of my antenna: https://images-na.ssl-images-amazon.com/images/I/61OG187Gu-L._AC_SL1500_.jpg
I'm also sitting in a full GSM reception in my room on my phone while using 2G/3G/4G, I even placed the antenna near the window for better result :)

by the end I still see no result in `grgsm_scanner`, it seems while running the grgsm_scanner entering some loop. Nor with `grgsm_livemon_headless` while trying to manipulate the frequency, band, gain etc..

I did also some tests with kalibrate `kal -s GSM900 -g 30`, manipulating some PPM, GAIN and suddenly, good news! :) I found few channels indeed, then going to grgsm_livemon and placing the frequency I found in kalibrate was indeed bring some packets in the wireshark!
But the mystery of grgsm_scanner yet didn't solve out. Do you have any idea?

Attached:
[cpuinfo.txt](https://github.com/ptrkrysik/gr-gsm/files/4542880/cpuinfo.txt)
[decodesh_output.txt](https://github.com/ptrkrysik/gr-gsm/files/4542882/decodesh_output.txt)

Thanks!
Bogdan

velichkov, 2020-04-28:
Hi @BogdanPau,

> Thank you very much, once again :) for the kind detailed help!

You are welcome!

> Now regarding your instructions:
> I ran the volk_profile and then I entered the config file and the only line with `volk_32fc_s32fc_x2_rotator_32fc` was actually written [...]

Great! This means the Rotator is fixed in these versions so you don't need to change this line.

> I believe the antenna is suitable for the wavelength, it is the stock default antenna came with the RTL-SDR, might not be the best, but I think it should work.
> Here is an approximate picture of my antenna: https://images-na.ssl-images-amazon.com/images/I/61OG187Gu-L._AC_SL1500_.jpg

Yes, it should work.

> I'm also sitting in a full GSM reception in my room on my phone while using 2G/3G/4G, I even placed the antenna near the window for better result :)

Is your phone configured in 2G only mode and registered in a 2G network?

> by the end I still see no result in `grgsm_scanner`, it seems while running the grgsm_scanner entering some loop. 

The loop is normal as it needs to capture parts of the GSM band. Do you still see O characters in the output? Have you tried different gain settings?

> Nor with `grgsm_livemon_headless` while trying to manipulate the frequency, band, gain etc..
>
> I did also some tests with kalibrate `kal -s GSM900 -g 30`, manipulating some PPM, GAIN and suddenly, [...]

Great!

> But the mystery of grgsm_scanner yet didn't solve out. Do you have any idea?

Nope.

P.S.
In the future it's better to send questions to our [mailing list](https://github.com/ptrkrysik/gr-gsm/#mailing-list)

blshkv, 2020-05-26:
Hello, I'm having exactly the same problem with Pentoo, different cpu, bare metal Pentoo Linux.
"kal" also works for me, but grgsm_scanner doesn't.

Please let me know what can I provide to help you.

my cpu is 4cores (2+2:
```
processor       : 0
vendor_id       : GenuineIntel
cpu family      : 6
model           : 58
model name      : Intel(R) Core(TM) i5-3320M CPU @ 2.60GHz
stepping        : 9
microcode       : 0x21
cpu MHz         : 1197.399
cache size      : 3072 KB
physical id     : 0
siblings        : 4
core id         : 0
cpu cores       : 2
apicid          : 0
initial apicid  : 0
fpu             : yes
fpu_exception   : yes
cpuid level     : 13
wp              : yes
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm cpuid_fault epb pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid fsgsbase smep erms xsaveopt dtherm ida arat pln pts md_clear flush_l1d
bugs            : cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs itlb_multihit
bogomips        : 5188.45
clflush size    : 64
cache_alignment : 64
address sizes   : 36 bits physical, 48 bits virtual
power management:
```

blshkv, 2020-05-26:
an important point to make: I turn in gqrx on frequency found by kal it shows some spikes at the correct frequencies. However, grgsm_livemon doesn't show it. So it looks like there is something wrong with gr-gsm (i'm using hack-rf)

ptrkrysik, 2026-11-09:
This was a performance/overflow problem on a VM with a low-power CPU and an old GNU Radio 3.8/VOLK; the VOLK rotator problem was fixed upstream. Please retry with the new GNU Radio 3.10 release on native hardware and open a new issue with logs if it still fails.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik