ptrkrysik/gr-gsm #529: Generating grgsm_livemon

dangquocdat97, 2020-04-29:
Hello,
. i'm using ubuntu 16.04. i try to install gr-gsm following https://osmocom.org/projects/gr-gsm/wiki/Installation but i got this error: i did: make VERBOSE=1 and got output:
[ 90%] Generating grgsm_livemon
cd /home/datdq/osmo/gr-gsm/build/apps && /usr/bin/cmake -E env PYTHONPATH="/home/datdq/osmo/gr-gsm/python:/home/datdq/osmo/gr-gsm/python/misc_utils:/home/datdq/osmo/gr-gsm/python/demapping:/home/datdq/osmo/gr-gsm/python/receiver:/home/datdq/osmo/gr-gsm/python/transmitter:/home/datdq/osmo/gr-gsm/python/trx:/home/datdq/osmo/gr-gsm/build/swig" GRC_BLOCKS_PATH=/home/datdq/osmo/gr-gsm/grc /usr/local/bin/grcc -d /home/datdq/osmo/gr-gsm/build/apps /home/datdq/osmo/gr-gsm/apps/grgsm_livemon.grc
No such file or directory
apps/CMakeFiles/pygen_apps.dir/build.make:84: recipe for target 'apps/grgsm_livemon' failed
make[2]: *** [apps/grgsm_livemon] Error 1
make[2]: Leaving directory '/home/datdq/osmo/gr-gsm/build'
CMakeFiles/Makefile2:1317: recipe for target 'apps/CMakeFiles/pygen_apps.dir/all' failed
make[1]: *** [apps/CMakeFiles/pygen_apps.dir/all] Error 2
make[1]: Leaving directory '/home/datdq/osmo/gr-gsm/build'
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

I can do the (cmake) but i got this error when i do the (make) i don't now how i could resolve.
Please support me to fix this error.
Thanks and best regards.

velichkov, 2020-04-29:
Hi @dangquocdat97,

> No such file or directory

Hmm, interesting! Provide the full output of the following commands
```
gnuradio-config-info --version
gnuradio-config-info --prefix
which grcc
/usr/local/bin/grcc  --help
ls /usr/local/bin/grcc
ls -al  /home/datdq/osmo/gr-gsm/apps/
git status
git describe
```

ptrkrysik, 2026-11-09:
This is a duplicate of #532 (continued there) and relates to the old grcc-based app generation on GR 3.7/3.8; that mechanism no longer exists in the 3.10 port.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik