ptrkrysik/gr-gsm #532: Generating grgsm_livemon error

dangquocdat97, 2020-05-04:
Dear @velichkov,

Sorry for late reply. I dont know why but i cannot comment in #529 so i creat a new one. After i remove folder build and install gr-gsm again, i got an other error. It still stop when generating grgsm_livemon: 
[ 90%] Generating grgsm_livemon
cd /home/datdq/osmo/gr-gsm/build/apps && /usr/local/lib/python3.5/dist-packages/cmake/data/bin/cmake -E env PYTHONPATH="/home/datdq/osmo/gr-gsm/python:/home/datdq/osmo/gr-gsm/python/misc_utils:/home/datdq/osmo/gr-gsm/python/demapping:/home/datdq/osmo/gr-gsm/python/receiver:/home/datdq/osmo/gr-gsm/python/transmitter:/home/datdq/osmo/gr-gsm/python/trx:/home/datdq/osmo/gr-gsm/build/swig::/usr/local/lib/python2.7/dist-packages" GRC_BLOCKS_PATH=/home/datdq/osmo/gr-gsm/grc /usr/local/bin/grcc -d /home/datdq/osmo/gr-gsm/build/apps /home/datdq/osmo/gr-gsm/apps/grgsm_livemon.grc
Permission denied
apps/CMakeFiles/pygen_apps.dir/build.make:104: recipe for target 'apps/grgsm_livemon' failed
make[2]: *** [apps/grgsm_livemon] Error 1
make[2]: Leaving directory '/home/datdq/osmo/gr-gsm/build'
CMakeFiles/Makefile2:1280: recipe for target 'apps/CMakeFiles/pygen_apps.dir/all' failed
make[1]: *** [apps/CMakeFiles/pygen_apps.dir/all] Error 2
make[1]: Leaving directory '/home/datdq/osmo/gr-gsm/build'
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2

Below is the output of commands that you request:

datdq@datdq:~/osmo/gr-gsm/build$ gnuradio-config-info --version
3.8.1.0
datdq@datdq:~/osmo/gr-gsm/build$ gnuradio-config-info --prefix
/usr/local
datdq@datdq:~/osmo/gr-gsm/build$ which grcc
/usr/bin/grcc
datdq@datdq:~/osmo/gr-gsm/build$ /usr/local/bin/grcc  --help
bash: /usr/local/bin/grcc: Is a directory
datdq@datdq:~/osmo/gr-gsm/build$ ls /usr/local/bin/grcc
datdq@datdq:~/osmo/gr-gsm/build$ ls -al  /home/datdq/osmo/gr-gsm/apps/
total 192
drwxrwxr-x  4 datdq datdq  4096 Th05  4 08:37 .
drwxrwxr-x 17 datdq datdq  4096 Th05  4 08:41 ..
drwxrwxr-x  2 datdq datdq  4096 Th05  4 08:37 apps_data
-rw-rw-r--  1 datdq datdq  1942 Th05  4 08:37 CMakeLists.txt
-rwxrwxr-x  1 datdq datdq 20323 Th05  4 08:37 grgsm_decode
-rw-rw-r--  1 datdq datdq 62212 Th05  4 08:37 grgsm_livemon.grc
-rw-rw-r--  1 datdq datdq 53884 Th05  4 08:37 grgsm_livemon_headless.grc
-rwxrwxr-x  1 datdq datdq 17262 Th05  4 08:37 grgsm_scanner
-rwxrwxr-x  1 datdq datdq  5278 Th05  4 08:37 grgsm_trx
drwxrwxr-x  2 datdq datdq  4096 Th05  4 08:37 helpers
-rw-rw-r--  1 datdq datdq  2227 Th05  4 08:37 README
datdq@datdq:~/osmo/gr-gsm/build$ git status
On branch master
Your branch is up-to-date with 'origin/master'.
nothing to commit, working directory clean
datdq@datdq:~/osmo/gr-gsm/build$ git describe
v0.42.2-55-gfa184a9

Can you support me to fix it ? Thank you so much

GOOD-Stuff, 2020-12-10:
I have the same issues.
```
$ gnuradio-config-info --version
3.7.9
$ gnuradio-config-info --prefix
/usr
$ git describe
v0.42.2-55-gfa184a9
```
@dangquocdat97 may be you should then close #529 issue?

GOOD-Stuff, 2020-12-10:
I fix this by installing libosmocore-dev:
`sudo apt install libosmocore-dev`

ptrkrysik, 2026-11-09:
Solved in the comments (install libosmocore-dev); the grcc-based generation step is also gone in the GNU Radio 3.10 port.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik