ptrkrysik/gr-gsm #534: grgsm no decoding on raspberry pi 3

jbarea, 2020-05-12:
Hello, i installed without issues grgsm on my raspberry pi as you described. I launch grgsm_livemon and i got the window with the frequency slider and spectrograph showing signals in the band. The problem i'm experiencing is that when i select a known frequency for a base station, the terminal should be showing the data stream but instead it remains silent. I know there should be data in the screen because when i change to my laptop and execute grgsm_livemon on the ubuntu installation there, data begins flowing and i can then work with it with wireshark, etc..

What could be causing this on the raspberry?. Building process went without errors when i followed the installation steps?.
Thanks in advance!

Xanuman, 2020-10-20:
I have a raspberry pi 4b (Rasbian 32 bit). GR-GSM decoding works, but data packets are being dropped all the time. Apparently there is not enough CPU power. It may be necessary to install a 64 bit system.
Raspberry 3 is less powerful. Maybe this is the problem?

ptrkrysik, 2026-11-09:
Most likely insufficient CPU on the Raspberry Pi 3 (dropped samples), as another user observed. Please retry with the new release and, if you still believe it is a bug, open a new issue with logs.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik