ptrkrysik/gr-gsm #537: Missing requirement for gr-gsm install, how do i install "libosmocoding" package on Ubuntu 18.04?

lovepie-ny, 2020-05-16:
I am new to ubuntu on rasp pi4b but i am becoming a fan instantly. 
Trying to tinker with a quick gr-gsm set up with a generic RTL dongle. 
I followed the instructions to the letter  in this link: 
https://osmocom.org/projects/gr-gsm/wiki/Installation
this section: 

Debbian-based distributions (Debian Testing, Ubuntu 16.04+, Kali Rolling Edition)

All the preceding steps are clean and successful. 

Got the following error in the cmake .. step: 
-- Checking for module 'libosmocoding'
--   No package 'libosmocoding' found
-- Could NOT find libosmocoding (missing: LIBOSMOCODING_LIBRARY LIBOSMOCODING_INCLUDE_DIR) 

I googled it and found nothing. please help. 
here are some of my set up info: 

cmake version 3.10.2

CMake suite maintained and supported by Kitware (kitware.com/cmake).

Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.4 LTS
Release:	18.04
Codename:	bionic

lovepie-ny, 2020-05-16:
Here is the full out of cmake .. 

-- The CXX compiler identification is GNU 7.5.0
-- The C compiler identification is GNU 7.5.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Found Git: /usr/bin/git (found version "2.17.1") 
-- Extracting version information from git describe...
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   filesystem
--   system
--   thread
--   chrono
--   date_time
--   atomic
-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.12.
-- Found SWIG: /usr/bin/swig3.0  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'gnuradio-runtime'
--   Found gnuradio-runtime, version 3.7.11
-- Found GNURADIO_RUNTIME: /usr/lib/aarch64-linux-gnu/libgnuradio-runtime.so;/usr/lib/aarch64-linux-gnu/libgnuradio-pmt.so;/usr/lib/aarch64-linux-gnu/liblog4cpp.so  
-- Checking for module 'gnuradio-filter'
--   Found gnuradio-filter, version 3.7.11
-- Found GNURADIO_FILTER: /usr/lib/aarch64-linux-gnu/libgnuradio-filter.so;/usr/lib/aarch64-linux-gnu/libgnuradio-fft.so;/usr/lib/aarch64-linux-gnu/libgnuradio-runtime.so;/usr/lib/aarch64-linux-gnu/libgnuradio-pmt.so;/usr/lib/aarch64-linux-gnu/liblog4cpp.so  
-- Checking for module 'gnuradio-runtime'
--   Found gnuradio-runtime, version 3.7.11
-- Found GNURADIO_PMT: /usr/lib/aarch64-linux-gnu/libgnuradio-runtime.so;/usr/lib/aarch64-linux-gnu/libgnuradio-pmt.so;/usr/lib/aarch64-linux-gnu/liblog4cpp.so  
-- Checking for module 'volk'
--   Found volk, version 1.3
-- Found VOLK: /usr/lib/aarch64-linux-gnu/libvolk.so  
-- Checking for module 'cppunit'
--   Found cppunit, version 1.14.0
-- Found CPPUNIT: /usr/lib/aarch64-linux-gnu/libcppunit.so;dl  
-- Checking for module 'libosmocore'
--   Found libosmocore, version 0.9.0
-- Found libosmocore: /usr/lib/aarch64-linux-gnu/libosmocore.so  
-- Checking for module 'libosmocodec'
--   Found libosmocodec, version 0.9.0
-- Found libosmocodec: /usr/lib/aarch64-linux-gnu/libosmocodec.so  
-- Checking for module 'libosmocoding'
--   No package 'libosmocoding' found
-- Could NOT find libosmocoding (missing: LIBOSMOCODING_LIBRARY LIBOSMOCODING_INCLUDE_DIR)** 
-- Checking for module 'libosmogsm'
--   Found libosmogsm, version 0.9.0
-- Found libosmogsm: /usr/lib/aarch64-linux-gnu/libosmogsm.so  
-- Compiling local version of libosmocoding
-- Loading build date  into constants...
-- Loading version 0.42.2. into constants...
-- Found PythonLibs: /usr/lib/aarch64-linux-gnu/libpython2.7.so (found suitable version "2.7.17", minimum required is "2") 
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- Performing Test HAVE_WNO_UNUSED_BUT_SET_VARIABLE
-- Performing Test HAVE_WNO_UNUSED_BUT_SET_VARIABLE - Success
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.13") found components:  doxygen dot 
-- Using install prefix: /usr/local
-- Building for version: 0.42.2. / 0.42.2.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ubuntu/gr-gsm/build

Yacinetel, 2020-05-17:
livepie-ny,  i am trying to install gr-gsm on ubuntu 18.04 , and it's gve me error on grgsm_livemon, like this:
grgsm_livemon -g 30
[INFO] [UHD] linux; GNU C++ version 7.5.0; Boost_106501; UHD_3.15.0.HEAD-0-g4e06022c
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

Cannot connect to server socket err = Aucun fichier ou dossier de ce type
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Found Rafael Micro R820T tuner
gr-osmosdr 0.1.5 (0.1.5) gnuradio 3.7.14.0
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf rfspace airspy soapy redpitaya 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000,052982 Hz
[R82XX] PLL not locked!
Erreur de segmentation (core dumped)

could you give me help to send me your step to install grgsm, UHD, gnu osmocom .....
please.

ptrkrysik, 2026-11-09:
Missing/too-old libosmocore development package on Ubuntu 18.04; install libosmocore-dev (or build libosmocore from source). Closing as an installation question.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik