ptrkrysik/gr-gsm #540: no results with porting_to_gr38 branch

blshkv, 2020-05-27:
Hello I'm using the latest stable packages including gnuradio 3.8.1.0 and the latest git from porting_to_gr38 branch.
grgsm_scanner does not find any networks (while kali does):
```
grgsm_livemon -f 940.0M -g 7
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file hackrf airspy soapy 
Using HackRF One with firmware 2018.01.1
```

grgsm_livemon does not show any networks, just a white noise (while gqrx does).

So something looks broken. Any hits?

```
hackrf_info 
hackrf_info version: 2018.01.1
libhackrf version: 2018.01.1 (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000457863c823441c1f
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x006d4751
```

ptrkrysik, 2026-11-09:
Report was against the old porting_to_gr38 branch with very low gain and no concrete defect; please retest with the new GNU Radio 3.10 release and open a new issue with logs if it persists.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik

blshkv, 2026-11-09:
it took you 6 years to come back. Seriously? I don't even remember what it is about and not going to waste my time reporting you any bugs.

ptrkrysik, 2026-11-10:
W dniu 9.11.2026 o 15:38, Anton Bolshakov pisze:
> *blshkv* left a comment (ptrkrysik/gr-gsm#540) <https://github.com/ 
> ptrkrysik/gr-gsm/issues/540#issuecomment-6742748320>
> 
> it took you 6 years to come back. Seriously? I don't even remember what 
> it is about and not going to waste my time reporting you any bugs.
> 
Sorry, the response was actually generated by an LLM and it is in fact a 
bit dumb. I'm trying to get the project up after long period of 
inactivity and I'm not expecting you to do anything with it.