ptrkrysik/gr-gsm #543: Kali linux 2020.2, gnuradio GSM reciever error

kolegmux, 2020-06-23:
Hello, please, help. I have error:

Block - gsm_receiver_0 - GSM Receiver(gsm_receiver):
Param - Cell allocation(cell_allocation):
Value "[arfcn.downlink2arfcn(fc)]" cannot be evaluated:
name 'arfcn' is not defined

Tried so many times to reinstall/install operating system, gnu radio, gr-osmocom and all all all, nothing helps. Kali linux 2020.2, gnu radio 3.8, and gr-gsm from github.com/velichkov

Thank you.

[image]

Xanuman, 2020-10-20:
Perhaps the problem is:
1)
/etc/gnuradio/conf.d/grc.conf
On the line local_blocks_path = / home / User / gr-gsm / grc

2) GRGSM and PYTHON do not see the contents of subfolders in /usr/lib/python2.7/dist_packages/grgsm
Or
/ usr / lib / python3 / dist_packages / grgsm

3) I understand that there is a problem of incompatibility between GnuRadio 3.8 and libosmoсore + Osmosdr

The creators of the OSMOCOM project, as I understand it, have not yet had time to change their instruments for the updated gnuradio. Therefore, you need to install the old Gnuradio 3.7 created for Python2

ptrkrysik, 2026-11-09:
This was an old GNU Radio 3.8 GRC module-path problem; the GNU Radio 3.10 port uses gnuradio.gsm and imports arfcn correctly. Closing as obsolete.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.