ptrkrysik/gr-gsm #549: GnuRadio Runtime required to compile gr-gsm error

Dm1027, 2020-07-23:
Hello, I am working to install gr-gsm and receiving this error every install attempt

CMake Error at CMakeLists.txt:150 (message):
  GnuRadio Runtime required to compile gr-gsm

I have manually installed GnuRadio several times, both with purging it/reinstalling and on completely fresh installs of ubuntu 20.04 to ensure no files are still around. 

I've tried different GnuRadio versions including the recommended 3.7 from other issues threads with no luck. 

I've followed 4 or 5 different install guides and just can't get it working. I am assuming its something I am missing, but if you can point me in the right direction I would appreciate the help. I am trying to use this software for a college project and am completely stuck after numerous attempts. 

Thank you

rfguru-rufus, 2021-07-12:
facing same issue ubuntu 20.04
any one able to fix? 

[image]

velichkov, 2021-07-13:
Hi @rfguru-rufus,

> facing same issue ubuntu 20.04
> any one able to fix?

Please next time copy-paste the text from console and do not attach screenshots.

How did you install `gnuradio`, `volk` and which versions? Which `gr-gsm` version are you trying to compile?

I see you have `volk` in `/usr/local` while your `gnuradio` is in `/usr/`, this probably means you have two `volk` versions installed on your system. Why do you install `volk` from source? Try removing it - go in its `build` directory and run `make uninstall`. Then install `libvolk2-dev` package, go in the `gr-gsm` build directory, remove CMakeCache.txt file and try again.

ptrkrysik, 2026-11-09:
This is an environment/dependency problem (GNU Radio or Volk not found by CMake, often from a mixed source/distro install) and the old build instructions no longer apply to the GNU Radio 3.10 port. Please re-open with a fresh log if it still happens with the new release.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik