ptrkrysik/gr-gsm #551: Runtime error in grgsm_scanner

accelleon, 2020-07-29:
I receive the following output when running

`grgsm_scanner -d -b GSM900`

`[bladeRF common] init: DEBUG: entering initialization
[bladeRF source] init: Opening Nuand bladeRF with device identifier string '*:instance=0'
[bladeRF source] Device: Nuand bladeRF 2.0 Serial # 5de4...cd2b FW v2.3.2 FPGA v0.11.0
[bladeRF source] init: Buffers: 512, samples per buffer: 4096, active transfers: 32
[bladeRF source] bladerf_source_c: DEBUG: initialization complete

Args= 
[INFO] [UHD] linux; GNU C++ version 9.3.0; Boost_107100; UHD_3.15.0.0-3
[ERROR] avahi_service_browser_new() failed: Bad state
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
[bladeRF source] bladerf_source_c: Warning: number of channels specified on command line (4) is greater than the maximum number supported by this device (2). Resetting to 2.
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_scanner", line 438, in <module>
    main()
  File "/usr/local/bin/grgsm_scanner", line 434, in main
    do_scan(options.samp_rate, options.band, options.speed,
  File "/usr/local/bin/grgsm_scanner", line 330, in do_scan
    scanner = wideband_scanner(rec_len=6 - speed,
  File "/usr/local/bin/grgsm_scanner", line 211, in __init__
    self.rtlsdr_source = osmosdr.source(args="numchan=" + str(1) + " " +
  File "/usr/lib/python3/dist-packages/osmosdr/osmosdr_swig.py", line 1074, in make
    return _osmosdr_swig.source_make(*args, **kwargs)
RuntimeError: destination port 1 out of range for source_impl(1)
`

zR4in, 2023-07-09:
So have you solved this?

ptrkrysik, 2026-11-09:
The scanner code that built the osmosdr source with a numchan argument has been rewritten in the GNU Radio 3.10 port. Please retest with the new release and open a new issue with logs if the bladeRF still fails.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik