ptrkrysik/gr-gsm #552: grgsm_scanner no output grgsm_livemon display issue

supashot, 2020-07-31:
OS: Linux kali 5.7.0-kali1-amd64 #1 SMP Debian 5.7.6-1kali2 (2020-07-01) x86_64 GNU/Linux on vmware 15
installation: sudo apt-get install gr-gsm (v0.42.2.20200214)
gnuradio-companion 3.8.1.0
gr-osmosdr 0.2.0.0
I use a PortaPack in HackRF one mode (usb connected to my laptop)

2 issues:

1. No output when I run `grgsm_scanner -b GSM900 -vd

Args= 
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file rtl rtl_tcp hackrf rfspace soapy redpitaya 
Using HackRF One with firmware local-9e5eebae
`
grgsm_scanner detects HackRF One, that
I also tried to run `sudo grgsm_scanner -b GSM900 -vd`, same no results.
Also when use ctrl-z it leave the process running in background so will have kill -9 pid the process.

2.running grgsm_livemon:
'grgsm_livemon
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file rtl rtl_tcp hackrf rfspace soapy redpitaya 
Using HackRF One with firmware local-9e5eebae'

It brings the monitor with a big spike in middle, and the display 
[image]
Changing frequency doesn't bring any significant changes on the graph.
 I use a frequency I scanned with another device (RTL2832u), no problems with it. I get the data but not with HackRF One.
Running `grgsm_scanner -b GSM900 -vd` with RTL2832u works also fine.

gr-gsm seems like installed properly and seems like being an issue with the HackRF one only.

Any hints would be much appreciated, despite hours of search I didn't find any solutions.

alexdparis, 2021-04-04:
Hi, same problem here. Did you find any solution ?

alexdparis, 2021-04-04:
Ok, for my case, I just went outisde my house (which has fat stone walls), and I gained +10 Pwr Value (from -60 to -50), then I successfully got  data. Check the Pwr value at the end of line in grgsm_scanner.

ptrkrysik, 2026-11-09:
This is a signal level/gain/location issue with the SDR, not a defect in gr-gsm (a later commenter fixed it by getting better reception). Please re-open with details if it persists with the new release.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik