ptrkrysik/gr-gsm #553: GSM reciever block invalid parameter

rapbando, 2020-08-01:
[image]
After a long day troubleshooting the installation in a stock fresh Kali rolling release I ended up with this last error and I don't know where to go from here, I've read all issues about gr-gsm but can t find any reference to fix this. 
I`m just one step away to be able to run the flowgraph but this error inside the GSM receiver block keeps me stuck:

`Param - Cell allocation(cell_allocation):
	Expression None is invalid for type'float_vector'.`

velichkov, 2021-05-05:
Hi @rapbando,

> Param - Cell allocation(cell_allocation): Expression None is invalid for type'float_vector'.

The easiest fix is to set Cell allocation to an empty vector - `[]`

> I`m just one step away to be able to run the flowgraph but this error inside the GSM receiver block keeps me stuck:

The flowgraph gets compiled and installed as a python script. Are you able to run the python script?

whatever2010, 2022-05-05:
@rapbando did you find any solution ?

rapbando, 2022-05-06:
> @rapbando did you find any solution ?

Actually for other reasons I ended up using pentoo linux for gnuradio stuff.

whatever2010, 2022-05-07:
oh finally i found the solution :)
`wget https://lab.nethence.com/elge/tarballs/arfcncalc.tar
    #wget http://www.runningserver.com/software/arfcncalc.tar
    tar xf arfcncalc.tar
    cp arfcncalc/arfcncalc /usr/local/bin/
    arfcncalc -h`

ptrkrysik, 2026-11-09:
The receiver block in the new port ships with a valid default for 'Cell allocation' ('[0]'), and the old hierarchical block was reworked. The old workaround was to set an explicit vector such as '[]'.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik