ptrkrysik/gr-gsm #554: I need grgsm_capture.grc file Where Can I find it ?

muratoznn, 2020-08-17:
I need grgsm_capture.grc file Where Can I find it ?

velichkov, 2021-05-05:
Hi @muratoznn,

The `grgsm_capture` is not generated from a flowgraph (grc file). It is a python script that uses gnuradio directly.

ptrkrysik, 2026-11-09:
grgsm_capture is a Python script (installed as `grgsm_capture`) and is not generated from a flowgraph, so there is no .grc file for it.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik