ptrkrysik/gr-gsm #556: No package 'libosmocoding' found

Mdxjj, 2020-09-15:
I got an error when running "cmake .."while compiling gr-gsm，after referring to https://github.com/ptrkrysik/gr-gsm/issues/416, I installed the necessary software,But still something went wrong。
Linux version: ubuntu16.04

-- Checking for module 'libosmocoding'
--   No package 'libosmocoding' found
-- Could NOT find libosmocoding (missing: LIBOSMOCODING_LIBRARY LIBOSMOCODING_INCLUDE_DIR) 
-- Compiling local version of libosmocoding
-- Loading build date Tue, 15 Sep 2020 09:13:29 into constants...
-- Loading version 0.42.2. into constants...
-- Using install prefix: /usr/local
-- Building for version: 0.42.2. / 0.42.2.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/md/gr-gsm/build

velichkov, 2021-05-05:
>  Linux version: ubuntu16.04

This version is ancient and probably does not have `libosmocoding-dev` package.  If `sudo apt-get install libosmocoding-dev` does not install it than my recommendation is to update your Ubuntu to a more recent version.

ptrkrysik, 2026-11-09:
Ubuntu 16.04 is unsupported and the output shown is not an error (cmake fell back to the local libosmocoding copy). Closing as an environment issue.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik