ptrkrysik/gr-gsm #560: RtApiAlsa::getDeviceInfo:

UbuntuEvangelist, 2020-11-22:
after run $ sudo grgsm_livemon -f 000.0M 

it shows this eror: 
RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (hw:0,0), Device or resource busy.

[image]

velichkov, 2021-05-05:
Hi @UbuntuEvangelist,

> RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (hw:0,0), Device or resource busy.

This error together with the Jack server and few others are generated during detection of the SDR device. As you are not specifying it the gr-osmosdr tries to detects it by trying to open all supported devices one by one.

To suppress these errors you can specify your SDR device in the `--args` arguments, for example if you have an RTL dongle you can use `--args=rtl` or `--args=rtl=0`. You can find some more info in the [gr-osmosdr  wiki](https://osmocom.org/projects/gr-osmosdr/wiki#Device-specification)

ptrkrysik, 2026-11-09:
These messages are printed by gr-osmosdr while probing devices; specifying the device with --args (for example --args=rtl=0) suppresses them.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik