ptrkrysik/gr-gsm #561: Error installing gr-gsm on Ubuntu 20.04.1 LTS

carlosrodriguesIT, 2020-12-20:
Hello,

After perform the steps:

git clone https://git.osmocom.org/gr-gsm
cd gr-gsm
mkdir build
cd build
cmake ..

A get the following error:

-- Extracting version information from git describe...
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.35") found components: filesystem system thread 
-- 
-- Checking for module SWIG
-- Found SWIG version 4.0.1.
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR) 
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.71.0") found components: date_time program_options filesystem system regex thread unit_test_framework 
CMake Warning at CMakeLists.txt:135 (find_package):
  Found package configuration file:

    /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GnuradioConfig.cmake

  but it set Gnuradio_FOUND to FALSE so package "Gnuradio" is considered to
  be NOT FOUND.  Reason given by package:

  Gnuradio could not be found because dependency Volk could not be found.

CMake Error at CMakeLists.txt:150 (message):
  GnuRadio Runtime required to compile gr-gsm

-- Configuring incomplete, errors occurred!
See also "/home/carlos/rf/gr-gsm/build/CMakeFiles/CMakeOutput.log".
See also "/home/carlos/rf/gr-gsm/build/CMakeFiles/CMakeError.log".

How can fix these errors? Dependencies?

Thanks in advance

FletcherMunson, 2020-12-30:
I'm experiencing the same issue.

Gnuradio 3.8.1.0 (Python 3.8.5)

```
sudo apt install gnuradio-dev 
 sudo apt install cmake
 sudo apt install autoconf libtool pkg-config g++ gcc libc6-dev libcppunit-1.15-0 libcppunit-dev swig doxygen liblog4cpp5v5 liblog4cpp5-dev  gr-osmosdr libosmocore libosmocore-dev 
 git clone https://git.osmocom.org/gr-gsm
 cd gr-gsm/
 mkdir build
 cd build/
 cmake ..
```

```
-- Extracting version information from git describe...
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.35") found components: filesystem system thread 
-- 
-- Checking for module SWIG
-- Found SWIG version 4.0.1.
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR) 
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.71.0") found components: date_time program_options filesystem system regex thread unit_test_framework 
CMake Warning at CMakeLists.txt:135 (find_package):
  Found package configuration file:

    /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GnuradioConfig.cmake

  but it set Gnuradio_FOUND to FALSE so package "Gnuradio" is considered to
  be NOT FOUND.  Reason given by package:

  Gnuradio could not be found because dependency Volk could not be found.

CMake Error at CMakeLists.txt:150 (message):
  GnuRadio Runtime required to compile gr-gsm

```

-- Configuring incomplete, errors occurred!
See also "/home/user0/gr-gsm/build/CMakeFiles/CMakeOutput.log".
See also "/home/user0/gr-gsm/build/CMakeFiles/CMakeError.log".

melfvelin, 2021-01-02:
I was experiencing the same problem but managed to successfully install after trying a bunch of different alternatives. I think the issue might have been trying to install gr-gsm from a branch that was not compatible with gnuradio 3.8

I followed the same instructions (installing gnuradio via apt-get, installing all dependencies (including volk, gr-osmosdr etc.)

Then I installed gr-gsm from Velichkovs github using the maint-3.8 fork by:
`git clone --branch maint-3.8 https://github.com/velichkov/gr-gsm.git
mkdir build
cd build
cmake ..
sudo make install
sudo ldconfig`

Try that and if it doesn't work still I can try to figure out the exact instructions I used and all the dependencies.

FletcherMunson, 2021-01-03:
The install got a lot farther, looks like I have some Makefile issues to sort out. 
```
...
[ 63%] Building CXX object lib/CMakeFiles/grgsm.dir/trx/trx_burst_if_impl.cc.o
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/liborc-0.4.so', needed by 'lib/libgrgsm.so.v0.41.1-346-g2efaa49e'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:953: lib/CMakeFiles/grgsm.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

MountainLogic, 2021-04-08:
FYI, I have been able to successfully build and install in Ubuntu 10.4 for GnuRadio  v3.8.2.0-106-gefac6d64 (Python 3.8.5) using a variation on @melfvelin 's instructions as I had already cloned the master branch: git checkout -b porting_to_gr38 remotes/origin/porting_to_gr38

velichkov, 2021-05-05:
Hi folks,

The support for gnuradio 3.8 was merged in the master branch so my recommendation is to switch back to using it.

ptrkrysik, 2026-11-09:
This was caused by building a GNU Radio 3.7 branch against 3.8 and missing Volk; 3.8 support was merged later and the new release targets GNU Radio 3.10.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik