ptrkrysik/gr-gsm #562: Make error

PHPoenX, 2021-01-12:
CMake output:
```-- Build type not specified: defaulting to release.
-- Checking for module 'mpir >= 3.0'
--   Package 'mpir', required by 'virtual:world', not found
-- User set python executable /usr/bin/python3
-- Found PythonLibs: /usr/lib64/libpython3.8.so (found suitable exact version "3.8.6") 
-- Extracting version information from git describe...
-- Checking for module 'libosmocodec'
--   Found libosmocodec, version 1.4.1
-- Found libosmocodec: /usr/lib64/libosmocodec.so  
-- Checking for module 'libosmocoding'
--   Found libosmocoding, version 1.4.1
-- Found libosmocoding: /usr/lib64/libosmocoding.so  
-- Loading build date Tue, 12 Jan 2021 16:17:50 into constants...
-- Loading version v0.41.1-346-g2efaa49e into constants...
-- Using install prefix: /usr/local
-- Building for version: v0.41.1-346-g2efaa49e / 1.0.0git
-- No C++ unit tests... skipping
-- 
-- Checking for module SWIG
-- Found SWIG version 4.0.2.
-- Found PythonLibs: /usr/lib64/libpython3.8.so (found version "3.8.6") 
-- Configuring done
-- Generating done
```

Make output:
```
gr-gsmUpstream/lib/decoding/control_channels_decoder_impl.cc:73:100: error: ‘_1’ was not declared in this scope
   73 |       set_msg_handler(pmt::mp("bursts"), boost::bind(&control_channels_decoder_impl::decode, this, _1));
```

openSUSE tumbleweed, boost installed from repo, current version is 1.75.0

PHPoenX, 2021-01-22:
Sorry, I forgot notice that I used branch porting_to_gr38

lordgurke, 2021-03-05:
Ran into the exact same problem, using current Arch Linux with boost 1.75.0-2 from repositories.
Is there any solution yet?

lordgurke, 2021-03-18:
Figured out what's happening:
Newer versions of boost don't understand the `_1` reference and these need to be replaced by `boost::placeholders::_1`.
Attached patchfile will fix these for "porting_to_gr38" branch.

```
diff --git a/lib/decoding/control_channels_decoder_impl.cc b/lib/decoding/control_channels_decoder_impl.cc
index 81ce161..b49582f 100644
--- a/lib/decoding/control_channels_decoder_impl.cc
+++ b/lib/decoding/control_channels_decoder_impl.cc
@@ -70,7 +70,7 @@ namespace gr {
     {
       //setup input/output ports
       message_port_register_in(pmt::mp("bursts"));
-      set_msg_handler(pmt::mp("bursts"), boost::bind(&control_channels_decoder_impl::decode, this, _1));
+      set_msg_handler(pmt::mp("bursts"), boost::bind(&control_channels_decoder_impl::decode, this, boost::placeholders::_1));
       message_port_register_out(pmt::mp("msgs"));
     }
 
diff --git a/lib/decoding/tch_f_decoder_impl.cc b/lib/decoding/tch_f_decoder_impl.cc
index 02dc45a..03e8fc8 100644
--- a/lib/decoding/tch_f_decoder_impl.cc
+++ b/lib/decoding/tch_f_decoder_impl.cc
@@ -92,7 +92,7 @@ namespace gr {
     {
         //setup input/output ports
         message_port_register_in(pmt::mp("bursts"));
-        set_msg_handler(pmt::mp("bursts"), boost::bind(&tch_f_decoder_impl::decode, this, _1));
+        set_msg_handler(pmt::mp("bursts"), boost::bind(&tch_f_decoder_impl::decode, this, boost::placeholders::_1));
         message_port_register_out(pmt::mp("msgs"));
         message_port_register_out(pmt::mp("voice"));    
 
diff --git a/lib/decoding/tch_h_decoder_impl.cc b/lib/decoding/tch_h_decoder_impl.cc
index b6827f7..f1ba5ba 100644
--- a/lib/decoding/tch_h_decoder_impl.cc
+++ b/lib/decoding/tch_h_decoder_impl.cc
@@ -81,7 +81,7 @@ namespace gr {
         {
             //setup input/output ports
             message_port_register_in(pmt::mp("bursts"));
-            set_msg_handler(pmt::mp("bursts"), boost::bind(&tch_h_decoder_impl::decode, this, _1));
+            set_msg_handler(pmt::mp("bursts"), boost::bind(&tch_h_decoder_impl::decode, this, boost::placeholders::_1));
             message_port_register_out(pmt::mp("msgs"));
             message_port_register_out(pmt::mp("voice"));
 
diff --git a/lib/decryption/decryption_impl.cc b/lib/decryption/decryption_impl.cc
index 551afa4..d189596 100644
--- a/lib/decryption/decryption_impl.cc
+++ b/lib/decryption/decryption_impl.cc
@@ -61,7 +61,7 @@ namespace gr {
         validate_k_c();
 
         message_port_register_in(pmt::mp("bursts"));
-        set_msg_handler(pmt::mp("bursts"), boost::bind(&decryption_impl::decrypt, this, _1));
+        set_msg_handler(pmt::mp("bursts"), boost::bind(&decryption_impl::decrypt, this, boost::placeholders::_1));
         message_port_register_out(pmt::mp("bursts"));
     }
 
diff --git a/lib/demapping/tch_f_chans_demapper_impl.cc b/lib/demapping/tch_f_chans_demapper_impl.cc
index bf1ff8f..ee0c05c 100644
--- a/lib/demapping/tch_f_chans_demapper_impl.cc
+++ b/lib/demapping/tch_f_chans_demapper_impl.cc
@@ -58,7 +58,7 @@ namespace gr {
         //        }
 
         message_port_register_in(pmt::mp("bursts"));
-        set_msg_handler(pmt::mp("bursts"), boost::bind(&tch_f_chans_demapper_impl::filter_tch_chans, this, _1));
+        set_msg_handler(pmt::mp("bursts"), boost::bind(&tch_f_chans_demapper_impl::filter_tch_chans, this, boost::placeholders::_1));
         message_port_register_out(pmt::mp("tch_bursts"));
         message_port_register_out(pmt::mp("acch_bursts"));
     }
[... 382 lines cut ...]
 
     /*
diff --git a/lib/trx/trx_burst_if_impl.cc b/lib/trx/trx_burst_if_impl.cc
index fed84a7..70d7d32 100644
--- a/lib/trx/trx_burst_if_impl.cc
+++ b/lib/trx/trx_burst_if_impl.cc
@@ -76,7 +76,7 @@ namespace gr {
 
         // Bind a port handler
         set_msg_handler(pmt::mp("bursts"),
-          boost::bind(&trx_burst_if_impl::handle_dl_burst, this, _1));
+          boost::bind(&trx_burst_if_impl::handle_dl_burst, this, boost::placeholders::_1));
 
         // Prepare port numbers
         std::string data_src_port = boost::lexical_cast<std::string> (base_port + 2);
@@ -88,7 +88,7 @@ namespace gr {
 
         // Bind DATA interface handler
         d_data_sock->udp_rx_handler = boost::bind(
-          &trx_burst_if_impl::handle_ul_burst, this, _1, _2);
+          &trx_burst_if_impl::handle_ul_burst, this, boost::placeholders::_1, boost::placeholders::_2);
     }
 
     /*
```

velichkov, 2021-05-05:
This is fixed in the latest master branch a116920. The support for gnuradio 3.8 was also merged in the master branch recently.

ptrkrysik, 2026-11-09:
Already fixed: the code now uses boost::placeholders::_1 (fixed in master and present in the GNU Radio 3.10 port).

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik