ptrkrysik/gr-gsm #563: grgsm.tch_h_decoder AttributeError

abbdulwahab86, 2021-01-22:
**Hi Folks**,
Everything working fine on my installation. However,  gnuradio produce error on every value (GSM-HS, GSM-AMR 7.4,GSM-AMR 7.95,GSM-AMR 6.7,GSM-AMR 5.15,GSM-AMR 4.75,GSM-AMR 5.9) of TCH/H decoder block drop down options.
**#uname** - a Linux butter-virtual-machine 5.3.0-53-generic #47~18.04.1-Ubuntu SMP Thu May 7 13:10:50 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

**GnuRadio 3.7.11**

 (return code 1)
**GSM-AMR 7.4** -> AttributeError: 'module' object has no attribute 'TCH_AHS7_4'
**GSM-AMR 7.95** -> AttributeError: 'module' object has no attribute 'TCH_AHS7_95'
**GSM-AMR 6.7** -> AttributeError: 'module' object has no attribute 'TCH_AHS6_7'
**GSM-AMR 5.15** -> AttributeError: 'module' object has no attribute 'TCH_AHS5_15'
**GSM-AMR 4.75** -> AttributeError: 'module' object has no attribute 'TCH_AHS4_75'
**GSM-AMR 5.9** -> AttributeError: 'module' object has no attribute 'TCH_AHS5_9'
**GSM-HS** -> TypeError: in method 'tch_h_decoder_make', argument 2 of type 'std::string'

velichkov, 2021-05-05:
Hi @abbdulwahab86,

Which gr-gsm version do you use and how did you installed it?

ptrkrysik, 2026-11-09:
The TCH/H decoder block was reworked and the TCH_AHS* constants referenced here no longer exist; the report was for GNU Radio 3.7 with an outdated install. Please open a new issue if you see problems with the new release.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik