ptrkrysik/gr-gsm #564: gr-gsm install error

woodpecker5, 2021-01-27:
When I execute the **cmake ..** the following error showing up on Ubuntu 20.04

-- Extracting version information from git describe...
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.35") found components: filesystem system thread 
-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.12.
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR) 
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.71.0") found components: date_time program_options filesystem system regex thread unit_test_framework 
CMake Warning at CMakeLists.txt:135 (find_package):
  Found package configuration file:

    /usr/local/lib/cmake/gnuradio/GnuradioConfig.cmake

  but it set Gnuradio_FOUND to FALSE so package "Gnuradio" is considered to
  be NOT FOUND.  Reason given by package:

  Gnuradio could not be found because dependency Volk could not be found.

-- Checking for module 'libosmocore'
--   No package 'libosmocore' found
-- Could NOT find libosmocore (missing: LIBOSMOCORE_LIBRARY LIBOSMOCORE_INCLUDE_DIR) 
-- Checking for module 'libosmocodec'
--   No package 'libosmocodec' found
-- Could NOT find libosmocodec (missing: LIBOSMOCODEC_LIBRARY LIBOSMOCODEC_INCLUDE_DIR) 
-- Checking for module 'libosmocoding'
--   No package 'libosmocoding' found
-- Could NOT find libosmocoding (missing: LIBOSMOCODING_LIBRARY LIBOSMOCODING_INCLUDE_DIR) 
-- Checking for module 'libosmogsm'
--   No package 'libosmogsm' found
-- Could NOT find libosmogsm (missing: LIBOSMOGSM_LIBRARY LIBOSMOGSM_INCLUDE_DIR) 
CMake Error at CMakeLists.txt:150 (message):
  GnuRadio Runtime required to compile gr-gsm

-- Configuring incomplete, errors occurred!
See also "/home/woodpecker/gr-gsm/build/CMakeFiles/CMakeOutput.log".
See also "/home/woodpecker/gr-gsm/build/CMakeFiles/CMakeError.log".

PHPoenX, 2021-01-30:
@woodpecker5
>  No package 'mpir' found
>  Volk could not be found
> No package 'libosmocore' found
> ...

You must install all build dependencies

ptrkrysik, 2026-11-09:
Missing build dependencies (Volk, libosmocore, etc.) on the user's system; please install them as described in the README for the new release.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik