ptrkrysik/gr-gsm #565: grgsm_scanner: no results, kernel driver loops over deattach/reattach

blackbit42, 2021-02-10:
Hi

I am on Debian 11/amd64 with gr-gsm installed via apt.
kalibrate-rtl works as expected.
grgsm_scanner seems to be stuck in a loop. Or is this behavior expected?
At any rate, it produces no results.

I am not sure how to debug this.

```
$ grgsm_scanner --args=rtl=0 -b DCS1800 -v -g 40 --debug

Args= rtl=0
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR
Detached kernel driver
Found Elonics E4000 tuner
Exact sample rate is: 2000000.052982 Hz
Allocating 15 zero-copy buffers
Reattached kernel driver
Args= rtl=0
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR
Detached kernel driver
Found Elonics E4000 tuner
Exact sample rate is: 2000000.052982 Hz
Allocating 15 zero-copy buffers
Reattached kernel driver
Args= rtl=0
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR
Detached kernel driver
Found Elonics E4000 tuner
Exact sample rate is: 2000000.052982 Hz
Allocating 15 zero-copy buffers
Reattached kernel driver
Args= rtl=0
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR
Detached kernel driver
Found Elonics E4000 tuner
Exact sample rate is: 2000000.052982 Hz
Allocating 15 zero-copy buffers
^Cterminate called after throwing an instance of 'boost::thread_interrupted'
Aborted
```

velichkov, 2021-05-05:
> grgsm_scanner seems to be stuck in a loop. Or is this behavior expected?

The loop is expected as it scans over all frequencies of the selected band `-b DCS1800`. The process is not fast so wait patiently until it finishes.  

> Found Elonics E4000 tuner

You have an Elonics tuner so the DCS1800 band should be supported. Just make sure that in the country and region where you live there are active GSM cells in this band.

In case you are testing inside of a big building try moving near a window or outside if possible.

blackbit42, 2021-05-18:
Indeed, it finishes after 3 minutes.
I ran into [this](https://stackoverflow.com/questions/24486153/gnu-radio-python-script-shmget-2-no-space-left-on-device) issue, but that was easy to overcome.

However, no results are being printed. I know there are multiple DCS1800 operators here though.
What could be wrong?

```$ /usr/bin/time grgsm_scanner --args=rtl=0 -b DCS1800 -v -g 40 --debug

Args= rtl=0
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR
Detached kernel driver
Found Elonics E4000 tuner
Exact sample rate is: 2000000.052982 Hz
Allocating 15 zero-copy buffers
Reattached kernel driver
Args= rtl=0
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR
Detached kernel driver
Found Elonics E4000 tuner
Exact sample rate is: 2000000.052982 Hz
Allocating 15 zero-copy buffers
Reattached kernel driver
Args= rtl=0
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR
Detached kernel driver
Found Elonics E4000 tuner
Exact sample rate is: 2000000.052982 Hz
Allocating 15 zero-copy buffers
Reattached kernel driver
Args= rtl=0
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR
Detached kernel driver
Found Elonics E4000 tuner
Exact sample rate is: 2000000.052982 Hz
Allocating 15 zero-copy buffers
Reattached kernel driver
Args= rtl=0
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR
Detached kernel driver
Found Elonics E4000 tuner
Exact sample rate is: 2000000.052982 Hz
Allocating 15 zero-copy buffers
Reattached kernel driver
Args= rtl=0
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR
Detached kernel driver
Found Elonics E4000 tuner
Exact sample rate is: 2000000.052982 Hz
Allocating 15 zero-copy buffers
Reattached kernel driver
Args= rtl=0
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR
Detached kernel driver
Found Elonics E4000 tuner
Exact sample rate is: 2000000.052982 Hz
Allocating 15 zero-copy buffers
[... 257 lines cut ...]
Detached kernel driver
Found Elonics E4000 tuner
Exact sample rate is: 2000000.052982 Hz
Allocating 15 zero-copy buffers
Reattached kernel driver
Args= rtl=0
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR
Detached kernel driver
Found Elonics E4000 tuner
Exact sample rate is: 2000000.052982 Hz
Allocating 15 zero-copy buffers
Reattached kernel driver
Args= rtl=0
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR
Detached kernel driver
Found Elonics E4000 tuner
Exact sample rate is: 2000000.052982 Hz
Allocating 15 zero-copy buffers
Reattached kernel driver
313.38user 12.73system 2:59.43elapsed 181%CPU (0avgtext+0avgdata 497092maxresident)k
0inputs+12464outputs (37795major+81552minor)pagefaults 0swaps```

velichkov, 2021-05-19:
> know there are multiple DCS1800 operators here though.

How do you know that?

> What could be wrong?

Does it find cells in the default `GSM900` band? Do you have a suitable antenna? Have you tried near a window or outside? Have you tried with a different gain setting?

```
  -g GAIN, --gain=GAIN  Set gain [default=24.0]
```

ptrkrysik, 2026-11-09:
The repeated device open messages are expected while scanning all channels, and the lack of results points to antenna/gain/band availability rather than a defect. Please re-open with more detail if it still fails with the new release.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik