ptrkrysik/gr-gsm #567: livemon dont start

vespertillo, 2021-04-06:
How i can solve? i think i update the system now dont work
grgsm_livemon -f 938.2M
[INFO] [UHD] linux; GNU C++ version 10.2.1 20201207; Boost_107400; UHD_3.15.0.0-4+b1
[ERROR] avahi_service_browser_new() failed: Bad state
Detached kernel driver
Found Rafael Micro R820T tuner
Reattached kernel driver
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000,052982 Hz
[R82XX] PLL not locked!
Could not find port: bursts in:
bursts
system

Traceback (most recent call last):
  File "/usr/local/bin/grgsm_livemon", line 376, in <module>
    main()
  File "/usr/local/bin/grgsm_livemon", line 354, in main
    tb = top_block_cls(args=options.args, collector=options.collector, collectorport=options.collectorport, fc=options.fc, gain=options.gain, osr=options.osr, ppm=options.ppm, samp_rate=options.samp_rate, serverport=options.serverport, shiftoff=options.shiftoff)
  File "/usr/local/bin/grgsm_livemon", line 162, in __init__
    self.gsm_sdcch8_demapper_0 = grgsm.gsm_sdcch8_demapper(
  File "/usr/local/lib/python3/dist-packages/grgsm/gsm_sdcch8_demapper.py", line 59, in __init__
    self.msg_connect((self.gsm_universal_ctrl_chans_demapper_0, 'bursts'), (self, 'bursts'))    
  File "/usr/lib/python3/dist-packages/gnuradio/gr/hier_block2.py", line 60, in wrapped
    func(self, src.to_basic_block(), srcport, dst.to_basic_block(), dstport)
  File "/usr/lib/python3/dist-packages/gnuradio/gr/hier_block2.py", line 132, in msg_connect
    self.primitive_msg_connect(*args)
  File "/usr/lib/python3/dist-packages/gnuradio/gr/runtime_swig.py", line 3006, in primitive_msg_connect
    return _runtime_swig.hier_block2_sptr_primitive_msg_connect(self, *args)
RuntimeError: invalid msg port in connect() or disconnect()
Reattached kernel driver

velichkov, 2021-05-05:
Hi @vespertillo,

> How i can solve? i think i update the system now dont work

My advice is to uninstall gr-gsm (run `make uninstall` from the build directory), then manually remove every grgsm's file in `/usr/local` sub-directories (`/usr/local/bin/`, `/usr/local/lib64`, `/usr/local/lib/python3/dist-packages/grgsm/`, ....) then remove the build directory and rebuild/reinstall from scratch.

ptrkrysik, 2026-11-09:
This was caused by a stale/mixed gr-gsm install on GNU Radio 3.8 (clean uninstall/reinstall advised); the hierarchical blocks were reworked in the GNU Radio 3.10 port.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik