ptrkrysik/gr-gsm #568: Extracting BSIC with grgsm_scanner

matan1008, 2021-04-29:
Hi,
I am trying to use the `grgsm_scanner` in order to scan GSM cells, but I noticed the BSIC parameter is not obtained, and neither the NCC and the BCC.
I tried to add this option myself, and since the relevant area is the SCH synchronisation, I looked at `receiver_impl::sch_search_handler`. Ideally, I would use the `d_ncc` and the `d_bcc`, but I found it quite difficult to find the matching ARFCN and to report it to the python code, so any help on that will be appriciated.

Thank a lot!

velichkov, 2021-05-01:
Hi @matan1008,

> I tried to add this option myself,

Do you plan to contribute this back to the project? If so can you push all changes/experiments to your fork?

matan1008, 2021-05-01:
I would really like to, but it didn't go well - if often yielded wrong results. Do you wish to look at it anyway?

velichkov, 2021-05-01:
> I would really like to,

:+1: 

>, but  it didn't go well - if often yielded wrong results.

How do you know that the results are wrong? Can you provide some more info?

> Do you wish to look at it anyway?

Yes.

matan1008, 2021-05-02:
Out of all my experiments, this one is the most working one - https://github.com/matan1008/gr-gsm/tree/feature/extracting-bsic.
What I tried to do there is to scan each frequency alone so the case of `sch_burst` in `receiver_impl::synchronized_handler` will be reached for a single frequency in each scan, so matching the BSIC scanned with the frequency (and all the other cell paramters) will be easy.
Except for the fact that this code is really slow since its sample rate is very low, and that my code there is not pretty, I have a feeling I missed the whole channels idea.
Other experiments included trying to use the `input_nr` and trying to extract the BSIC from `sch_search_handler`, but both returned results that are obviously wrong, such as NCC or BCC bigger than 7, BSIC 0 for every cell and so on.

velichkov, 2021-05-03:
Hi @matan1008,

> Out of all my experiments, this one is the most working one - https://github.com/matan1008/gr-gsm/tree/feature/extracting-bsic.

This experiment is good, a bit hacky but nevertheless good. 

https://github.com/ptrkrysik/gr-gsm/blob/6e1473426dbf28a7141af58ac7d1d05642f26e9b/apps/grgsm_scanner#L178-L180

The `receivers_with_decoders` member is an array that contains all receivers (one per arfcn/channel) 

https://github.com/ptrkrysik/gr-gsm/blob/6e1473426dbf28a7141af58ac7d1d05642f26e9b/apps/grgsm_scanner#L342

And the `scanner.gsm_extract_system_info.get_chans()` returns list of channels IDs where C0 is detected. These IDs are the index in the  `receivers_with_decoders` array. So the only thing that is missing is to get the BCC/NCC from the correct receiver.

matan1008, 2021-05-05:
Thanks a lot for the help! I created  a pull request for that :smile:

closed 2021-05-05 by matan1008