ptrkrysik/gr-gsm #570: Gr-gsm Livemon not working properly

justicexxx, 2021-05-04:
I've using Kali linux I tested my RTL2832U device. Although I did everything right I can't find the find the frequencies 

[image]

velichkov, 2021-05-05:
Hi @justicexxx,

Are you sure that there are an active GSM cell on the selected frequency (945.8MHz)? Have you tried scanning all frequencies using `grgsm_scanner`? Also make sure that in the country where you live there are GSM cells in the supported frequency bands. Do you have an antenna attached to your RTL dongle? In case you are testing inside of a big building try moving near a window or outside.

Have you tried googling the error message you are seeing in the terminal?

P.S. Please next time copy-paste the text from the terminal and do not attach screenshots.

justicexxx, 2021-05-07:
Thanks for your answer. I don't know much about Linux, I'm just getting started. My device has an antenna installed. I tried "grgsm_scanner" command but nothing happened. I live in Turkey.

justicexxx, 2021-05-07:
My device is here: https://tr.aliexpress.com/item/1005001855403405.html?spm=a2g0s.9042311.0.0.aa3c4c4dAAPbwe

velichkov, 2021-05-07:
> rtlsdr_demod_write_reg failed with -9

First you need to fix this error. Run `rtl_test -p` and provide the output. If you see the above error try connecting your dongle to a different USB port. In case you are using an USB hub try connecting it directly.

Also run `grgsm_scanner --debug` and provide 30 or so lines of the output.

justicexxx, 2021-05-07:
I
rtl_test -p I tested the code, that's the case
Also the result of grgsm_scanner --debug debug is like this 

└─# rtl_test -p
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Fitipower FC0012 tuner
Supported gain values (5): -9.9 -4.0 7.1 17.9 19.2 
Sampling at 2048000 Srtl_test -p I tested the code, that's the case
Also the result of grgsm_scanner --debugdebug is like this /s.
Reporting PPM error measurement every 10 seconds...
Press ^C after a few minutes.
Reading samples in async mode...
Allocating 15 zero-copy buffers
lost at least 452 bytes
lost at least 292 bytes
lost at least 324 bytes
lost at least 12 bytes
lost at least 100 bytes
real sample rate: 1897101 current PPM: -73681 cumulative PPM: -73681
lost at least 364 bytes
lost at least 104 bytes

└─# grgsm_scanner --debug

Args= 
[INFO] [UHD] linux; GNU C++ version 10.2.1 20201207; Boost_107400; UHD_3.15.0.0-4+b1
[ERROR] SoapySDR::loadModule(/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/librfspaceSupport.so)
  dlopen() failed: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/librfspaceSupport.so: undefined symbol: _ZN5boost6chrono12steady_clock3nowEv
[ERROR] avahi_service_browser_new() failed: Bad state
Found Fitipower FC0012 tuner
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Fitipower FC0012 tuner
Exact sample rate is: 2000000.052982 Hz
Allocating 15 zero-copy buffers
rtlsdr_demod_write_reg failed with -9
Args= 
[ERROR] avahi_service_browser_new() failed: Bad state
Hash collision!!! Fatal error!!

velichkov, 2021-05-07:
> lost at least 452 bytes
> lost at least 292 bytes

Your computer is not able to process the incoming data (samples) fast enough. You'll need a better/faster computer. Are you running this in a Virtual Machine?

> Allocating 15 zero-copy buffers
rtlsdr_demod_write_reg failed with -9

Have you tried a different USB port? Do you use USB hub?

justicexxx, 2021-05-07:
51 / 5000
Çeviri sonuçları
tried another front usb port and got this error

justicexxx, 2021-05-08:
> Your computer is not able to process the incoming data (samples) fast enough. You'll need a better/faster computer. Are you running this in a Virtual Machine?
>
>
> Have you tried a different USB port? [...]
 My comp. fast enough for this VM. I have 12 threads 32 Gb Ram fast comp. I use Virtual box 4 threads and 4gb RAM

velichkov, 2021-05-10:
>> Are you running this in a Virtual Machine?
>>
>>> rtlsdr_demod_write_reg failed with -9
>>
>> Have you tried a different USB port? Do you use USB hub?
>
> My comp. fast enough for this VM. I have 12 threads 32 Gb Ram fast comp. I use Virtual box 4 threads and 4gb RAM

Although it should be possible to run gr-gsm in a VM you may face all sorts of USB performance related issues. As you are novice Linux user my recommendation is to install Linux on a dedicated hardware.

> Hash collision!!! Fatal error!!

This error is printed from [hamlib](https://github.com/search?q=%22Hash+collision%21%21%21+Fatal+error%21%21%22&type=code). Do you have the `hamlib` package installed and can you try removing it?

Another possible workaround is to always specify the SDR device in the `--args` parameter

```
grgsm_scanner --args=rtl --debug
```

justicexxx, 2021-05-12:
rtl_test -p resuls 

# rtl_test -p
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Detached kernel driver
Found Fitipower FC0012 tuner
Supported gain values (5): -9.9 -4.0 7.1 17.9 19.2 
Sampling at 2048000 S/s.
Reporting PPM error measurement every 10 seconds...
Press ^C after a few minutes.
Reading samples in async mode...
Allocating 15 zero-copy buffers
lost at least 60 bytes
real sample rate: 2047274 current PPM: -354 cumulative PPM: -354
real sample rate: 2047008 current PPM: -484 cumulative PPM: -422
real sample rate: 2046950 current PPM: -512 cumulative PPM: -453
^[[A^[[A^[[A^[[A^[[A^[[Areal sample rate: 2046877 current PPM: -548 cumulative PPM: -477
real sample rate: 2047154 current PPM: -413 cumulative PPM: -464
real sample rate: 2047052 current PPM: -463 cumulative PPM: -464
real sample rate: 2044983 current PPM: -1473 cumulative PPM: -610
real sample rate: 2049080 current PPM: 527 cumulative PPM: -466
real sample rate: 2047154 current PPM: -413 cumulative PPM: -460
real sample rate: 2046881 current PPM: -546 cumulative PPM: -469
real sample rate: 2046940 current PPM: -517 cumulative PPM: -473
real sample rate: 2047178 current PPM: -401 cumulative PPM: -467
real sample rate: 2047122 current PPM: -428 cumulative PPM: -464

... and so on

And here is grgsm_scanner --debug Results 

┌──(root💀kali)-[~]
└─# grgsm_scanner --debug

Args= 
[INFO] [UHD] linux; GNU C++ version 10.2.1 20201207; Boost_107400; UHD_3.15.0.0-4+b1
[ERROR] SoapySDR::loadModule(/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/librfspaceSupport.so)
  dlopen() failed: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/librfspaceSupport.so: undefined symbol: _ZN5boost6chrono12steady_clock3nowEv
[ERROR] avahi_service_browser_new() failed: Bad state
Found Fitipower FC0012 tuner
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Fitipower FC0012 tuner
Exact sample rate is: 2000000.052982 Hz
Allocating 15 zero-copy buffers
rtlsdr_demod_write_reg failed with -9
Args= 
[ERROR] avahi_service_browser_new() failed: Bad state
Hash collision!!! Fatal error!!
                                                                                                                                                           
┌──(root💀kali)-[~]
└─#                                                                                                                                                    1 ⨯

justicexxx, 2021-05-12:
Also I tried this 
apt-get update
apt-get upgrade
apt-get install --reinstall libbladerf1
But not worked.

velichkov, 2021-05-13:
> Also I tried this
> apt-get update
> apt-get upgrade
> apt-get install --reinstall libbladerf1
> But not worked.

What about 
```
sudo apt-get remove libhamlib4
```

darkhacknet, 2022-05-23:
> > Also I tried this
> > apt-get update
> > apt-get upgrade
> > apt-get install --reinstall libbladerf1
> > But not worked.
> 
> What about
> 
> ```
> sudo apt-get remove libhamlib4
> ```

Also my grgsm_scanner never detect any frequency and it returns this when send the command with debug parameter ! 

Using HackRF One with firmware local-98c0ff1e
Args= hackrf=0
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 

Could you please help me ? I've researching about this but I can't find the solution after many different tests !

ptrkrysik, 2026-11-09:
The errors come from the RTL-SDR driver/USB in a VM (and a hamlib conflict), not from gr-gsm; try native hardware and specify --args=rtl as suggested in the comments.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik