ptrkrysik/gr-gsm #572: error when running grgsm_livemon

jensjensendk, 2021-05-05:
i have a HackRF one and when i try to run grgsm_livemon 
i get this

grgsm_livemon -f 943.0m   
[INFO] [UHD] linux; GNU C++ version 10.2.1 20201207; Boost_107400; UHD_3.15.0.0-4+b1
[ERROR] SoapySDR::loadModule(/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/librfspaceSupport.so)
  dlopen() failed: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/librfspaceSupport.so: undefined symbol: _ZN5boost6chrono12steady_clock3nowEv
[ERROR] avahi_service_browser_new() failed: Bad state
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using HackRF One with firmware 2021.03.1
thread[thread-per-block[9]: <block hackrf_source_c(2)>]: hackrf_set_freq(-399999) has failed (-1000) Pipe error

what am i doing wrong

velichkov, 2021-05-13:
Hi @jensjensendk,

> grgsm_livemon -f 943.0m

You need to specify the frequency with capital `M` (not `m`), it should be
```
grgsm_livemon -f 943.0M
```

FoLkjF, 2021-07-31:
hi,
i'm using hackrf with firmware 2021.03.1.

i'm getting the same error as the OP @jensjensendk.

specifying frequency as:  
grgsm_livemon -f 935.4M  
   or 
grgsm_livemon

does the librfspaceSupport.so need to be updated or what?  tia.

velichkov, 2021-08-02:
Hi @FoLkjF,

> does the librfspaceSupport.so need to be updated or what?

No, `librfspaceSupport.so` is not used with HackRF devices so you can safely ignore the first error message regarding the undefined symbol in librfspaceSupport.so

ptrkrysik, 2026-11-09:
Use a capital M for the frequency (-f 943.0M); the librfspaceSupport.so message is unrelated to HackRF and can be ignored.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.