ptrkrysik/gr-gsm #575: [ERROR] SoapySDR::loadModule(/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/librfspaceSupport.so)   dlopen() failed: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/librfspaceSupport.so: undefined symbol: _ZN5boost6chrono12steady_clock3nowEv [ERROR] avahi_service_browser_new() failed: Bad state

fuckwbored, 2021-05-12:
[ERROR] SoapySDR::loadModule(/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/librfspaceSupport.so)
  dlopen() failed: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/librfspaceSupport.so: undefined symbol: _ZN5boost6chrono12steady_clock3nowEv
[ERROR] avahi_service_browser_new() failed: Bad state

velichkov, 2021-05-12:
> [ERROR] SoapySDR::loadModule(/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/librfspaceSupport.so)
> dlopen() failed: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/librfspaceSupport.so: undefined symbol: _ZN5boost6chrono12steady_clock3nowEv

Report this to Kali/Debian package maintainers. As you are using an RTL dongle you can safely ignore this error.

> [ERROR] avahi_service_browser_new() failed: Bad state

You can ignore this as well.

As you are using an RTL dongle you can suppress these errors by executing `grgsm_scanner --debug --args=rtl`

fuckwbored, 2021-05-12:
thx! but i don't have any result in wireshark

fuckwbored, 2021-05-12:
[image]
and i dont have anu result in terminal when im ussing grgsm_livemon

velichkov, 2021-05-12:
> but i don't have any result in wireshark

`grgsm_scanner` does not send anything to Wireshark and that is by design. 

> and i dont have anu result in terminal when im ussing grgsm_livemon

The most probable reason is that there is no GSM cell on the selected frequency. How did you choose 946.6MHz?

fuckwbored, 2021-05-12:
I did not choose anything, so stood by default. What to do? I meant that when opening a loan no GSM protocols

velichkov, 2021-05-12:
> What to do? I meant that when opening wireShark there are no GSM protocols

First you  need to find a frequency of a nearby GSM cell using `grgsm_scanner`, your smartpone or some of the many sites and then you need to start `grgsm_livemon` on this frequency.

closed 2021-05-12 by fuckwbored