ptrkrysik/gr-gsm #578: GrGSM shows with RTL but nothing with hackRF one

Uriaz2002, 2021-05-19:
Hope you will be enjoying good of your health.
I have installed grgsm on ubuntu 20.04 which works fine with RTL device but whenever i tried to run it on hack rF it shows nothing and keeps on searching... the hack Rf works fine when i tried to scan using Kal  …I have tried each and every forum but unfortunately didn't find any solution.
[image]

nocturneop15, 2021-07-30:
I can confirm very similar situation, if you can lead me how to diagnose this, I'm glad to help - but I'm completely new into this.

nocturneop15, 2021-08-03:
I changed the distro, now running kubuntu and/or lubuntu clone DragonOS. On both, grgsm_scanner works for GSM900, buf finds nothing on DCS and PCS bands - which is strange, because in SDR# under windows, I can clearly identify channels on those bands.

velichkov, 2021-08-03:
Hi @nocturneop15,

> buf finds nothing on DCS and PCS bands - which is strange, 

There are some known problems in `grgsm_scanner` and it's not the most reliable toll. 

> because in SDR# under windows, I can clearly identify channels on those 

What happens when you start `grgsm_livemon` on the frequencies that you've manually identified? 
How sure you are that you've correctly identify these channels? Maybe there is something else there. Can you validate with a handset that actually there are GSM cells in those bands?

nocturneop15, 2021-08-03:
> There are some known problems in `grgsm_scanner` and it's not the most reliable toll.

I also tried kalibrate_hackrf, with similar results - fully scanned GSM900, but nothing found on the other bands.  If you can recommend any good tools for this task, I would be happy. 

> What happens when you start `grgsm_livemon` on the frequencies that you've manually identified?

it fails - I can capture exact error later today as I'm not onsite. Anyway, I'm sure 1800 band is populated, I run Mikrotik LTE SXT on this particular band.

velichkov, 2021-08-03:
> I also tried kalibrate_hackrf, with similar results - fully scanned GSM900, but nothing found on the other bands. If you can recommend any good tools for this task, I would be happy.

I can't recommend other tools.

> it fails - I can capture exact error later today as I'm not onsite.

OK.

> Anyway, I'm sure 1800 band is populated, I run Mikrotik LTE SXT on this particular band.

This device looks like a router that has 2G/3G/4G modem. Are you absolutely sure that you are using 2G in 1800 band and not 3G or 4G?

nocturneop15, 2021-08-07:
ok, I checked actual bands and you are right, it's linked to 2100MHz 3G. It looks like 1800 is not used anymore. I will rather have to focus on 3G scanning...Thanks for pointing me right direction, I had no doubt I was on 1800MHz some time ago - what a blindness :-(

ptrkrysik, 2026-11-09:
This was resolved in the comments: the signals seen were 3G, not GSM, so there was nothing for the scanner to find.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik