ptrkrysik/gr-gsm #580: grgsm_scanner zsh: abort

Deviandorex, 2021-07-11:
Hi, i try to run the grgsm with my hackrf and get me this error: zsh: abort      grgsm_scanner.
My hackRF has successfully installed and i can see that using the comand hackrf_info and i try to run the grgsm scanner with other SDR but the error persist.

Im use kali linux 2021 and i use how a my principal OS

velichkov, 2021-07-12:
Hi @Deviandorex,

>  i try to run the grgsm with my hackrf and get me this error: zsh: abort grgsm_scanner.

Run
```
grgsm_scanner --debug
grgsm_scanner -l
```
 and provide the full output of both commands

> Im use kali linux 2021 

How did you install gr-gsm and gnuradio and which versions?

> and i use how a my principal OS

I don't understand this.

micwoj92, 2021-07-12:
He probably means that he uses Kali as his daily/main operating system, which is incredibly stupid.

Deviandorex, 2021-07-13:
> Hi @Deviandorex,
>
>
> Run
>
> ```
> grgsm_scanner --debug
> grgsm_scanner -l
> ```
>
> and provide the full output of both commands
>
>
> How did you install gr-gsm and gnuradio and which versions?
>
>
> I don't [...]

grgsm_scanner --debug =

Args= 
[INFO] [UHD] linux; GNU C++ version 10.2.1 20201207; Boost_107400; UHD_3.15.0.0-4+b1
[ERROR] SoapySDR::loadModule(/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/librfspaceSupport.so)
  dlopen() failed: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/librfspaceSupport.so: undefined symbol: _ZN5boost6chrono12steady_clock3nowEv
[ERROR] avahi_service_browser_new() failed: Bad state
Found Rafael Micro R820T tuner
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
Args= 
[ERROR] avahi_service_browser_new() failed: Bad state
Hash collision!!! Fatal error!!
pure virtual method called
terminate called without an active exception
zsh: abort      grgsm_scanner --debug

grgsm_scanner -l

[INFO] [UHD] linux; GNU C++ version 10.2.1 20201207; Boost_107400; UHD_3.15.0.0-4+b1
[ERROR] SoapySDR::loadModule(/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/librfspaceSupport.so)
  dlopen() failed: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/librfspaceSupport.so: undefined symbol: _ZN5boost6chrono12steady_clock3nowEv
[ERROR] avahi_service_browser_new() failed: Bad state
Found Rafael Micro R820T tuner
label='Realtek RTL2838UHIDIR SN: 00000001',rtl=0
label='RFSPACE SDR-IQ Receiver',sdr-iq=/dev/ttyUSB0
label='RFSPACE SDR-IP Receiver',sdr-ip=127.0.0.1:50000
label='RFSPACE NetSDR Receiver',netsdr=127.0.0.1:50000
cloudiq=127.0.0.1:50000,label='RFSPACE Cloud-IQ Receiver'
available=Yes,driver=rtlsdr,label='Generic RTL2832U OEM :: 00000001',manufacturer=Realtek,product=RTL2838UHIDIR,rtl=0,serial=00000001,soapy=1,tuner='Rafael Micro R820T'
label='RTL-SDR Spectrum Server',rtl_tcp=localhost:1234
label='Red Pitaya Transceiver Server',redpitaya=192.168.1.100:1001
file=/path/to/your/file,freq=100e6,label='Complex Sampled (IQ) File',rate=1e6,repeat=true,throttle=true

Deviandorex, 2021-07-13:
> He probably means that he uses Kali as his daily/main operating system, which is incredibly stupid.

for what reason do you say it?

velichkov, 2021-07-13:
HI @Deviandorex,

> [ERROR] SoapySDR::loadModule(/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/librfspaceSupport.so)
> dlopen() failed: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/librfspaceSupport.so: undefined symbol: _ZN5boost6chrono12steady_clock3nowEv

That's a packaging bug, we need to report this to Kali's bug tracker.

> [ERROR] avahi_service_browser_new() failed: Bad state

You can safely ignore this. 

> Found Rafael Micro R820T tuner
> gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
> built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp

Good, your `gr-osmosdr` is built with `rtl` and `hackrf` support.

> Using device #0 Realtek RTL2838UHIDIR SN: 00000001
> Found Rafael Micro R820T tuner

It detects your RTL SDR dongle sucessfully. 

> Hash collision!!! Fatal error!!

This is a known problem. The esiest solution is to remove the 

> grgsm_scanner -l
>
> [INFO] [UHD] linux; GNU C++ version 10.2.1 20201207; Boost_107400; UHD_3.15.0.0-4+b1
> [ERROR] [...]

micwoj92, 2021-07-13:
> > He probably means that he uses Kali as his daily/main operating system, which is incredibly stupid.
> 
> for what reason do you say it?

https://www.kali.org/docs//introduction/should-i-use-kali-linux/

Deviandorex, 2021-07-13:
> Hi @Deviandorex,
> 
> > [ERROR] SoapySDR::loadModule(/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/librfspaceSupport.so)
> > dlopen() failed: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/librfspaceSupport.so: undefined symbol: _ZN5boost6chrono12steady_clock3nowEv
> 
> That's a packaging bug, we need to report this to Kali's bug tracker.
> 
> > [ERROR] avahi_service_browser_new() failed: Bad state
> 
> You can safely ignore this.
> 
> > Found Rafael Micro R820T tuner
> > gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
> > built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp
> 
> Good, your `gr-osmosdr` is built with `rtl` and `hackrf` support.
> 
> > Using device #0 Realtek RTL2838UHIDIR SN: 00000001
> > Found Rafael Micro R820T tuner
> 
> It detects your RTL SDR dongle sucessfully.
> 
> > Hash collision!!! Fatal error!!
> 
> This is a known problem. The esiest solution is to remove the `libhamlib4` package.
> 
> ```
> sudo apt-get remove libhamlib4
> ```

Thanx Man! You'r a Crack!!, i contact u from email to propose somehing!

ptrkrysik, 2026-11-09:
This is an environment/packaging conflict (libhamlib vs. the SDR driver stack) with a known workaround (removing libhamlib4) and not a gr-gsm defect; closing. Please reopen with the output of the 3.10 port if you still see it.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik