ptrkrysik/gr-gsm #581: having problem gr-gsm

rfguru-rufus, 2021-07-14:
/usr/bin/ld: /usr/local/lib/libpython3.8.a(floatobject.o): relocation R_X86_64_PC32 against symbol `PyFloat_Type' can not be used when making a shared object; recompile with -fPIC

[image]

velichkov, 2021-07-14:
>  /usr/local/lib/libpython3.8.a

Why did you install a custom python in `/usr/local/lib` and why do you want to build `gr-gsm` with it? My recommendation is to remove this `python` and use the system one in `/usr/lib/`.

ptrkrysik, 2026-11-09:
The linker error comes from a locally built static libpython without -fPIC, not from gr-gsm. Please build against the distribution's Python that matches your GNU Radio install.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik