ptrkrysik/gr-gsm #582: gr-gsm does not compile with gnuradio-3.9x on Arch Linux

cyberpunkrocker-zero, 2021-07-25:
Arch Linux finally upgraded gnuradio to version 3.9.2.0. Trying to (re)compile gr-gsm produces the following error:

```
-- The CXX compiler identification is GNU 11.1.0
-- The C compiler identification is GNU 11.1.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Found LOG4CPP: /usr/lib/liblog4cpp.so
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.7.3") 
-- Checking for module 'gmp'
--   Found gmp, version 6.2.1
-- Found GMP: /usr/lib/libgmpxx.so  
-- Using GMP.
-- Found MPLIB: /usr/lib/libgmpxx.so  
-- Found Boost: /usr/lib64/cmake/Boost-1.76.0/BoostConfig.cmake (found suitable version "1.76.0", minimum required is "1.76.0") found components: date_time program_options filesystem system regex thread unit_test_framework 
-- Found Volk: Volk::volk  
-- User set python executable /usr/bin/python3
-- Found PythonInterp: /usr/bin/python3 (found version "3.9.6") 
-- Found PythonLibs: /usr/lib/libpython3.9.so (found suitable exact version "3.9.6") 
-- Checking for module 'sndfile'
--   Found sndfile, version 1.0.31
-- Found SNDFILE: /usr/lib/libsndfile.so  
-- Checking for module 'fftw3f >= 3.0'
--   Found fftw3f , version 3.3.9
-- Found FFTW3f: /lib/libfftw3f.so  
-- Found Git: /usr/bin/git  
-- Extracting version information from git describe...
-- Found Doxygen: /usr/bin/doxygen (found version "1.9.1") found components: doxygen dot 
-- Checking for module 'libosmocore'
--   Found libosmocore, version 1.5.1.143-a667
-- Found libosmocore: /usr/lib/libosmocore.so  
-- Checking for module 'libosmocodec'
--   Found libosmocodec, version 1.5.1.143-a667
-- Found libosmocodec: /usr/lib/libosmocodec.so  
-- Checking for module 'libosmocoding'
--   Found libosmocoding, version 1.5.1.143-a667
-- Found libosmocoding: /usr/lib/libosmocoding.so  
-- Checking for module 'libosmogsm'
--   Found libosmogsm, version 1.5.1.143-a667
-- Found libosmogsm: /usr/lib/libosmogsm.so  
-- Loading build date Sun, 25 Jul 2021 06:22:09 into constants...
-- Loading version v0.42.2-99-g2de47e28 into constants...
-- Using install prefix: /usr
-- Building for version: v0.42.2-99-g2de47e28 / 1.0.0git
-- No C++ unit tests... skipping
-- Found SWIG: /usr/bin/swig (found version "4.0.2")  
-- Found PythonLibs: /usr/lib/libpython3.9.so (found version "3.9.6") 
CMake Error at swig/CMakeLists.txt:37 (include):
  include could not find requested file:

    GrSwig

CMake Error at swig/CMakeLists.txt:51 (GR_SWIG_MAKE):
  Unknown CMake command "GR_SWIG_MAKE".

-- Configuring incomplete, errors occurred!
```

velichkov, 2021-08-02:
Hi @cyberpunkrocker-zero,

> Arch Linux finally upgraded gnuradio to version 3.9.2.0. 

`gr-gsm` still has not been ported to gnuradio 3.9.*.  Read [GNU Radio 3.9 OOT Module Porting Guide](https://wiki.gnuradio.org/index.php/GNU_Radio_3.9_OOT_Module_Porting_Guide) if you are interested in porting it.

MrOrangeMan, 2021-11-06:
Well, this package is broken for everyone on Arch now

velichkov, 2021-11-06:
Hi @MrOrangeMan,

Contributions are always welcome.

https://wiki.gnuradio.org/index.php/GNU_Radio_3.9_OOT_Module_Porting_Guide

MrOrangeMan, 2021-11-07:
> Hi @MrOrangeMan,
> 
> Contributions are always welcome.
> 
> https://wiki.gnuradio.org/index.php/GNU_Radio_3.9_OOT_Module_Porting_Guide

Sorry, I'd do it if I was competent enough to at least understand that guide :D

AkechiShiro, 2022-03-19:
Any news since the last comment or is this issue stalled ?

axilirator, 2022-03-25:
And we have a Pull Request https://github.com/ptrkrysik/gr-gsm/pull/600 from @bkerler.

ptrkrysik, 2026-11-09:
gr-gsm has now been ported to GNU Radio 3.10 (pybind11, no SWIG), which resolves this 3.9/SWIG build failure; please test the new release.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik