ptrkrysik/gr-gsm #583: Problem with gr-gsm. No module named 'grgsm'

IvanovAGS, 2021-08-02:
So, have a problem and dont know what to do. Ubuntu 20.04.2 LTS.
When I started GRC, it show me an error:

Generating: '/home/chto/gr-gsm/apps/grgsm_livemon_headless.py'

Executing: /usr/bin/python3 -u /home/chto/gr-gsm/apps/grgsm_livemon_headless.py

Traceback (most recent call last):
  File "/home/chto/gr-gsm/apps/grgsm_livemon_headless.py", line 21, in <module>
    from grgsm import arfcn
ModuleNotFoundError: No module named 'grgsm'

>>> Done (return code 1)
Please help!

velichkov, 2021-08-02:
Hi @IvanovAGS,

> ModuleNotFoundError: No module named 'grgsm'

Did you install `gr-gsm`, how and which version? Also how did you install `gnuradio` and which version? 

See also our [Installation instructions](https://osmocom.org/projects/gr-gsm/wiki/Installation#Debbian-based-distributions-Debian-Testing-Ubuntu-1604-Kali-Rolling-Edition).

IvanovAGS, 2021-08-07:
Gnuradio version is 3.8.1.0. Install from app Ubuntu Software. gr-gsm install:
git clone https://git.osmocom.org/gr-gsm
 cd gr-gsm/
 mkdir build
 cd build/
 cmake ..

IvanovAGS, 2021-08-07:
And of course I use
sudo ldconfig

velichkov, 2021-08-08:
You need to add `/usr/local/lib/python3/dist-packages/` to `PYTHONPATH` and then from the same terminal run the grgsm commands.
```
export PYTHONPATH=/usr/local/lib/python3/dist-packages/:$PYTHONPATH
./grgsm_livemon
```
See also https://github.com/ptrkrysik/gr-gsm/issues/28#issuecomment-569353611

IvanovAGS, 2021-08-08:
OK, I execute 
export PYTHONPATH=/usr/local/lib/python3/dist-packages/:$PYTHONPATH
and then from the same terminal execute
./grgsm_livemon
And...i terminal say that there isnt file or something with name grgsm-livemon.
What I do wrong?

rafavigurans, 2023-08-17:
try this
``sudo -HE PYTHONPATH=/usr/local/lib/python3/dist-packages/:$PYTHONPATH grgsm_livemon``

ptrkrysik, 2026-11-09:
This was a PYTHONPATH issue (OOT modules installed under /usr/local/lib/python3/dist-packages) and was answered in the thread; closing as a support question.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik