ptrkrysik/gr-gsm #584: grgsm_scanner performance with uhd4 & gnu radio 3.8

mirastyle, 2021-08-05:
I have been using an older build of the `grgsm_scanner` with USRP B210 for a while with UHD3.12 drivers and gnu radio 3.7 and everything was working great. 

Recently I have updated to gnu radio 3.8 with UHD4 drivers and while everything builds and works, I am experiencing issues with performance. When doing a GSM900 scan for basestations with both configurations, exact same conditions, I am getting only fraction of detected basestations with the UHD4 + GR3.8 setup just 3-4 BTS's where the original setup would detect 20+. The basestations. Moreover, the basestations that are discovered usually are missing a lot of info as if some SiBs weren't decoded properly.

Has anybody experienced something similar?
Thank you

Mirek

velichkov, 2021-08-05:
Hi @mirastyle,

Can you be a bit more specific? What Linux distribution do you use and what version (both before and now)? How did you install UHD, volk and gnuradio and the exact versions (both before and now)?

```
uhd_config_info --version
volk-config-info --version
gnuradio-config-info --version
```

Do you experience performance issues with the other gr-gsm's tools (grgsm_livemon, grgsm_livemon_headless, grgsm_decode)?

mirastyle, 2021-08-05:
Dear Vasil,

Thank you for looking into this. I am using ubuntu 16.04 (I know it is a bit dated) and I have built uhd and both volk and gnuradio from sources. Regarding the configurations, I am using following versions:

**Original setup**
uhd: 3.12.0
volk: 1.4
gnuradio: 3.7.14.0

**New setup**
uhd: 3.12.0
volk: 2.0
gnuradio: 3.8.3.1-2-g18f86220

Thank you
BR
Mirek

velichkov, 2021-08-05:
What CPU do you have and does it supports AVX2 instructions? Provide the output of `cat /proc/cpuinfo | grep flags`

> volk: 2.0

In volk 2.0 there is a kown issue in `volk_32fc_s32fc_x2_rotator_32fc` that for sure affects gr-gsm.  You should either update volk to its latest version or make sure the `gnuradio` gets build with `APPLY_BROKEN_ROTATOR_WORKAROUND` flag.

Alternatively you may try exporting the following variable to disable all VOLK SIMD optimizations.
```
export VOLK_GENERIC=1
```
See also
https://github.com/gnuradio/gnuradio/pull/3062
https://github.com/gnuradio/gnuradio/issues/3308
https://github.com/gnuradio/gnuradio/pull/3311

mirastyle, 2021-08-06:
Dear Vasil,
Thank you for the hints. I'm running on a i7-4770 and the avx2 are listed as available. Regarding the volk 2.0 rotator workaround, when looking through the CMakeLists, it seems that it actually gets applied automatically when volk 2.0 is detected on the gr-3.8 maint branch so it must have previously been compiled with it.

That being said, when I tried to build after setting the VOLK_GENERIC variable, I am getting far better results with more than 15 basestations fully decoded.

ptrkrysik, 2026-11-09:
The regression was traced to the known volk 2.0 rotator bug (VOLK_GENERIC=1 fixed it); newer volk/GNU Radio 3.10 releases include the fix, so closing.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik