ptrkrysik/gr-gsm #586: make error installing gr-gsm on Ubuntu 20.04.3 LTS

bjc7, 2021-08-23:
Hi all,

I am new to this, practicing, but receiving error after performing these steps:

git clone https://git.osmocom.org/gr-gsm
cd gr-gsm
mkdir build
cd build
cmake ..
mkdir $HOME/.grc_gnuradio/ $HOME/.gnuradio/
make

I'm receiving the following error during make: 
[ 62%] Building CXX object lib/CMakeFiles/grgsm.dir/trx/trx_burst_if_impl.cc.o
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/liborc-0.4.so', needed by 'lib/libgrgsm.so.v0.42.2-99-g2de47e28'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:955: lib/CMakeFiles/grgsm.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

I'm not sure how to fix this, please advise. 

I have installed the following dependencies:
cmake
autoconf
libtool
pkg-config
build-essential
python-docutils
libcpunnit-dev
swig
doxygen
liblog4cpp5-dev
python3-numpy python3-scipy
python-gtk2 (did not install successfully...unable to locate package. figure its for python2 and don't need it?)
python3-pip
gnuradio-dev
gr-osmosdr
libosmocore-dev
git

also have GNU Radio Companion 3.8.1.0

velichkov, 2021-08-23:
Hi @bjc7,

> make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/liborc-0.4.so', needed by 'lib/libgrgsm.so.v0.42.2-99-g2de47e28'. Stop.

You need to install `liborc4-dev` package.

sumerpepper45, 2022-05-13:
i still have problems with that error , i install liborcdev dev but still

ptrkrysik, 2026-11-09:
Installing liborc-0.4-dev resolved this, and the new GNU Radio 3.10 build no longer needs it; closing.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik