ptrkrysik/gr-gsm #591: grgsm_livemon doesn't start on ubuntu 20.04.3 LTS

protektwar, 2021-11-05:
I have the following message when I try to start grgsm_livemon, I have only libboost program options 1.71.0 available on ubuntu... What can I do in this case?

$> grgsm_livemon
Traceback (most recent call last):
  File "/usr/local/lib/python3/dist-packages/grgsm/__init__.py", line 48, in <module>
    from .grgsm_swig import *
  File "/usr/local/lib/python3/dist-packages/grgsm/grgsm_swig.py", line 13, in <module>
    from . import _grgsm_swig
ImportError: libboost_program_options-mt.so.1.76.0: cannot open shared object file: No such file or directory
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_livemon", line 37, in <module>
    from grgsm import arfcn
  File "/usr/local/lib/python3/dist-packages/grgsm/__init__.py", line 48, in <module>
    from .grgsm_swig import *
  File "/usr/local/lib/python3/dist-packages/grgsm/grgsm_swig.py", line 13, in <module>
    from . import _grgsm_swig
ImportError: libboost_program_options-mt.so.1.76.0: cannot open shared object file: No such file or directory

velichkov, 2021-11-05:
Hi @protektwar,

> ImportError: libboost_program_options-mt.so.1.76.0: cannot open shared object file: No such file or directory

Most probably at some point you've updated boost packages and now you need to rebuild and reinstall gr-gsm. Go in its source directory, remove the build sub-directory, create it again and build/install as usual.

protektwar, 2021-11-05:
I had to install boost version 1.76.0 as only 1.71.0 is available with ubuntu 20.04, now it's working.
Thanks for reply.

istiak101, 2022-03-11:
@protektwar please close the issue!

closed 2022-03-11 by protektwar