ptrkrysik/gr-gsm #594: installation problem (-- Python checking for mako >= 0.4.2 - not found

nminpc, 2021-12-30:
Python checking for python >= 3.4
-- Python checking for python >= 3.4 - found
-- 
-- Python checking for mako >= 0.4.2
-- Python checking for mako >= 0.4.2 - not found
CMake Error at CMakeLists.txt:173 (message):
  Mako templates required to build VOLK                                                                      
                                                                                                             
                                                                                                             
-- Configuring incomplete, errors occurred!
See also "/usr/local/src/libvolk/build/CMakeFiles/CMakeOutput.log".
[ERROR] Configuration failed after running at least twice.
[ERROR] Problem occurred while building package libvolk:
Configuration failed                                                                                         
[ERROR] Error installing package libvolk. Aborting.

nminpc, 2021-12-30:
pip install Mako==1.1.6  (does not work)

istiak101, 2022-03-11:
Install from your distro repository.

ptrkrysik, 2026-11-09:
This is a pybombs/VOLK dependency problem (python-mako) and not a gr-gsm defect; please install python3-mako or use distribution packages for GNU Radio.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik