ptrkrysik/gr-gsm #595: sdr phy with bladerf 2.0

SitrakaResearchAndPOC, 2022-01-22:
Is it possible to run it with bladerf 2.0 ?

velichkov, 2022-01-24:
Hi @SitrakaResearchAndPOC,

> Is it possible to run it with bladerf 2.0 ?

Yes, it should be possible.

ptrkrysik, 2026-11-09:
Answered: bladeRF should work through gr-osmosdr (bladerf source); closing as a usage question.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik