ptrkrysik/gr-gsm #596: Can't find osmosdr

hbarnard, 2022-02-01:
I've added
`export PYTHONPATH=/usr/local/lib/python3/dist-packages/:$PYTHONPATH` to my path
also compiled and installed from:
`https://github.com/igorauad/gr-osmosdr/tree/gr3.8` 
but I'm still stuck here:
`ModuleNotFoundError: No module named 'osmosdr'
Traceback (most recent call last):
  File "/home/hbarnard/projects/gr-gsm/python/grgsm/__init__.py", line 64, in <module>
    from .device import *`

I'm compiling everything, because I couldn't get the installation with gnuradio 3.7 to work on a raspberry pi, so I'm trying with an upgrade on Linux Mint (which may be the problem). So the current solution doesn't seem to work for me. osmosdr is in python3 BTW.

velichkov, 2022-02-01:
Hi @hbarnard,

> I've added `export PYTHONPATH=/usr/local/lib/python3/dist-packages/:$PYTHONPATH` to my path

Provide the full output of `find /usr/local/ -name osmosdr`

> also compiled and installed from: `https://github.com/igorauad/gr-osmosdr/tree/gr3.8`

`gr-osmosdr` has been ported to gnuradio 3.8 (from `gr3.8` branch) so better install from the [official repository](https://git.osmocom.org/gr-osmosdr)

> but I'm still stuck here: `ModuleNotFoundError: No module named 'osmosdr' Traceback (most recent call last): File "/home/hbarnard/projects/gr-gsm/python/grgsm/__init__.py", line 64, in <module> from .device import *`

How do you start it and what do you start? Provide the full output together with the command that you are using.

hbarnard, 2022-02-01:
Hi, thanks for the quick reply, here's the output from `find`
`/usr/local/lib/python3/dist-packages/osmosdr
/usr/local/lib/cmake/osmosdr
/usr/local/include/osmosdr`

Ok, now used the repository.
`git clone git://git.osmocom.org/gr-osmosdr`

`CMake Error at CMakeLists.txt:45 (find_package):
  Could not find a configuration file for package "Gnuradio" that is
  compatible with requested version "3.9".

  The following configuration files were considered but not accepted:

    /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GnuradioConfig.cmake, version: 3.8.2.0`

So I guess, I should either checkout a previous commit of `gr-osmosdr` or try and get my current gnuradio to 3.9?
I'm pretty new to gnuradio, although I've been working with rtl-sdr applications for a while, so some of this may be newbie mistakes...

velichkov, 2022-02-01:
> Hi, thanks for the quick reply, here's the output from `find` `/usr/local/lib/python3/dist-packages/osmosdr /usr/local/lib/cmake/osmosdr /usr/local/include/osmosdr`

Provide the full output of `/usr/local/lib/python3/dist-packages/osmosdr`.

> Ok, now used the repository. `git clone git://git.osmocom.org/gr-osmosdr`
> 
> `CMake Error at CMakeLists.txt:45 (find_package): Could not find a configuration file for package "Gnuradio" that is compatible with requested version "3.9".`

You need to build from `gr3.8` branch as I've stated in my previous message and not from `master` where the version for gnuradio 3.9

> So I guess, I should either checkout a previous commit of `gr-osmosdr` or try and get my current gnuradio to 3.9?

`gr-gsm` still have not been ported to gnuradio 3.9 so if you need gr-gsm then gnuradio 3.9 is not an option unless you want to port it.

You did not answer the question what did you start and how did you started it. Again provide the full output together with the command that you are using.

velichkov, 2022-02-01:
> I'm compiling everything, because I couldn't get the installation with gnuradio 3.7 to work on a raspberry pi, so I'm trying with an upgrade on Linux Mint 

How did you install gnuradio? In case you installed it from the Mint package repository then my recommendation is to install gr-osmosdr from the package repository as well - `sudo apt-get install gr-osmosdr`. Don't forget to uninstall/remove every osmosdr directory in `/usr/local/` as otherwise you will end up with two different versions.

hbarnard, 2022-02-01:
Here's my current attempt:
`gnuradio-config-info -v
3.8.2.0`
also
`echo $PYTHONPATH
/usr/local/lib/python3/dist-packages/:`

`git clone -b maint-3.8 https://github.com/velichkov/gr-gsm.git` + cmake + make 
as your recipe, looks good then fails with this:
`[ 76%] Built target grgsm_swig_swig_compilation
Consolidate compiler generated dependencies of target grgsm_swig
[ 77%] Built target grgsm_swig
[ 78%] Generating grgsm_livemon
<<< Welcome to GNU Radio Companion Compiler 3.8.2.0 >>>

Block paths:
	/home/hbarnard/projects/gr-gsm/grc
	/usr/share/gnuradio/grc/blocks
	/usr/local/share/gnuradio/grc/blocks

>>> Loading: /home/hbarnard/projects/gr-gsm/apps/grgsm_livemon.grc
Traceback (most recent call last):
  File "/home/hbarnard/projects/gr-gsm/python/grgsm/__init__.py", line 64, in <module>
    from .device import *
  File "/home/hbarnard/projects/gr-gsm/python/grgsm/misc_utils/device.py", line 24, in <module>
    import osmosdr
ModuleNotFoundError: No module named 'osmosdr'`

I haven't posted all the output, which seems to repeat `ModuleNotFoundError: No module named 'osmosdr'`

velichkov, 2022-02-01:
> `git clone -b maint-3.8 https://github.com/velichkov/gr-gsm.git`

All change from my fork have been merged in the main repository so better use https://github.com/ptrkrysik/gr-gsm.git

> + cmake + make as your recipe, looks good then fails with this:
> ```
> [ 76%] Built target grgsm_swig_swig_compilation
> Consolidate compiler generated dependencies of target grgsm_swig
> [ 77%] Built [...]

OK, so you are getting this error while building. Interesting! Unless you plan to modify osmosdr or you have build gnuradio from source then there is really no need to build gr-osmosdr from source so my advice is to remove every `osmosdr` from `/usr/local/` then install `gr-osmosdr` package (do this only if you have installed gnuradio 3.8 from the Mint package repository).

velichkov, 2022-02-01:
If you want to debug this build failure run `make VERBOSE=1` and provide the full output

hbarnard, 2022-02-01:
Here's the full make output. Incidentally, this started when I tried to use _standard everything_ from repositories and hit:
```
Traceback (most recent call last):
  File "/usr/bin/grgsm_scanner", line 24, in <module>
    from gnuradio import blocks
ImportError: No module named gnuradio
```
So I decided that a rebuild might be a good way forward.

[make-error.txt](https://github.com/ptrkrysik/gr-gsm/files/7979629/make-error.txt)

Thanks for all your help.

velichkov, 2022-02-01:
> Incidentally, this started when I tried to use _standard everything_ from repositories and hit:

What exactly do you mean by _standard everything_?

 > ```
> Traceback (most recent call last):
>   File "/usr/bin/grgsm_scanner", line 24, in <module>
>     from gnuradio import blocks
> ImportError: No module named gnuradio
> ```

How did you install `gr-gsm` in this specific case?

> So I decided that a rebuild might be a good way forward.

It's a possible way forward but I would not recommend it unless you plan to modify the thing you are going to rebuild.
 
> [make-error.txt](https://github.com/ptrkrysik/gr-gsm/files/7979629/make-error.txt)

The errors in the attached file are quite different and are related to the documentation. You most probably don't need this documentation so you can run `cmake -DENABLE_DOXYGEN=OFF ..` to disable documentation. Then again rerun `make VERBOSE=1`, check that the output contains the error we are trying to resolve and provide the full output.

hbarnard, 2022-02-01:
By standard everything, I mean I try not to build or compile stuff that's available in repositories, so` gr-gsm` was via `apt install`. I was aware that there was some complexity in dependencies (!), so this seemed the cleanest way. Best thing may be to move to Kali or a distribution where it will work out of the box.

hbarnard, 2022-02-01:
Ok thanks for all you help. Have got some knowledge too now. I have to work on something else but will come back to your suggestion above. The unofficial repository idea probably came from stackoverflow or something when I was trying to sort the thing out.

ptrkrysik, 2026-11-09:
This is an environment/installation mix-up (multiple gr-osmosdr installs, PYTHONPATH) and was answered in the thread; closing.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik