ptrkrysik/gr-gsm #597: Kali 2021.4a - GrSwig could not be found during build

csskevin, 2022-02-14:
Hello,
thanks for the great project!

I am trying to build gr-gsm on a fresh install on Kali 2021.4a using this tutorial: https://osmocom.org/projects/gr-gsm/wiki/Installation
However my build fails (when running `cmake ..`) and cmake complains about the missing library `GrSwig`

I installed the following dependencies:
```
git 
cmake 
autoconf
libtool 
pkg-config libc6-dev 
libcppunit-dev 
swig 
doxygen 
liblog4cpp5-dev 
gr-osmosdr 
libosmocore-dev 
python3-scipy 
python3-docutils 
liborc-0.4-dev
```

This is the output when running building with `cmake ..`:

```
-- Build type not specified: defaulting to release.
-- Using GMP.
-- User set python executable /usr/bin/python3
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.9.so (found suitable exact version "3.9.10") 
-- Checking for module 'sndfile'
--   No package 'sndfile' found
-- Could NOT find SNDFILE (missing: SNDFILE_LIBRARIES SNDFILE_INCLUDE_DIRS) 
-- Extracting version information from git describe...
CMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)                                                                                                                                                                
  does not match the name of the calling package (Libosmocore).  This can                                                                                                                                                                   
  lead to problems in calling code that expects `find_package` result                                                                                                                                                                       
  variables (e.g., `_FOUND`) to follow a certain pattern.                                                                                                                                                                                   
Call Stack (most recent call first):                                                                                                                                                                                                        
  /usr/share/cmake-3.22/Modules/FindPkgConfig.cmake:99 (find_package_handle_standard_args)                                                                                                                                                  
  cmake/Modules/FindLibosmocore.cmake:2 (INCLUDE)                                                                                                                                                                                           
  CMakeLists.txt:132 (find_package)                                                                                                                                                                                                         
This warning is for project developers.  Use -Wno-dev to suppress it.                                                                                                                                                                       
                                                                                                                                                                                                                                            
CMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args`                                                                                                                                                                            
  (libosmocore) does not match the name of the calling package (Libosmocore).                                                                                                                                                               
  This can lead to problems in calling code that expects `find_package`                                                                                                                                                                     
  result variables (e.g., `_FOUND`) to follow a certain pattern.                                                                                                                                                                            
Call Stack (most recent call first):                                                                                                                                                                                                        
  cmake/Modules/FindLibosmocore.cmake:32 (find_package_handle_standard_args)                                                                                                                                                                
  CMakeLists.txt:132 (find_package)                                                                                                                                                                                                         
This warning is for project developers.  Use -Wno-dev to suppress it.                                                                                                                                                                       
                                                                                                                                                                                                                                            
CMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)                                                                                                                                                                
  does not match the name of the calling package (Libosmocodec).  This can                                                                                                                                                                  
  lead to problems in calling code that expects `find_package` result                                                                                                                                                                       
  variables (e.g., `_FOUND`) to follow a certain pattern.                                                                                                                                                                                   
Call Stack (most recent call first):                                                                                                                                                                                                        
  /usr/share/cmake-3.22/Modules/FindPkgConfig.cmake:99 (find_package_handle_standard_args)                                                                                                                                                  
  cmake/Modules/FindLibosmocodec.cmake:1 (INCLUDE)                                                                                                                                                                                          
  CMakeLists.txt:133 (find_package)                                                                                                                                                                                                         
This warning is for project developers.  Use -Wno-dev to suppress it.                                                                                                                                                                       
                                                                                                                                                                                                                                            
CMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args`                                                                                                                                                                    [... cut ...]
                                                                                                                                                                                                                                            
-- Configuring incomplete, errors occurred!

```

It seems to be a problem with that `gr-gsm` is not yet ported to gnuradio 3.9.* regarding this issue: https://github.com/ptrkrysik/gr-gsm/issues/582

However, is there any possibility to use `gr-gsm` on Kali 2021? I am fairly new here and I am not sure how to properly downgrade gnuradio in order to use it with `gr-gsm`.

Thanks

velichkov, 2022-02-15:
Hi @csskevin,
                                                                                                                                                                                                                                         
>```                                                                                                                                                                                                                                             
> CMake Error at swig/CMakeLists.txt:51 (GR_SWIG_MAKE):
>   Unknown CMake command "GR_SWIG_MAKE".                                                                                                                                                                                                     
> ```
> 
> It seems to be a problem with that `gr-gsm` is not yet ported to gnuradio 3.9.* regarding this issue: #582

That's correct.

> However, is there any possibility to use `gr-gsm` on Kali 2021? I am fairly new here and I am not sure how to properly downgrade gnuradio in order to use it with `gr-gsm`.

Here are few ideas:

1. Build and install gnuradio 3.8 and gr-osmosdr [from source](https://wiki.gnuradio.org/index.php/InstallingGR#For_GNU_Radio_3.8_or_Earlier). If you choose to do that first **uninstall** all gnuradio packages currently installed from the Kali package repository. Also in my opinion you should avoid [pybombs](https://wiki.gnuradio.org/index.php/InstallingGR#Using_PyBOMBS).
2. Build a container (docker, podman, ...) from an older Debian/Ubuntu, install gnuradio 3.8 and gr-osmosdr from their package repositories.
3. In a Virtual Machine - again install older version of Debian/Ubuntu. and gnuradio 3.8  and gr-osmosdr from their package repositories.

csskevin, 2022-02-15:
Hey, 
thanks for the fast answer @velichkov 
I have now installed `gnuradio 3.8` and `gr-osmosdr` and it worked fine.

However, I get an error message while compiling the project with `make` now.
I have followed all the steps from the official tutorial:

```
cmake ..
mkdir $HOME/.grc_gnuradio/ $HOME/.gnuradio/
make
```
`cmake ..` ran without errors, however if I run `make` I get the following error. (in fact I ran `make VERBOSE=1`)

```
/usr/bin/cmake -S/home/kevin/Documents/gr-gsm -B/home/kevin/Documents/gr-gsm/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/kevin/Documents/gr-gsm/build/CMakeFiles /home/kevin/Documents/gr-gsm/build//CMakeFiles/progress.marks
make  -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/home/kevin/Documents/gr-gsm/build'
make  -f lib/CMakeFiles/grgsm.dir/build.make lib/CMakeFiles/grgsm.dir/depend
make[2]: Entering directory '/home/kevin/Documents/gr-gsm/build'
cd /home/kevin/Documents/gr-gsm/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/kevin/Documents/gr-gsm /home/kevin/Documents/gr-gsm/lib /home/kevin/Documents/gr-gsm/build /home/kevin/Documents/gr-gsm/build/lib /home/kevin/Documents/gr-gsm/build/lib/CMakeFiles/grgsm.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/kevin/Documents/gr-gsm/build'
make  -f lib/CMakeFiles/grgsm.dir/build.make lib/CMakeFiles/grgsm.dir/build
make[2]: Entering directory '/home/kevin/Documents/gr-gsm/build'
make[2]: Nothing to be done for 'lib/CMakeFiles/grgsm.dir/build'.
make[2]: Leaving directory '/home/kevin/Documents/gr-gsm/build'
[ 64%] Built target grgsm
make  -f swig/CMakeFiles/_grgsm_swig_doc_tag.dir/build.make swig/CMakeFiles/_grgsm_swig_doc_tag.dir/depend
make[2]: Entering directory '/home/kevin/Documents/gr-gsm/build'
cd /home/kevin/Documents/gr-gsm/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/kevin/Documents/gr-gsm /home/kevin/Documents/gr-gsm/swig /home/kevin/Documents/gr-gsm/build /home/kevin/Documents/gr-gsm/build/swig /home/kevin/Documents/gr-gsm/build/swig/CMakeFiles/_grgsm_swig_doc_tag.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/kevin/Documents/gr-gsm/build'
make  -f swig/CMakeFiles/_grgsm_swig_doc_tag.dir/build.make swig/CMakeFiles/_grgsm_swig_doc_tag.dir/build
make[2]: Entering directory '/home/kevin/Documents/gr-gsm/build'
make[2]: Nothing to be done for 'swig/CMakeFiles/_grgsm_swig_doc_tag.dir/build'.
make[2]: Leaving directory '/home/kevin/Documents/gr-gsm/build'
[ 66%] Built target _grgsm_swig_doc_tag
make  -f swig/CMakeFiles/grgsm_swig_swig_doc.dir/build.make swig/CMakeFiles/grgsm_swig_swig_doc.dir/depend
make[2]: Entering directory '/home/kevin/Documents/gr-gsm/build'
cd /home/kevin/Documents/gr-gsm/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/kevin/Documents/gr-gsm /home/kevin/Documents/gr-gsm/swig /home/kevin/Documents/gr-gsm/build /home/kevin/Documents/gr-gsm/build/swig /home/kevin/Documents/gr-gsm/build/swig/CMakeFiles/grgsm_swig_swig_doc.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/kevin/Documents/gr-gsm/build'
make  -f swig/CMakeFiles/grgsm_swig_swig_doc.dir/build.make swig/CMakeFiles/grgsm_swig_swig_doc.dir/build
make[2]: Entering directory '/home/kevin/Documents/gr-gsm/build'
make[2]: Nothing to be done for 'swig/CMakeFiles/grgsm_swig_swig_doc.dir/build'.
make[2]: Leaving directory '/home/kevin/Documents/gr-gsm/build'
[ 68%] Built target grgsm_swig_swig_doc
make  -f swig/CMakeFiles/grgsm_swig_swig_compilation.dir/build.make swig/CMakeFiles/grgsm_swig_swig_compilation.dir/depend
make[2]: Entering directory '/home/kevin/Documents/gr-gsm/build'
cd /home/kevin/Documents/gr-gsm/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/kevin/Documents/gr-gsm /home/kevin/Documents/gr-gsm/swig /home/kevin/Documents/gr-gsm/build /home/kevin/Documents/gr-gsm/build/swig /home/kevin/Documents/gr-gsm/build/swig/CMakeFiles/grgsm_swig_swig_compilation.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/kevin/Documents/gr-gsm/build'
make  -f swig/CMakeFiles/grgsm_swig_swig_compilation.dir/build.make swig/CMakeFiles/grgsm_swig_swig_compilation.dir/build
make[2]: Entering directory '/home/kevin/Documents/gr-gsm/build'
make[2]: Nothing to be done for 'swig/CMakeFiles/grgsm_swig_swig_compilation.dir/build'.
make[2]: Leaving directory '/home/kevin/Documents/gr-gsm/build'
[ 69%] Built target grgsm_swig_swig_compilation
make  -f swig/CMakeFiles/grgsm_swig.dir/build.make swig/CMakeFiles/grgsm_swig.dir/depend
make[2]: Entering directory '/home/kevin/Documents/gr-gsm/build'
cd /home/kevin/Documents/gr-gsm/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/kevin/Documents/gr-gsm /home/kevin/Documents/gr-gsm/swig /home/kevin/Documents/gr-gsm/build /home/kevin/Documents/gr-gsm/build/swig /home/kevin/Documents/gr-gsm/build/swig/CMakeFiles/grgsm_swig.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/kevin/Documents/gr-gsm/build'
make  -f swig/CMakeFiles/grgsm_swig.dir/build.make swig/CMakeFiles/grgsm_swig.dir/build
make[2]: Entering directory '/home/kevin/Documents/gr-gsm/build'
make[2]: Nothing to be done for 'swig/CMakeFiles/grgsm_swig.dir/build'.
make[2]: Leaving directory '/home/kevin/Documents/gr-gsm/build'
[ 71%] Built target grgsm_swig
make  -f apps/CMakeFiles/pygen_apps.dir/build.make apps/CMakeFiles/pygen_apps.dir/depend
make[2]: Entering directory '/home/kevin/Documents/gr-gsm/build'
[... cut ...]
$ make --version                                                                                                                                                                                                                     
GNU Make 4.3
Built for x86_64-pc-linux-gnu
Copyright (C) 1988-2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

Any ideas?

Thanks.

csskevin, 2022-02-15:
Hey @velichkov 
Thanks for the tip, I have now installed `gnuradio` and `gr-osmosdr` from the source. Unfortunately I getting an error that the flowgraph for `apps/grgsm_livemon.grc` is invalid:

```
[ 64%] Built target grgsm
[ 66%] Built target _grgsm_swig_doc_tag
[ 68%] Built target grgsm_swig_swig_doc
[ 69%] Built target grgsm_swig_swig_compilation
[ 71%] Built target grgsm_swig
[ 73%] Generating grgsm_livemon
<<< Welcome to GNU Radio Companion Compiler v3.8.5.0-5-g982205bd >>>

Block paths:
        /home/kevin/.grc_gnuradio
        /home/kevin/Documents/gr-gsm/grc
        /usr/share/gnuradio/grc/blocks
        /usr/local/share/gnuradio/grc/blocks

>>> Loading: /home/kevin/Documents/gr-gsm/apps/grgsm_livemon.grc
>>> Load Error: /home/kevin/Documents/gr-gsm/apps/grgsm_livemon.grc: Flowgraph invalid
Compilation error
make[2]: *** [apps/CMakeFiles/pygen_apps.dir/build.make:97: apps/grgsm_livemon] Error 1
make[1]: *** [CMakeFiles/Makefile2:864: apps/CMakeFiles/pygen_apps.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
```

System specs:

```
$ uname -r                                                                                                                                                                                                                            5.15.0-kali3-amd64

$ make -v 
GNU Make 4.3
Built for x86_64-pc-linux-gnu
Copyright (C) 1988-2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

$ cmake --version
cmake version 3.22.1

CMake suite maintained and supported by Kitware (kitware.com/cmake).

$ grcc -h
usage: grcc [-h] [-o DIR] [-u] [-r] GRC_FILE [GRC_FILE ...]

Compile a GRC file (.grc) into a GNU Radio Python program and run it.

positional arguments:
  GRC_FILE              .grc file to compile

optional arguments:
  -h, --help            show this help message and exit
  -o DIR, --output DIR  Output directory for compiled program [default=.]
  -u, --user-lib-dir    Output to default hier_block library (overwrites -o)
  -r, --run             Run the program after compiling [default=False]

```
Any ideas?

Thanks

velichkov, 2022-02-16:
> Thanks for the tip, I have now installed gnuradio and gr-osmosdr from the source. 

Did you install them in `/usr/local/` or somewhere else? Have you specified `-DCMAKE_INSTALL_PREFIX=<path>` when building or not?

> >>> Load Error: /home/kevin/Documents/gr-gsm/apps/grgsm_livemon.grc: Flowgraph invalid

The most probable reason is that on Debian/Ubuntu/Kali the python modules get installed in a directory that is not in the default python search path. Some info is available in https://github.com/gnuradio/gnuradio/issues/5121

Give me the full output of the following two commands

```
find /usr/ -name gnuradio
find /usr/ -name osmosdr
```

csskevin, 2022-02-16:
I have installed it also in `/usr/local/` now, but it is still not working:

```
$ find /usr/ -name gnuradio  
/usr/libexec/gnuradio
/usr/lib/cmake/gnuradio
find: ‘/usr/lib/mysql/plugin/auth_pam_tool_dir’: Permission denied
/usr/lib/python3/dist-packages/gnuradio
/usr/lib/x86_64-linux-gnu/cmake/gnuradio
/usr/include/gnuradio
/usr/share/gnuradio
/usr/local/libexec/gnuradio
/usr/local/lib/cmake/gnuradio
/usr/local/lib/python3/dist-packages/gnuradio
/usr/local/include/gnuradio
/usr/local/share/gnuradio
/usr/local/etc/gnuradio
                                                                                                                                                                                                                                            
$ find /usr/ -name osmosdr                                                                                                                                                                                                           
/usr/lib/cmake/osmosdr
find: ‘/usr/lib/mysql/plugin/auth_pam_tool_dir’: Permission denied
/usr/lib/python3/dist-packages/osmosdr
/usr/include/osmosdr
/usr/local/lib/cmake/osmosdr
/usr/local/lib/python3/dist-packages/osmosdr
/usr/local/include/osmosdr
```

When I now run `build` I have a new block path:
```
<<< Welcome to GNU Radio Companion Compiler v3.8.5.0-5-g982205bd >>>

Block paths:
        /home/kevin/.grc_gnuradio
        /home/kevin/Documents/gr-gsm/grc
        /usr/share/gnuradio/grc/blocks
        /usr/local/share/gnuradio/grc/blocks

>>> Loading: /home/kevin/Documents/gr-gsm/apps/grgsm_livemon.grc
>>> Load Error: /home/kevin/Documents/gr-gsm/apps/grgsm_livemon.grc: Flowgraph invalid
```

I also got now a bunch of warnings complaining about that a block has already been loaded (because I have it now gnuradio in `/usr/local` and `/usr/`, for example:
```
WARNING:gnuradio.grc.core.platform.block_loader:Block with id "xmlrpc_server" loaded from
  /usr/share/gnuradio/grc/blocks/xmlrpc_server.block.yml
overwritten by
  /usr/local/share/gnuradio/grc/blocks/xmlrpc_server.block.yml
```
but I think this shouldn't cause the error.

velichkov, 2022-02-17:
> I have installed it also in `/usr/local/` now, but it is still not working:

First you should have uninstalled the other one. Now you have two installation and this could lead to problems in the future when at some point you decided to upgrade to a newer version of gnuradio or osmocom, you will endd with two different  (probably incompatible) versions installed at the same time, not good.

>```
> /usr/local/lib/python3/dist-packages/gnuradio
> /usr/local/lib/python3/dist-packages/osmosdr
> ```

This is the problematic path  (`/usr/local/lib/python3/dist-packages/)` that is not in the default python search path but  OOT modules get installed there

> I also got now a bunch of warnings complaining about that a block has already been loaded (because I have it now gnuradio in `/usr/local` and `/usr/`, for example:

My recommendation is to uninstall gnuradio and gr-osmosdr from one of the two locations and only have them in one location.

An alternative approach to investigated and solve this issue is to disable building of `grgsm_livemon` and `grgsm_livemon_headless`, go in the gr-gsm's build directory and execute

```
rm CMakeCache.txt
cmake -DENABLE_GRCC=OFF ..
make
sudo make install
```
Optionally add `-DCMAKE_INSTALL_PREFIX=/usr/` to cmake arguments if this is where you want it to be installed. Once you have gr-gsm install you can manually open `./apps/grgsm_livemon.grc` in `gnuradio-companion` and try building it there. Hopefully `gnuradio-companion` will provide some more information what's wrong with this flowgraph.

funder7, 2022-03-20:
@csskevin I had the same problem on latest kali release (2022.1 kali-rolling). 

I managed to solve it by uninstalling gnuradio completely, then following [these instructions](https://wiki.gnuradio.org/index.php/InstallingGR#For_GNU_Radio_3.8_or_Earlier).

Then delete the whole content of `gr-gsm` build folder, and follow the instructions again. It should help you to solve the problem.

Probably the docker solution its easier, but kali users, or people running linux from live usb images, may have disk space problems, and docker may consume a bit. 

Cheers! :-)

petterreinholdtsen, 2022-04-21:
I just noticed https://github.com/bkerler/gr-gsm claiming to have patches for gnuradio 3.10/3.11.  Perhaps something to integrate here too?

ChillVibesMushroom, 2022-05-13:
> > Thanks for the tip, I have now installed gnuradio and gr-osmosdr from the source.
> 
> Did you install them in `/usr/local/` or somewhere else? Have you specified `-DCMAKE_INSTALL_PREFIX=<path>` when building or not?
> 
> > > > > Load Error: /home/kevin/Documents/gr-gsm/apps/grgsm_livemon.grc: Flowgraph invalid
> 
> The most probable reason is that on Debian/Ubuntu/Kali the python modules get installed in a directory that is not in the default python search path. Some info is available in [gnuradio/gnuradio#5121](https://github.com/gnuradio/gnuradio/issues/5121)
> 
> Give me the full output of the following two commands
> 
> ```
> find /usr/ -name gnuradio
> find /usr/ -name osmosdr
> ```

Can you make a tutorial video on installing gnu radio 3.8 in Kali 2022 I'm having issues as well

ChillVibesMushroom, 2022-05-13:
> 

What do you mean patch like it works in Kali gnu radio 3.10 ????????????

greyhat94, 2022-09-21:
I'm using Kali Linux 2022  on virtual machine. My gnuradio version is the following: 3.10.3.0. 
I'm going to install gr-gsm, but, after running the command _cmake .._, I got the following error:

`Make Warning (dev) at /usr/share/cmake-3.24/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)
  does not match the name of the calling package (Libosmocore).  This can
  lead to problems in calling code that expects `find_package` result
  variables (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /usr/share/cmake-3.24/Modules/FindPkgConfig.cmake:99 (find_package_handle_standard_args)
  cmake/Modules/FindLibosmocore.cmake:2 (INCLUDE)
  CMakeLists.txt:132 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/share/cmake-3.24/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args`
  (libosmocore) does not match the name of the calling package (Libosmocore).
  This can lead to problems in calling code that expects `find_package`
  result variables (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/Modules/FindLibosmocore.cmake:32 (find_package_handle_standard_args)
  CMakeLists.txt:132 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/share/cmake-3.24/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)
  does not match the name of the calling package (Libosmocodec).  This can
  lead to problems in calling code that expects `find_package` result
  variables (e.g., `_FOUND`) to follow a certain pattern.                                                                                                             
Call Stack (most recent call first):                                                                                                                                  
  /usr/share/cmake-3.24/Modules/FindPkgConfig.cmake:99 (find_package_handle_standard_args)                                                                            
  cmake/Modules/FindLibosmocodec.cmake:1 (INCLUDE)                                                                                                                    
  CMakeLists.txt:133 (find_package)                                                                                                                                   
This warning is for project developers.  Use -Wno-dev to suppress it.                                                                                                 
                                                                                                                                                                      
CMake Warning (dev) at /usr/share/cmake-3.24/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args`                                                                                                      
  (libosmocodec) does not match the name of the calling package                                                                                                       
  (Libosmocodec).  This can lead to problems in calling code that expects                                                                                             
  `find_package` result variables (e.g., `_FOUND`) to follow a certain                                                                                                
  pattern.                                                                                                                                                            
Call Stack (most recent call first):                                                                                                                                  
  cmake/Modules/FindLibosmocodec.cmake:31 (find_package_handle_standard_args)                                                                                         
  CMakeLists.txt:133 (find_package)                                                                                                                                   
This warning is for project developers.  Use -Wno-dev to suppress it.                                                                                                 
                                                                                                                                                                      
CMake Warning (dev) at /usr/share/cmake-3.24/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args`                                                                                                      
  (libosmocoding) does not match the name of the calling package                                                                                                      
  (Libosmocoding).  This can lead to problems in calling code that expects                                                                                            
[... cut ...]
./lib/autoconf/general.m4:2249: AC_CACHE_VAL is expanded from...
./lib/autoconf/general.m4:2270: AC_CACHE_CHECK is expanded from...
m4/ax_pthread.m4:88: AX_PTHREAD is expanded from...
configure.ac:85: the top level
Makefile.am:72: warning: DOXYGEN was already defined in condition TRUE, which includes condition !HAVE_DOXYGEN ...
configure.ac:117: ... 'DOXYGEN' previously defined here
```
I'd like to know if the error I got during the installation gr-gsm depends on this error I got with libosmocore. How can I fix them?

ptrkrysik, 2026-11-09:
Duplicate of #582. Duplicate of #582 (GrSwig missing because of the unported GNU Radio 3.9+ API); the new GNU Radio 3.10 port fixes this.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik