ptrkrysik/gr-gsm #598: Gr-gsm failures =(

jackattack21, 2022-02-18:
Hey all - I have been having some issues getting things going. I've worked through a number of the different repos that reference different dependencies and such,  but many were throwing errors (unbuntu 20.04 mostly). I couldn't for the life of me figure out what it was. 

Anyways, I now have a system running on a raspberry pi 4, and gr-gsm is running well with no errors. 

Still when I run grgsm_scanner I get no output, just 00000000

however when I run kal -s GSM850 I am able to identify nearby BTS's their relative frequencies. 849.6 seems to be the highest power. 

With that, I then try: 
grgsm_livemon -f 849.6 -g 40 

and I again just see 0000000000000000000000000000000000000000

My spectrum chart does show peaks at the GSM freqs, but I am not identifying any packets. 

Any ideas!??

jackattack21, 2022-02-18:
Interesting. Thanks for the insight. Do you think I can lower the sample rate and be ok? There's a number of users who have success with the rpi4 - this was the post I followed: https://github.com/gordonwelchman/rpi_imsi_catcher

i also saw a similar issue with my Ubuntu install. All O's and that was on a dell xps developer edition

jackattack21, 2022-02-20:
Bumping this in desperation =(

velichkov, 2022-02-21:
>  Do you think I can lower the sample rate and be ok? 

Yes, you can try different values.

> There's a number of users who have success with the rpi4 - this was the post I followed: https://github.com/gordonwelchman/rpi_imsi_catcher

Better ask them how to properly install, configure and optimize `gr-gsm` for rpi4. Also read older discussions here in the github issues or in the [mailing list](https://github.com/ptrkrysik/gr-gsm#mailing-list), for sure there were few threads about rpi.

> i also saw a similar issue with my Ubuntu install. All O's and that was on a dell xps developer edition

That's strange! There was something wrong with your installation.

jackattack21, 2022-02-24:
I cant for the life of me get this thing to work - multiple distros, multiple tutorials, multiple SDRs, same story.

example attached... ugh
[image]
[image]
[image]

jackattack21, 2022-02-24:
Same issue when I run a more powerful system that can handle greater sample rate also.

jackattack21, 2022-02-27:
Any ideas given the new info?

velichkov, 2022-02-28:
> I cant for the life of me get this thing to work - multiple distros, multiple tutorials, multiple SDRs, same story.
> Same issue when I run a more powerful system that can handle greater sample rate also.
>  Any ideas given the new info?

Do you have a suitable antenna attached to your SDR devices? In case you are inside of a big building try moving near a window or outside if possible. Have you tried different `gain` settings and why you did you choose `-g 40`?

Are you sure that there are GSM cells on these frequencies? `kal` is known to report false positives.

HackRF has a [big DC offset](https://hackrf.readthedocs.io/en/latest/faq.html#what-is-the-big-spike-in-the-center-of-my-received-spectrum) that can negatively affect some of the gr-gsm's tools.

ptrkrysik, 2026-11-09:
The 'O' characters indicate SDR overflows (host too slow/sample rate), and the other points (antenna, gain, HackRF DC offset) were discussed; this is a usage/hardware question, so closing.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik