ptrkrysik/gr-gsm #599: gr-gsm stuck upon make compile

nihal993, 2022-02-28:
I'm trying to install gr-gsm on a Raspberry Bi 3 Model B.
Operating System: Debian GNU/Linux 11 (bullseye)
Kernel: Linux 5.10.92-v8+
Architecture: arm64

I'm following the official installation wiki [here](https://github.com/ptrkrysik/gr-gsm/wiki/Installation-on-RaspberryPi-3).

Everything runs smooth until the execution of `make` command inside `/home/pi/gr-gsm/build` that remains stuck at 70%, without returning any error. It simply does not continue compiling and I've waited for over 2 hours before aborting.
Here is the output I'm getting:
`pi@raspberrypi:~/gr-gsm/build $ make
Scanning dependencies of target grgsm
[  1%] Building CXX object lib/CMakeFiles/grgsm.dir/constants.cc.o
[  2%] Building CXX object lib/CMakeFiles/grgsm.dir/decoding/openbts/AmrCoder.cpp.o
[  3%] Building CXX object lib/CMakeFiles/grgsm.dir/decoding/openbts/BitVector.cpp.o
[  4%] Building CXX object lib/CMakeFiles/grgsm.dir/decoding/openbts/GSM610Tables.cpp.o
[  5%] Building CXX object lib/CMakeFiles/grgsm.dir/decoding/openbts/GSM660Tables.cpp.o
[  6%] Building CXX object lib/CMakeFiles/grgsm.dir/decoding/openbts/GSM503Tables.cpp.o
[  7%] Building CXX object lib/CMakeFiles/grgsm.dir/decoding/openbts/ViterbiR204.cpp.o
[  8%] Building CXX object lib/CMakeFiles/grgsm.dir/decoding/control_channels_decoder_impl.cc.o
[ 10%] Building CXX object lib/CMakeFiles/grgsm.dir/decoding/tch_f_decoder_impl.cc.o
[ 11%] Building CXX object lib/CMakeFiles/grgsm.dir/decoding/tch_h_decoder_impl.cc.o
[ 12%] Building C object lib/CMakeFiles/grgsm.dir/decoding/sch.c.o
[ 13%] Building CXX object lib/CMakeFiles/grgsm.dir/decryption/decryption_impl.cc.o
[ 14%] Building CXX object lib/CMakeFiles/grgsm.dir/demapping/tch_f_chans_demapper_impl.cc.o
[ 15%] Building CXX object lib/CMakeFiles/grgsm.dir/demapping/tch_h_chans_demapper_impl.cc.o
[ 16%] Building CXX object lib/CMakeFiles/grgsm.dir/demapping/universal_ctrl_chans_demapper_impl.cc.o
[ 17%] Building CXX object lib/CMakeFiles/grgsm.dir/flow_control/burst_fnr_filter_impl.cc.o
[ 19%] Building CXX object lib/CMakeFiles/grgsm.dir/flow_control/burst_sdcch_subslot_filter_impl.cc.o
[ 20%] Building CXX object lib/CMakeFiles/grgsm.dir/flow_control/burst_sdcch_subslot_splitter_impl.cc.o
[ 21%] Building CXX object lib/CMakeFiles/grgsm.dir/flow_control/burst_timeslot_filter_impl.cc.o
[ 22%] Building CXX object lib/CMakeFiles/grgsm.dir/flow_control/burst_timeslot_splitter_impl.cc.o
[ 23%] Building CXX object lib/CMakeFiles/grgsm.dir/flow_control/burst_type_filter_impl.cc.o
[ 24%] Building CXX object lib/CMakeFiles/grgsm.dir/flow_control/dummy_burst_filter_impl.cc.o
[ 25%] Building CXX object lib/CMakeFiles/grgsm.dir/flow_control/uplink_downlink_splitter_impl.cc.o
[ 26%] Building CXX object lib/CMakeFiles/grgsm.dir/misc_utils/burst_file_sink_impl.cc.o
[ 28%] Building CXX object lib/CMakeFiles/grgsm.dir/misc_utils/burst_file_source_impl.cc.o
[ 29%] Building CXX object lib/CMakeFiles/grgsm.dir/misc_utils/bursts_printer_impl.cc.o
[ 30%] Building CXX object lib/CMakeFiles/grgsm.dir/misc_utils/collect_system_info_impl.cc.o
[ 31%] Building CXX object lib/CMakeFiles/grgsm.dir/misc_utils/controlled_fractional_resampler_cc_impl.cc.o
[ 32%] Building CXX object lib/CMakeFiles/grgsm.dir/misc_utils/controlled_rotator_cc_impl.cc.o
[ 33%] Building CXX object lib/CMakeFiles/grgsm.dir/misc_utils/extract_cmc_impl.cc.o
[ 34%] Building CXX object lib/CMakeFiles/grgsm.dir/misc_utils/extract_assignment_cmd_impl.cc.o
[ 35%] Building CXX object lib/CMakeFiles/grgsm.dir/misc_utils/extract_immediate_assignment_impl.cc.o
[ 37%] Building CXX object lib/CMakeFiles/grgsm.dir/misc_utils/extract_system_info_impl.cc.o
[ 38%] Building CXX object lib/CMakeFiles/grgsm.dir/misc_utils/message_file_sink_impl.cc.o
[ 39%] Building CXX object lib/CMakeFiles/grgsm.dir/misc_utils/message_file_source_impl.cc.o
[ 40%] Building CXX object lib/CMakeFiles/grgsm.dir/misc_utils/message_printer_impl.cc.o
[ 41%] Building CXX object lib/CMakeFiles/grgsm.dir/misc_utils/msg_to_tag_impl.cc.o
[ 42%] Building CXX object lib/CMakeFiles/grgsm.dir/misc_utils/tmsi_dumper_impl.cc.o
[ 43%] Building CXX object lib/CMakeFiles/grgsm.dir/misc_utils/time_spec.cc.o
[ 44%] Building CXX object lib/CMakeFiles/grgsm.dir/misc_utils/fn_time.cc.o
[ 46%] Building CXX object lib/CMakeFiles/grgsm.dir/misc_utils/udp_socket.cc.o
In file included from /usr/include/boost/smart_ptr/detail/sp_thread_sleep.hpp:22,
                 from /usr/include/boost/smart_ptr/detail/yield_k.hpp:23,
                 from /usr/include/boost/smart_ptr/detail/spinlock_gcc_atomic.hpp:14,
                 from /usr/include/boost/smart_ptr/detail/spinlock.hpp:42,
                 from /usr/include/boost/smart_ptr/detail/spinlock_pool.hpp:25,
                 from /usr/include/boost/smart_ptr/shared_ptr.hpp:29,
                 from /usr/include/boost/shared_ptr.hpp:17,
                 from /usr/include/gnuradio/thread/thread.h:26,
                 from /home/pi/gr-gsm/lib/misc_utils/udp_socket.cc:27:
/usr/include/boost/bind.hpp:36:1: note: ‘#pragma message: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.’
   36 | BOOST_PRAGMA_MESSAGE(
      | ^~~~~~~~~~~~~~~~~~~~
[ 47%] Building CXX object lib/CMakeFiles/grgsm.dir/misc_utils/burst_to_fn_time_impl.cc.o
[ 48%] Building CXX object lib/CMakeFiles/grgsm.dir/qa_utils/burst_sink_impl.cc.o
[ 49%] Building CXX object lib/CMakeFiles/grgsm.dir/qa_utils/burst_source_impl.cc.o
[ 50%] Building CXX object lib/CMakeFiles/grgsm.dir/qa_utils/message_sink_impl.cc.o
[ 51%] Building CXX object lib/CMakeFiles/grgsm.dir/qa_utils/message_source_impl.cc.o
[ 52%] Building CXX object lib/CMakeFiles/grgsm.dir/receiver/clock_offset_control_impl.cc.o
[... 25 lines cut ...]
Scanning dependencies of target _grgsm_swig_doc_tag
[ 65%] Building CXX object swig/CMakeFiles/_grgsm_swig_doc_tag.dir/_grgsm_swig_doc_tag.cpp.o
[ 66%] Linking CXX executable _grgsm_swig_doc_tag
[ 66%] Built target _grgsm_swig_doc_tag
Scanning dependencies of target grgsm_swig_swig_doc
[ 67%] Generating doxygen xml for grgsm_swig_doc docs
warning: Tag 'TCL_SUBST' at line 207 of file '/home/pi/gr-gsm/build/swig/grgsm_swig_doc_swig_docs/Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'COLS_IN_ALPHA_INDEX' at line 854 of file '/home/pi/gr-gsm/build/swig/grgsm_swig_doc_swig_docs/Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'PERL_PATH' at line 1654 of file '/home/pi/gr-gsm/build/swig/grgsm_swig_doc_swig_docs/Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'MSCGEN_PATH' at line 1675 of file '/home/pi/gr-gsm/build/swig/grgsm_swig_doc_swig_docs/Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: argument 'a4wide' for option PAPER_TYPE is not a valid enum value
Using the default: a4!
[ 68%] Generating python docstrings for grgsm_swig_doc
[ 68%] Built target grgsm_swig_swig_doc
Scanning dependencies of target grgsm_swig_swig_compilation
[ 69%] Swig source grgsm_swig.i
[ 69%] Built target grgsm_swig_swig_compilation
Scanning dependencies of target grgsm_swig
[ 70%] Building CXX object swig/CMakeFiles/grgsm_swig.dir/CMakeFiles/grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx.o
`
Please note that I'm trying to build `maint-3.8` branch from velichkov fork by doing `git clone -b maint-3.8 https://github.com/velichkov/gr-gsm.git`

velichkov, 2022-02-28:
Hi @nihal993,

> Please note that I'm trying to build maint-3.8 branch from velichkov fork by doing git clone -b maint-3.8 https://github.com/velichkov/gr-gsm.git

You don't have to build from my fork, all changes have already been merged in the main repository.

> that remains stuck at 70%, without returning any error.

Wait... it takes time. You can try adding some swap space but other then then you need to wait. Raspberry Pi devices are not powerful enough to build as fast as you may expect. 

Also most probably your RPi is not powerful enough to run all gr-gsm tools. My advice is first to install gr-gsm on a normal PC and get used to all tools there. There are various Raspberry Pi related discussions in other issues and in the [mailing list](https://github.com/ptrkrysik/gr-gsm#mailing-list) - read them carefully.

nihal993, 2022-02-28:
Thank you @velichkov for your fast reply.

> You don't have to build from my fork, all changes have already been merged in the main repository.

Thank you for the information.

> Wait... it takes time. You can try adding some swap space but other then then you need to wait. Raspberry Pi devices are not powerful enough to build as fast as you may expect.

I've already added swap space, probably wasn't still enough.

> Also most probably your RPi is not powerful enough to run all gr-gsm tools. My advice is first to install gr-gsm on a normal PC and get used to all tools there. There are various Raspberry Pi [...]

That's exactly what I've already done, as a matter of fact I've tested gr-gsm on my computer before attempting to install on the RPi. Thank you anyway for pointing out all RPi related issues, it is very helpful for me.

For the record I was finally able to build by forcing the use of `python 3.8.0` with `pyenv`.
I don't know if it was the right approach, but it eventually worked out.

velichkov, 2022-02-28:
> For the record I was finally able to build by forcing the use of python 3.8.0 with pyenv.
I don't know if it was the right approach, but it eventually worked out.

You should build with the same python version your gnuradio was built with.

Let us know how it goes and whether you are able to use gr-gsm on your RPi 3.

ptrkrysik, 2026-11-09:
Building on a Raspberry Pi is simply slow and the user resolved it by matching Python versions; this is a build-environment question, not a code defect.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.